open-mmlab/OpenPCDet #734: Dose anyone have config of pointpillars_multi_head on kitti dataset?

DuZzzs, 2022-01-04:
I modified a kitti_multi_head config based on nusences pointpillars_multi_head config, but the performance is 10 points lower than  a single header. Dose anyone have multi_head config on kitti dataset? Thanks.
poor performance: 
```
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:59.5336, 55.9778, 54.3823
bev  AP:53.2788, 48.9597, 46.2426
3d   AP:46.3634, 42.3372, 39.8900
aos  AP:38.48, 36.06, 34.48
```

daeunni, 2022-01-11:
Hi, 
I did the same thing as you. And the following performance was obtained.
can i show your config file?

```
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:68.5132, 64.3342, 61.1776
bev  AP:63.6460, 57.6637, 54.4449
3d   AP:57.6386, 52.4393, 48.2287
aos  AP:51.53, 48.54, 45.88
```

daeunni, 2022-01-11:
Also I wonder why the performance improved when I changed to multihead.

DuZzzs, 2022-01-12:
@daeunni Using a multi head configuration can slightly improve performance. You can refer to [here](https://github.com/open-mmlab/OpenPCDet/issues/677).
Here is my bad config:
```
CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist', 'Truck']

DATA_CONFIG:
  _BASE_CONFIG_: cfgs/dataset_configs/untouch_dataset.yaml

  POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
  DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    -   NAME: shuffle_points
        SHUFFLE_ENABLED: {
          'train': True,
          'test': False
        }

    -   NAME: transform_points_to_voxels
        VOXEL_SIZE: [0.2, 0.2, 4.0]
        MAX_POINTS_PER_VOXEL: 32
        MAX_NUMBER_OF_VOXELS: {
          'train': 30000,
          'test': 30000
        }

  DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
      - NAME: gt_sampling
        USE_ROAD_PLANE: False
        DB_INFO_PATH:
          - kitti_dataset/kitti_dbinfos_train.pkl
        PREPARE: {
          filter_by_min_points: ['Car:20', 'Pedestrian:20', 'Cyclist:20', 'Truck:20'],
          filter_by_difficulty: [-1],
        }

        SAMPLE_GROUPS: ['Car:15','Pedestrian:10', 'Cyclist:10', 'Truck:15']
        NUM_POINT_FEATURES: 4
        DATABASE_WITH_FAKELIDAR: False
        REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
        LIMIT_WHOLE_SCENE: False

      - NAME: random_world_flip
        ALONG_AXIS_LIST: ['x']

      - NAME: random_world_rotation
        WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

      - NAME: random_world_scaling
        WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: PointPillar

    VFE:
        NAME: PillarVFE
        WITH_DISTANCE: False
        USE_ABSLOTE_XYZ: True
        USE_NORM: True
        NUM_FILTERS: [64]

    MAP_TO_BEV:
        NAME: PointPillarScatter
        NUM_BEV_FEATURES: 64

    BACKBONE_2D:
        NAME: BaseBEVBackbone
        LAYER_NUMS: [3, 5, 5]
        LAYER_STRIDES: [2, 2, 2]
        NUM_FILTERS: [64, 128, 256]
        UPSAMPLE_STRIDES: [1, 2, 4]  # kitti
        NUM_UPSAMPLE_FILTERS: [128, 128, 128]

    DENSE_HEAD:
        NAME: AnchorHeadMulti
        CLASS_AGNOSTIC: False

        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        USE_MULTIHEAD: True
        SEPARATE_MULTIHEAD: True
        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            },
            {
                'class_name': 'Pedestrian',
                'anchor_sizes': [[0.8, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            },
            {
                'class_name': 'Cyclist',
                'anchor_sizes': [[1.76, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            },
            {
                'class_name': 'Truck',
                'anchor_sizes': [[10.5, 2.94, 3.47]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.085],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.55,
                'unmatched_threshold': 0.4
            },
        ]

        SHARED_CONV_NUM_FILTER: 64

        RPN_HEAD_CFGS: [
            {
                'HEAD_CLS_NAME': ['Car'],
            },
            {
                'HEAD_CLS_NAME': ['Pedestrian'],
            },
            {
                'HEAD_CLS_NAME': ['Cyclist'],
            },
            {
                'HEAD_CLS_NAME': ['Truck'],
            },
        ]
        SEPARATE_REG_CONFIG:
            NUM_MIDDLE_CONV: 1
            NUM_MIDDLE_FILTER: 64
            REG_LIST: ['reg:2', 'height:1', 'size:3', 'angle:2']

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder
            BOX_CODER_CONFIG: {
                'code_size': 7,
                'encode_angle_by_sincos': True
            }

        LOSS_CONFIG:
            REG_LOSS_TYPE: WeightedL1Loss
            LOSS_WEIGHTS: {
                'pos_cls_weight': 1.0,
                'neg_cls_weight': 2.0,
                'cls_weight': 1.0,
                'loc_weight': 0.25,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: True
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.2
            NMS_PRE_MAXSIZE: 1000
            NMS_POST_MAXSIZE: 100

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 2
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.001
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

```
Can you share your config? This is my email：duzchn@gmail.com. Thanks.

daeunni, 2022-01-12:
Did you use kitti dataset?

I'm not sure, but this part doesn't exist in my config. 
```
        SEPARATE_REG_CONFIG:
            NUM_MIDDLE_CONV: 1
            NUM_MIDDLE_FILTER: 64
            REG_LIST: ['reg:2', 'height:1', 'size:3', 'angle:2']
```
because I thought this part was not necessary in the Pointpillar model.

daeunni, 2022-01-12:
BTW, I want to ask how you found the **anchor size** of the truck class.

`Truck's anchor_sizes : [[10.5, 2.94, 3.47]],`

and if you need my config file, i can share it using email.

DuZzzs, 2022-01-12:
I searched in other configs, this size is the average size of the data set, I did not use the code to make statistics, if you count it, it will be more accurate. If it is convenient for you, please send your config to my email. Thanks.

Keenthan, 2022-02-08:
> Did you use kitti dataset?
>
> I'm not sure, but this part doesn't exist in my config.
>
> ```
>         SEPARATE_REG_CONFIG:
>             NUM_MIDDLE_CONV: 1
>             NUM_MIDDLE_FILTER: 64 [...]

I'm also confused about these lines. What are these lines of code for? Would you please help explain? Thanks!

closed 2022-03-25 by github-actions[bot]

huihui6666, 2023-04-07:
> 顺便说一句，我想问一下你是如何找到卡车类的**锚尺寸的。**
> 
> `Truck's anchor_sizes : [[10.5, 2.94, 3.47]],`
> 
> 如果您需要我的配置文件，我可以使用电子邮件分享它。

Hi,daeunni
I am doing some training of multi-head,could you please share me your config file of the multi-head pointpillars.
Thanks very much.

huihui6666, 2023-04-08:
> 顺便说一句，我想问一下你是如何找到卡车类的**锚尺寸的。**
> 
> `Truck's anchor_sizes : [[10.5, 2.94, 3.47]],`
> 
> 如果您需要我的配置文件，我可以使用电子邮件分享它。

My email is 1763719731@qq.com
Thanks.