open-mmlab/OpenPCDet #735: How to access "forward_ret_dict"

daeunni, 2022-01-05:
Hi, i want to print "**forward_ret_dict"** 
but I don't know how to access using this code 

"**forward_ret_dict"**  is in here 
> The path is /pcdet/models/dense_heads/**anchor_head_template.py**

```
def get_cls_layer_loss(self):
        
        cls_preds = self.forward_ret_dict['cls_preds'] 
        box_cls_labels = self.forward_ret_dict['box_cls_labels']     

        batch_size = int(cls_preds.shape[0])
        cared = box_cls_labels >= 0  # [N, num_anchors]
        
        positives = box_cls_labels > 0
        negatives = box_cls_labels == 0
        negative_cls_weights = negatives * 1.0`
```

plz help me ~~

jihanyang, 2022-01-05:
Please provide more information such as where you want to access forward_ret_dict

closed 2022-01-06 by daeunni

AJAY31797, 2024-01-27:
Hi @jihanyang 

I am also trying to debug some problems I am having while running the code, for which I want to understand how exactly the forward_ret_dict is getting its values. 
In the anchor_head_template.py, it is first initialized as follows:
`self.forward_ret_dict = {}`

Then, in the get_cls_layer_loss function, the values from the dictionary are being accessed:
`def get_cls_layer_loss(self):
        cls_preds = self.forward_ret_dict['cls_preds']
        box_cls_labels = self.forward_ret_dict['box_cls_labels']`

But I could not understand how is the dictionary getting the values exactly? For instance, from where are the values corresponding to the 'cls_preds' key coming in the dictionary? Could you please provide some guidance on this?

Thanks, 
Ajay

thelandord, 2024-05-20:
Is there a solution?