open-mmlab/OpenPCDet #736: Waymo-Open-Dataset Version Update

aalmuzairee, 2022-01-05:
Hi @sshaoshuai,

Could you please provide support for a newer version of waymo-open-dataset instead of waymo-open-dataset-tf-2-0-0==1.2.0?

The reason being, spconv 2.x only supports CUDA 10.2 and forward, and waymo-open-dataset-tf-2-0-0 uses tensorflow-gpu 2.0 which was built on CUDA 10.0, so its really hard to match newer packages with older ones.

Appreciate the help.

sshaoshuai, 2022-01-05:
I have updated the documents about this command. Actually we do not need to do any changes the codes of OpenPCDet. It should work well with the latest version of waymo-open-dataset.

aalmuzairee, 2022-01-05:
Awesome, thanks!

Do the numbers in this [post](370#issuecomment-736683423) still hold using waymo-open-dataset-tf-2-5-0?

sshaoshuai, 2022-01-05:
I always worked with waymo-open-dataset-tf-2-5-0 recently and do not find any problems currently.

closed 2022-01-06 by sshaoshuai