open-mmlab/OpenPCDet #741: pcdet052 multi gpu training problem

AndyYuan96, 2022-01-07:
Hi, I just update to pcdet052, and use Waymo to train, I find that when use multi gpu training and workers > 0, and when using gt aug, the code will stuck long time before start to training , but if I set workers = 0, quickly start to train. so either set workers = 0 and use gt aug, or set workers > 0 and don't use gtaug, I don't meet that problem in pcdet030.

jihanyang, 2022-01-07:
Hello, I guess this is also caused by the larger size of gt database compared to pcdet030. A preivous issue also mentioned this but focus more on the larger memory usage. I think you can try to use shm to solve this.

AndyYuan96, 2022-01-08:
@sshaoshuai, it works, after comment the line, quickly start to train.

AndyYuan96, 2022-01-08:
what's more, the reason why starting to train so slow is not related with gtaug, even I use the gtaug generated before which sample interval is 10, it still use very long time to start to training, and when use more gpu like 8 v100, after stuck 2 hours, it still don't start to train.

closed 2022-01-10 by AndyYuan96

Angericky, 2022-01-20:
Hi, I meet a similar problem.

In pcdet052, when I run detection on waymo dataset, gt sampling comsumes lots of time, which causes my training process unreasonble slow.
If I comment `gt sampling` in config files, the training is quick.

I checked running time of gt sampling, and noticed large time cost in [database_sampler.py](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/augmentor/database_sampler.py#L181-L182)

Where could the problem lie in? 

BTW, this does not work in my condition.

> If you use `dist_train.sh`, please try to comment the following two lines:
> 
> https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/utils/common_utils.py#L162-L163
> 
> It is expected to be much faster to start the training.
> 
> After your try, I appreciate if you could come back to tell us whether it works.

jihanyang, 2022-01-20:
Use shared_memory might solve your problem
https://github.com/open-mmlab/OpenPCDet/blob/5861246c31c876f0e5a3e125086ed2b0be7bc9da/tools/cfgs/dataset_configs/waymo_dataset.yaml#L31

eriche2016, 2022-02-11:
> If you use `dist_train.sh`, please try to comment the following two lines:
>
> https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/utils/common_utils.py#L162-L163
>
> It is expected to be much [...]

I have tried it, but it is still to slow to start on kitti dataset.