open-mmlab/OpenPCDet #743: 训练SECOND时出现错误Error training on SECOND

kellen5l, 2022-01-08:
我在训练SECOND时出现了下面的问题，查看其他Issues了解到是`USE_ROAD_PLANE`这个参数的问题，但我在`second.yaml`文件中未找到这个参数，请问该怎么解决？
环境：python3.7, pytorch 1.10, cuda11.3
I had the following problem when training second. I checked other issues and learned that it was the problem of the parameter "use_road_plane", but I didn't find this parameter in the "second.yaml" file. How can I solve it?
environment: python3.7, pytorch 1.10, cuda11.3
```
2022-01-08 21:30:17,060   INFO  **********************Start training kitti_models/second(default)**********************
epochs:   0%|                                                                                                                                                                          | 0/80 [00:00<?, ?it/s]
Traceback (most recent call last):                                                                                                                                                    | 0/928 [00:00<?, ?it/s]
  File "train.py", line 201, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/featurize/OpenPCDet/tools/train_utils/train_utils.py", line 118, in train_model
    dataloader_iter=dataloader_iter
  File "/home/featurize/OpenPCDet/tools/train_utils/train_utils.py", line 25, in train_one_epoch
    batch = next(dataloader_iter)
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/_utils.py", line 434, in reraise
    raise exception
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/environment/miniconda3/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 424, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "../pcdet/datasets/dataset.py", line 130, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 218, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "../pcdet/datasets/augmentor/database_sampler.py", line 245, in __call__
    data_dict = self.add_sampled_boxes_to_scene(data_dict, sampled_gt_boxes, total_valid_sampled_dict)
  File "../pcdet/datasets/augmentor/database_sampler.py", line 163, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
KeyError: 'road_plane'
```

closed 2022-01-09 by kellen5l

Dev-HJYoo, 2023-02-22:
@kellen5l How to solve this problem? i meet same error :(

Bennyliiiiii, 2024-01-11:
@kellen5l How to solve this problem? i meet same error :(

Cocoon-Ray, 2024-04-27:
Corresponding key is inherited in the kitti_dataset.yaml, please modify the USE_ROAD_PLANE in that place