open-mmlab/OpenPCDet #745: RuntimeError: Cannot install on Python version 3.6.13

RonaldYuren, 2022-01-10:
1.as mentioned, I can't run the " python setup.py develop "  mentioned in the INSTALL.md by python3.6.13
//
RuntimeError: Cannot install on Python version 3.6.13; only versions >=3.7,<3.11                                                                                                                               are supported.
//

2. I am using the CUDA 9.1 and Pytorch1.1 in UBUNTU , I think that the sponcv V2.x would not come up any errors, right ?

jihanyang, 2022-01-10:
spconv v2.x don't not support CUDA 9.1.

RonaldYuren, 2022-01-10:
> spconv v2.x don't not support CUDA 9.1.

then how to download the former version of spconv?

jihanyang, 2022-01-10:
You should raise this issue in Spconv.

closed 2022-01-10 by jihanyang

HenryJunW, 2022-01-16:
I am using Python 3.6.13 and spconv 1.2, and face the similar #1 problem. Solved by 'conda install numba==0.49.1'. Since default 'numba-0.55.0' requires python 3.7. Hope that helps.