open-mmlab/OpenPCDet #747: PV-RCNN++ KITTI config

MartinHahner, 2022-01-10:
@sshaoshuai: Can you please also provide your PV-RCNN++ config **for the KITTI dataset**?

sshaoshuai, 2022-01-12:
For PV-RCNN++, I did most experiments on the Waymo dataset and I have only checked the performance of current released codes on Waymo dataset (the code has been refactored for code release). 

For the KITTI results, you may firstly try to simply replace the VSA config and RoI-grid config of PV-RCNN by using the config of PV-RCNN++ (waymo config), since currently I do not have the KITTI config with my released codes (I may find time to write a config of KITTI and check the performance before it is released).

MartinHahner, 2022-01-12:
> I may find time to write a config of KITTI and check the performance before it is released

That would be awesome!

MartinHahner, 2022-03-01:
This topic is not solved yet.

sshaoshuai, 2022-03-12:
Close this issue since the config has been given in https://github.com/open-mmlab/OpenPCDet/issues/829.

closed 2022-03-12 by sshaoshuai