open-mmlab/OpenPCDet #748: demo.py error

yun9993, 2022-01-12:
--data_path 000014.binb6808:/data8/lyp/OpenPCDet/tools# python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth -
/root/anaconda3/envs/pillars/lib/python3.7/site-packages/skimage/io/manage_plugins.py:23: UserWarning: Your installed pillow version is < 8.1.2. Several security issues (CVE-2021-27921, CVE-2021-25290, CVE-2021-25291, CVE-2021-25293, and more) have been fixed in pillow 8.1.2 or higher. We recommend to upgrade this library.
  from .collection import imread_collection_wrapper
2022-01-12 20:13:11,089   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2022-01-12 20:13:11,089   INFO  Total number of samples:        1
/root/anaconda3/envs/pillars/lib/python3.7/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1639180589158/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2022-01-12 20:13:13,015   INFO  ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2022-01-12 20:13:13,343   INFO  ==> Done (loaded 367/367)
2022-01-12 20:13:13,546   INFO  Visualized sample index:        1
[Open3D WARNING] GLFW Error: X11: The DISPLAY environment variable is missing
[Open3D WARNING] Failed to initialize GLFW
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 102, in main
    ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
  File "/data8/lyp/OpenPCDet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
    vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'

LouisSF, 2022-01-12:
Your error seems to come from Open3D, do you have access to a graphics display through your terminal, with OpenGL installed and the latest drivers ? Working in ssh without the -X option will produce the same error for instance.

IzzeddinTeeti, 2022-02-03:
> 

Hi there, I have the same problem, I working on a server through ssh, and when I run the demo, it shoots the following error:

`[Open3D WARNING] GLFW Error: X11: The DISPLAY environment variable is missing
[Open3D WARNING] Failed to initialize GLFW
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 100, in main
    V.draw_scenes(
  File "OpenPCDet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
    vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'`

MartinHahner, 2022-02-08:
demo.py only works when you run it on a machine where a display is connected.
I implemented a workaround to store the predictions as pickle on disk and can then visualize the predictions with my [viewer](https://github.com/MartinHahner/LiDAR_fog_sim/blob/main/pointcloud_viewer.py).

Here is my demo.py, hope it helps.
```python
import os
import glob
import torch
import pickle
import shutil
import argparse

import numpy as np

from tqdm import tqdm
from pathlib import Path

from pcdet.config import cfg, cfg_from_yaml_file
from pcdet.datasets import DatasetTemplate
from pcdet.models import build_network, load_data_to_gpu
from pcdet.utils import common_utils

if os.name == 'posix' and "DISPLAY" not in os.environ:
    headless_server = True
else:
    headless_server = False
    from tools.visual_utils import visualize_utils as V
    import mayavi.mlab as mlab

def parse_config():

    parser = argparse.ArgumentParser(description='arg parser')

    parser.add_argument('-m', '--method', type=str, default='pv_rcnn', help='specify the method')
    parser.add_argument('-d', '--dataset', type=str, default='DENSE', help='specify the dataset')
    parser.add_argument('-r', '--dataset_root', type=str, help='specify the dataset root',
                        default='/srv/beegfs-benderdata/scratch/tracezuerich/data/datasets/')
    parser.add_argument('-c', '--copy_image', type=bool, help='specify if images should be copied to DEMO folder',
                        default=True)

    args = parser.parse_args()

    cfg_file = get_cfg_file(args.method)

    cfg_from_yaml_file(cfg_file, cfg)

    return args, cfg

class DemoDataset(DatasetTemplate):

    def __init__(self, dataset_cfg, class_names, training=True, root_path=None, logger=None, ext='.bin',
                 num_features=4):
        """
        Args:
            root_path:
            dataset_cfg:
            class_names:
            training:
            logger:
        """
        super().__init__(
            dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
        )
        self.root_path = root_path
        self.ext = ext
        data_file_list = glob.glob(str(root_path / f'*{self.ext}')) if self.root_path.is_dir() else [self.root_path]

        data_file_list.sort()
        self.sample_file_list = data_file_list
        self.num_features = num_features
[... 146 lines cut ...]

                with open(f'{export_dir}/data_dict_{idx+1:06d}.pkl', 'wb') as f:
                    pickle.dump(data_dict_cpu, f, pickle.HIGHEST_PROTOCOL)

                for pred_dict in pred_dicts:

                    for key, value in pred_dict.items():
                        if isinstance(pred_dict[key], torch.Tensor):
                            pred_dict[key] = value.cpu().numpy()

                with open(f'{export_dir}/pred_dicts_{idx + 1:06d}.pkl', 'wb') as f:
                    pickle.dump(pred_dicts, f, pickle.HIGHEST_PROTOCOL)

            else:
                V.draw_scenes(
                    points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
                    ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
                )
                mlab.show(stop=True)

    logger.info('Demo done.')

if __name__ == '__main__':
    main()
```

jinhuilin97, 2022-09-29:
if x11 has some problem, maybe it can help 
https://blog.csdn.net/weixin_44287798/article/details/126925297#comments_23422299

souravraha, 2022-10-17:
@yun9993 I was facing the exact same error. I had been using Xming (obsolete, free version) + PuTTY for remote display on my Windows laptop. When I switched to VcXsrv (in place of Xming), there were no more glitches. See if this works for you.

zbaishancha, 2022-11-26:
> 
但是当我使用本地的pycharm去连接远程服务器去运行相同的demo，如何将其显示呢？

zbaishancha, 2022-11-26:
@souravraha

souravraha, 2023-01-19:
> > 
> 
> 但是当我使用本地的pycharm去连接远程服务器去运行相同的demo，如何将其显示呢？

Sorry, I am not very sure about PyCharm. I use VcXrcv for remote display and PuTTy (with port forwarding and X11 forwarding enabled) for establishing the ssh connection.

wanglaotou, 2023-02-10:
https://github.com/open-mmlab/OpenPCDet/blob/31f6758af378c20e6200cd8e16307b1afa8c8a42/tools/demo.py#L100

pred_dicts, _ = model.forward(data_dict)
            
print('result:', pred_dicts[0]['pred_boxes'], pred_dicts[0]['pred_scores'])

# V.draw_scenes(
#     points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
#     ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
# )

# if not OPEN3D_FLAG:
#     mlab.show(stop=True)

getabear, 2023-07-11:
maybe this website will help you solve this problem. Insatll the X11. https://xie.infoq.cn/article/31c321e9fbf4c0fc1427600c2.

stu-zhangxl, 2023-07-14:
> 
Hello, I have also encountered this problem. 
```bash
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 100, in main
    V.draw_scenes(
  File "open_pcdet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
    vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'
```
May I ask how to use the visualization of demo.py based on the py file you provided, if it is saved as a local file? Is it still similar to the following method? 
```bash
python -s demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml  --ckpt xxx/pointpillar_7728.pth --data_path xxx/kitti_test_data/000000.bin
```

whzz101, 2023-08-02:
用ssh连接主机运行demo时报该错误，是因为渲染点云时，需要gpu的支持，但在windows环境下，没有LINUX对应的GPU，所以是不可能构建窗口并显示的，最简单的方式就是在远程桌面登录linux，在linux就可以完美运行