open-mmlab/OpenPCDet #749: mayavi  ；pyqt5；“xcb”

yun9993, 2022-01-13:
Got keys from plugin meta data ("webgl")
QFactoryLoader::QFactoryLoader() looking at "/root/anaconda3/envs/environment_name/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqxcb.so"
Found metadata in lib /root/anaconda3/envs/environment_name/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/root/anaconda3/envs/environment_name/bin/platforms" ...
loaded library "/root/anaconda3/envs/environment_name/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqxcb.so"
qt.qpa.xcb: could not connect to display :0.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

jihanyang, 2022-01-14:
If you want to visualize the point cloud, maybe open3d will have less problem on environment setup.

yun9993, 2022-01-14:
> 如果你想可视化点云，也许open3d在环境设置上的问题会更少。

I also encountered errors when trying Open3D.
----------------------
--data_path 000014.binb6808:/data8/lyp/OpenPCDet/tools# python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth -
/root/anaconda3/envs/pillars/lib/python3.7/site-packages/skimage/io/manage_plugins.py:23: UserWarning: Your installed pillow version is < 8.1.2. Several security issues (CVE-2021-27921, CVE-2021-25290, CVE-2021-25291, CVE-2021-25293, and more) have been fixed in pillow 8.1.2 or higher. We recommend to upgrade this library.
from .collection import imread_collection_wrapper
2022-01-12 20:13:11,089 INFO -----------------Quick Demo of OpenPCDet-------------------------
2022-01-12 20:13:11,089 INFO Total number of samples: 1
/root/anaconda3/envs/pillars/lib/python3.7/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/conda/conda-bld/pytorch_1639180589158/work/aten/src/ATen/native/TensorShape.cpp:2157.)
return _VF.meshgrid(tensors, **kwargs) # type: ignore[attr-defined]
2022-01-12 20:13:13,015 INFO ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2022-01-12 20:13:13,343 INFO ==> Done (loaded 367/367)
2022-01-12 20:13:13,546 INFO Visualized sample index: 1
[Open3D WARNING] GLFW Error: X11: The DISPLAY environment variable is missing
[Open3D WARNING] Failed to initialize GLFW
Traceback (most recent call last):
File "demo.py", line 112, in
main()
File "demo.py", line 102, in main
ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
File "/data8/lyp/OpenPCDet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'

jihanyang, 2022-01-14:
Do you have display device in this machine? If not, please try to use x11 forward with open3d.

yun9993, 2022-01-14:
> 

There is. I tested it on the Ubuntu terminal. So, I don't know how to solve this problem. Please help me

jihanyang, 2022-01-15:
Sorry, I don't have experience on this situation. I just heard this solution from my friend previously.  I think this will not be difficult with simple ssh x11 forwarding.

closed 2022-03-07 by github-actions[bot]