open-mmlab/OpenPCDet #750: my CenterPoint config for KITTI dataset

OuyangJunyuan, 2022-01-13:
This config file is modified from both [official code released for KITTI](https://github.com/tianweiy/CenterPoint-KITTI) and [OpenPCDet version for WAYMO](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/waymo_models/centerpoint_without_resnet.yaml).
And if anyone wants to get a pre-training model, please let me know.
```yaml
CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml

MODEL:
    NAME: CenterPoint

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelResBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5]
        LAYER_STRIDES: [1]
        NUM_FILTERS: [128]
        UPSAMPLE_STRIDES: [2]
        NUM_UPSAMPLE_FILTERS: [256]

    DENSE_HEAD:
        NAME: CenterHead
        CLASS_AGNOSTIC: False

        CLASS_NAMES_EACH_HEAD: [
            [ 'Car', 'Pedestrian', 'Cyclist' ]
        ]

        SHARED_CONV_CHANNEL: 64
        USE_BIAS_BEFORE_NORM: True
        NUM_HM_CONV: 2 #  heatmap
        SEPARATE_HEAD_CFG:
            HEAD_ORDER: ['center', 'center_z', 'dim', 'rot']
            HEAD_DICT: {
                'center': {'out_channels': 2, 'num_conv': 2}, # offset
                'center_z': {'out_channels': 1, 'num_conv': 2},
                'dim': {'out_channels': 3, 'num_conv': 2},
                'rot': {'out_channels': 2, 'num_conv': 2},
            }

        TARGET_ASSIGNER_CONFIG:
            FEATURE_MAP_STRIDE: 4
            NUM_MAX_OBJS: 500
            GAUSSIAN_OVERLAP: 0.1
            MIN_RADIUS: 2

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }
        POST_PROCESSING:
            SCORE_THRESH: 0.1
            POST_CENTER_LIMIT_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
            MAX_OBJ_PER_SAMPLE: 500
            NMS_CONFIG:
                NMS_TYPE: nms_gpu
                NMS_THRESH: 0.7
                NMS_PRE_MAXSIZE: 4096
                NMS_POST_MAXSIZE: 500

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.01
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10
```

EmiyaNing, 2022-01-14:
Can you show the result on kitti val set?

OuyangJunyuan, 2022-01-15:
the result get from tools/test.py are as follwing:
```
Car AP@0.70, 0.70, 0.70:
bbox AP:92.4371, 89.7997, 89.1875
bev  AP:89.0399, 87.5611, 86.4736
3d   AP:87.3393, 79.4999, 77.3796
aos  AP:92.42, 89.72, 89.07
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:97.1744, 93.6886, 92.1958
bev  AP:92.7800, 89.0816, 88.0308
3d   AP:88.9477, 81.6019, 79.1999
aos  AP:97.15, 93.59, 92.05
Car AP@0.70, 0.50, 0.50:
bbox AP:92.4371, 89.7997, 89.1875
bev  AP:92.3280, 91.4396, 89.4304
3d   AP:92.3222, 89.8208, 89.3696
aos  AP:92.42, 89.72, 89.07
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:97.1744, 93.6886, 92.1958
bev  AP:97.1386, 94.7012, 93.3835
3d   AP:97.1264, 93.9522, 93.1182
aos  AP:97.15, 93.59, 92.05
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:74.4678, 71.9525, 69.1918
bev  AP:60.7042, 59.1109, 55.1364
3d   AP:57.3194, 54.2235, 50.6287
aos  AP:72.65, 69.44, 66.21
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:75.1197, 72.9895, 69.8245
bev  AP:60.5372, 58.2984, 54.0787
3d   AP:55.7938, 53.0594, 48.9036
aos  AP:73.24, 70.18, 66.46
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:74.4678, 71.9525, 69.1918
bev  AP:78.9672, 78.2260, 75.2896
3d   AP:78.8915, 77.9085, 75.0464
aos  AP:72.65, 69.44, 66.21
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:75.1197, 72.9895, 69.8245
bev  AP:81.4859, 80.1064, 76.6350
3d   AP:81.3868, 79.6834, 76.2700
aos  AP:73.24, 70.18, 66.46
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:87.5590, 75.4489, 71.9498
bev  AP:88.6395, 68.9562, 65.1039
3d   AP:85.7151, 65.6614, 62.1453
aos  AP:87.46, 74.95, 71.55
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:91.6160, 76.7370, 72.7161
bev  AP:89.0516, 69.3149, 65.6969
3d   AP:85.7833, 65.5833, 62.2746
aos  AP:91.49, 76.25, 72.24
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:87.5590, 75.4489, 71.9498
bev  AP:87.8537, 72.0860, 68.6397
3d   AP:87.8537, 72.0850, 68.6397
aos  AP:87.46, 74.95, 71.55
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:91.6160, 76.7370, 72.7161
bev  AP:91.3929, 73.1575, 69.0485
3d   AP:91.3929, 73.1573, 69.0482
aos  AP:91.49, 76.25, 72.24
```

YoushaaMurhij, 2022-02-16:
Would you please provide centerpoint-pointpillars config?

OuyangJunyuan, 2022-02-17:
> Would you please provide centerpoint-pointpillars config?

it's sry that i didn't try. But it's easy to implement, just using pillar as VFE and checking that `BACKBONE_2D feature map size` equal to `Voxel Grid Size_XY / CenterHead stride `

YoushaaMurhij, 2022-02-17:
@OuyangJunyuan , Thanks for you comment! I have implemented it.

EmiyaNing, 2022-03-11:
It's seems that I can't reproduce your result in my enviroment( RTX 3090 + cuda11.0 + spconv1.2.1 + OpenPCDet v0.5.0). I'm puzzed that the POST_CENTER_LIMIT_RANGE is not suit for KITTI in your configure file.

hhhmrcscs, 2022-03-25:
I need a pre-training mode, can you give me?

angryhen, 2022-03-28:
Hi, can you share your pretrained model? Thanks
copainsir@foxmail.com

OuyangJunyuan, 2022-03-28:
> I need a pre-training mode, can you give me?

> Hi, can you share your pretrained model? Thanks [copainsir@foxmail.com](mailto:copainsir@foxmail.com)

has been send~

OuyangJunyuan, 2022-03-28:
> I need a pre-training mode, can you give me?

any email will be provided？

hhhmrcscs, 2022-03-28:
1780337699@qq.com, Thanks

angryhen, 2022-03-28:
> > I need a pre-training mode, can you give me?
> 
> > Hi, can you share your pretrained model? Thanks [copainsir@foxmail.com](mailto:copainsir@foxmail.com)
> 
> has been sent~

Hi, I used your pretrain-model to test kitti testing velodyne bin, but got error result. 
`
python demo.py --cfg_file cfgs/kitti_models/centerpoints.yaml --ckpt ../checkpoints/centerpoint_kitti_80.pth --data_path ../testing/velodyne/000003.bin
`
[image]

HuangCongQing, 2022-04-19:
> 

Hi, can you share your yaml file? Thanks

HuangCongQing, 2022-04-19:
> @OuyangJunyuan , Thanks for your comment! I have implemented it.

Hi, can you share your yaml file? Thanks
1756260160@qq.com

OuyangJunyuan, 2022-04-23:
Aha, the CenterPoint's yaml file I used in kitti just like aforemention. You can try it, and train&test  in normal way.

Rapisurazurite, 2022-04-23:
> > > 我需要一个预训练模式，你能给我吗？
> > 
> > 
> > > 嗨，你能分享你的预训练模型吗？谢谢[copainsir@foxmail.com](mailto:copainsir@foxmail.com)
> > 
> > 
> > 已发送~
> 
> 嗨，我使用你的预训练模型来测试 kitti 测试 velodyne bin，但得到了错误结果。 `python demo.py --cfg_file cfgs/kitti_models/centerpoints.yaml --ckpt ../checkpoints/centerpoint_kitti_80.pth --data_path ../testing/velodyne/000003.bin` [image]

you can set NMS_THRESH to 0.01 in MODEL.DENSE_HEAD.POST_PROCESSING config. Centerpoint model do nms in centerhead, so the config of MODEL.POST_PROCESSING is useless.

Rapisurazurite, 2022-04-23:
> Hi, I used your pretrain-model to test kitti testing velodyne bin, but got error result. `python demo.py --cfg_file cfgs/kitti_models/centerpoints.yaml --ckpt ../checkpoints/centerpoint_kitti_80.pth [...]

you can set NMS_THRESH to 0.01 in MODEL.DENSE_HEAD.POST_PROCESSING config. Centerpoint model do nms in centerhead, so the config of MODEL.POST_PROCESSING is useless.

OuyangJunyuan, 2022-06-30:
> @OuyangJunyuan 你好，可以分享一下预训练模型吗 [gong20160317@163.com](mailto:gong20160317@163.com)

> Hi, can you share your pretrained model? Thanks a lot! [kuangyongshao1995@163.com](mailto:kuangyongshao1995@163.com)

sry for replying late. the pretrain model can obttain is available [here](https://github.com/open-mmlab/OpenPCDet/issues/750#issue-1101618318)

ghb0224, 2022-06-30:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Sorry,I have not received.
        
    
        
            
        
    
    
     
    
        
                    ***@***.***
        
    
 

  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            ***@***.***>
            
        
      
      
        
         Date 
        
        
        6/30/2022 09:25
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [open-mmlab/OpenPCDet] my CenterPoint config for KITTI dataset (Issue #750)
        
      
    
  
  

@OuyangJunyuan 你好，可以分享一下预训练模型吗 ***@***.***

Hi, can you share your pretrained model? Thanks a lot! ***@***.***

sry for replying late. the pretrain model can obttain is available here

xiaoxin83121, 2022-07-16:
Hi, I have a question about 'FEATURE_MAP_STRIDE: 4'. It seems to be 8, since the backbone downsamples the feature map to 1/8. I have read the code of centerhead and config file of that on Waymo, I think 8 should be the right value. However, stride=4 gets better results than 8. And I have also conducted experiments on stride=1 and stride=2, they get nearly 0 mAP. Hope to get your reply.

OuyangJunyuan, 2022-07-18:
> Hi, I have a question about 'FEATURE_MAP_STRIDE: 4'. It seems to be 8, since the backbone downsamples the feature map to 1/8. I have read the code of centerhead and config file of that on Waymo, I [...]

Indeed, what you thought are approximately right about the `Backbone3D`. But there is one thing you has ignored, i.e., `BEV_Backbone`. Please kindly refer to configuration file that you can find `UPSAMPLE_STRIDES: [2]` in it, resulting in `1/8 * 2 = 1/4` here.

xiaoxin83121, 2022-07-18:
> 

Get. Many thanks!

closed 2022-09-02 by github-actions[bot]

DezeZhao, 2023-03-15:
> sry for replying late. the pretrain model can obttain is available [here](https://github.com/open-mmlab/OpenPCDet/issues/750#issue-1101618318)

hello, can you share your pretrained model ? I dont't find it at the issue.

DezeZhao, 2023-03-15:
> hello, can you share your pretrained model ? I dont't find it at the issue.

zhaodeze@163.com, thank u.

HezyC, 2023-09-07:
> 嗨，你能分享一下你的预训练模型吗？谢谢 [copainsir@foxmail.com](mailto:copainsir@foxmail.com)

Hello, can you share the pretrained model you obtained? Thanks 1977055971@qq.com

lifan67, 2024-01-19:
> @OuyangJunyuan , 
May I ask how do you achieve it? I tried to achieve it, but the loss has been kept around 8, and all aps are 0.

lifan67, 2024-01-19:
> @OuyangJunyuan ,
May I ask how do you achieve it? I tried to achieve it, but the loss has been kept around 8, and all AP are 0.