open-mmlab/OpenPCDet #755: ops.iou3d_nms import iou3d_nms_utils

makeradventures, 2022-01-14:
Hello, 

I am stuck with the following error and don't know what to do:
_command:_
(AI) uses@user:~/practical_project/OpenPCDet/tools$ python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path ${POINT_CLOUD_DATA}
_error:_
Traceback (most recent call last):
  File "/home/bernhard/practical_project/OpenPCDet/tools/demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/bernhard/practical_project/OpenPCDet/pcdet/datasets/__init__.py", line 7, in <module>
    from .dataset import DatasetTemplate
  File "/home/bernhard/practical_project/OpenPCDet/pcdet/datasets/dataset.py", line 8, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/bernhard/practical_project/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 6, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/bernhard/practical_project/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 9, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/bernhard/practical_project/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /home/bernhard/practical_project/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZNSt15__exception_ptr13exception_ptr10_M_releaseEv

I originally wanted to write an issue about another problem I was facing, where I was stuck in a loop where gcc <10 was a requirement but downgrading it broke my nvcc. And doing sudo apt install 'nvidia-cuda-toolkit' upgraded my gcc version to 11 again! For this issue **(which did not occur this time)** I started on a fresh Ubuntu installation and have documented every step I have done so far! Maybe this can help solving this issue I am now facing:

### My system stats:
[image]

### My original nvidia--smi stats:
[image]

### My full .bash_history leading to the **first occurrence** of this error:

```
nano "documentation_of_issue.txt"
nvidia-smi
nvcc --version
gcc --version
[200~sudo apt install nvidia-cuda-toolkit
sudo apt install nvidia-cuda-toolkit
nvcc --version
gcc --version
bash ~/Downloads/Anaconda3-2021.11-Linux-x86_64.sh
conda create --name AI
conda activate AI
conda install pip
pip install spconv-cu114
pip install torch==1.10.1+cu114 torchvision==0.11.2+cu114 torchaudio==0.10.1+cu114 -f https://download.pytorch.org/whl/cu113/torch_stable.html
conda install pytorch torchvision torchaudio cudatoolkit=11.4 -c pytorch
conda install pytorch torchvision torchaudio cudatoolkit=11.3 -c pytorch
[200~pip install tensorflow
pip install tensorflow
git clone https://github.com/open-mmlab/OpenPCDet.git
sudo apt install git
git clone https://github.com/open-mmlab/OpenPCDet.git
cd OpenPCDet/
conda activate AI
python setup.py develop
sudo apt install -y g++-10
gcc --version
python setup.py develop
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64
export PATH=$PATH:/usr/local/cuda/bin:/usr/local/bin
export CUDA_HOME=/usr/local/cuda
export C_INCLUDE_PATH=${CUDA_HOME}/include:${C_INCLUDE_PATH}
export CPATH=$CPATH:/usr/local/cuda/include
export LIBRARY_PATH=${CUDA_HOME}/lib64:$LIBRARY_PATH
python setup.py develop
python
conda activate AI
python
conda activate AI
cd OpenPCDet/
python setup.py develop
sudo apt-get install build-essential
sudo apt-get install build-essential gcc=10
gcc --version
sudo apt install gcc-10
python setup.py develop
g++ --version

python setup.py develop
which g++
sudo apt-get install build-essential
python setup.py develop
python
pip install open3d
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path ${POINT_CLOUD_DATA}
ls
cd tools
ls
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path ${POINT_CLOUD_DATA}
```

### Fishy things I have done, so far in hopes of fixing the orginal problem, with the infinite gcc requirments loop:

since `sudo apt-get install build-essential` showed that it would install g++-11 and gcc-11 which are incompatible with OenPCDet. I manually executed `sudo apt install -y g++-10` and `sudo apt install gcc-10` before installing the build-essentials! Resulting in the following apt information:
[image]

I however did NOT type `sudo-apt autoremove` yet and pytorch also still sees my gpu, inidicating that cuda is still working!

I would really appreciate your help solving the 'ops.iou3d_nms import iou3d_nms_utils' error!
Thanks in advance!

jihanyang, 2022-01-15:
Hello, can you install gcc with conda to get a compatible version with pcdet?

makeradventures, 2022-01-24:
Hello, thank you very much for the answer! I was not aware that one can install gcc with conda! Now that issue is clear! However I am still stuck on the 'ops.iou3d_nms import iou3d_nms_utils' error!

makeradventures, 2022-02-15:
I solved the Problem! The issue was the line 
`pip install spconv-cu114`
my `nvcc --version` showed that my cuda toolkit was version 11.3!
Also I installed pytorch with:
```pip3 install --pre torch torchvision torchaudio -f https://download.pytorch.org/whl/nightly/cu113/torch_nightly.html``` this time.

Here are all my bash commands for the now better version in case it helps somebody:
```
conda create --name right_sp
conda activate right_sp
conda install pip
nvcc --version
pip install spconv-cu113
pip3 install --pre torch torchvision torchaudio -f https://download.pytorch.org/whl/nightly/cu113/torch_nightly.html
cd OpenPCDet/
cd practical_project/OpenPCDet/
python setup.py develop
sudo apt install -y g++-10
conda install -y g++-10
sudo apt install -y g++-10
gcc --version
sudo apt install gcc-10
gcc --version
python setup.py develop
gcc --version
sudo apt install gcc-10
python setup.py develop
cd ..
mkdir experiment
cd experiment/
git clone https://github.com/open-mmlab/OpenPCDet.git~
git clone https://github.com/open-mmlab/OpenPCDet.git
cd OpenPCDet/
python setup.py develop
sudo apt-get install build-essential
gcc --version
pip list
pip install tensorflow
which g++
g++ --version
python setup.py develop
pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html
python setup.py develop
pip install ~/Downloads/packaging-21.3-py3-none-any.whl
pip install imageio>=2.4.1
pip install networkx>=2.2
pip install PyWavelets>=1.1.1
pip install networkx>=2.2
pip install PyWavelets>=1.1.1
pip install ~/Downloads/packaging-21.3-py3-none-any.whl
pip install ~/Downloads/packaging-21.3-py3-none-any.whl --force-reinstall
pip install open3d
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path ${POINT_CLOUD_DATA}
nvcc --version showed
```
As I side note I want to mention that I got this error during the installation:
[image]

which I solved by manually downloading the package and installing it with:
```pip install ~/Downloads/packaging-21.3-py3-none-any.whl`
and then I had to run:
`pip install imageio>=2.4.1
pip install networkx>=2.2
pip install PyWavelets>=1.1.1
pip install networkx>=2.2
pip install PyWavelets>=1.1.1
pip install ~/Downloads/packaging-21.3-py3-none-any.whl```

makeradventures, 2022-02-15:
Also in hope that it might help somone: While trying to solve this I looked through all issues. Here is a list of somehow related issues in no particular order. Some are more related, some are less related:
#403  
#705 
#245 
#254 
#273 
#468 
#569 
#624 
#650 
#672 
#776 
#268 
#420 
#774 
#655 
#553

makeradventures, 2022-02-16:
restarted my computer and the error is here again :( this makes no senes!

makeradventures, 2022-02-17:
I found a workaround using Docker #811 !

MartinHahner, 2022-02-17:
I assumed the new GPU generation was the issue but wasn't sure. I am glad this adjusted docker container works for you!
But maybe not everyone can use docker on their work/university machine due to missing admin rights.
Would be great if someone finds another solution without the help of docker, too.

SaMeEr9597, 2022-03-10:
> _**Edit:**_ I did _**not solve the Problem**_. I got a new error python: `can't open file '/home/bernhard/practical_project/experiment/OpenPCDet/demo.py': [Errno 2] No such file or directory`and was happy that the old error disapeared and went to bed, but apperently I was just tied and tried running the demo from the wrong directory. So no nothing is solved:
> 
> _**original post from me:**_ I solved the Problem! The issue was the line `pip install spconv-cu114` my `nvcc --version` showed that my cuda toolkit was version 11.3! Also I installed pytorch with: `pip3 install --pre torch torchvision torchaudio -f https://download.pytorch.org/whl/nightly/cu113/torch_nightly.html` this time.
> 
> Here are all my bash commands for the now working version in case it helps somebody:
> 
> ```
> conda create --name right_sp
> conda activate right_sp
> conda install pip
> nvcc --version
> pip install spconv-cu113
> cd OpenPCDet/
> cd practical_project/OpenPCDet/
> python setup.py develop
> sudo apt install -y g++-10
> conda install -y g++-10
> sudo apt install -y g++-10
> gcc --version
> sudo apt install gcc-10
> gcc --version
> python setup.py develop
> gcc --version
> sudo apt install gcc-10
> python setup.py develop
> cd ..
> mkdir experiment
> cd experiment/
> git clone https://github.com/open-mmlab/OpenPCDet.git~
> git clone https://github.com/open-mmlab/OpenPCDet.git
> cd OpenPCDet/
> python setup.py develop
> sudo apt-get install build-essential
> gcc --version
> pip list
> pip install tensorflow
> which g++
> g++ --version
> python setup.py develop
> pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html
> python setup.py develop
> pip install ~/Downloads/packaging-21.3-py3-none-any.whl
> pip install imageio>=2.4.1
> pip install networkx>=2.2
> pip install PyWavelets>=1.1.1
> pip install networkx>=2.2
> pip install PyWavelets>=1.1.1
> pip install ~/Downloads/packaging-21.3-py3-none-any.whl
> pip install ~/Downloads/packaging-21.3-py3-none-any.whl --force-reinstall
> pip install open3d
> python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path ${POINT_CLOUD_DATA}
> nvcc --version showed
> ```
> 
> As I side note I want to mention that I got this error during the installation: [image]
> 
> which I solved by manually downloading the package and installing it with: `pip install ~/Downloads/packaging-21.3-py3-none-any.whl` and then I had to run:
> 
> ```
> pip install networkx>=2.2
> pip install PyWavelets>=1.1.1
> pip install networkx>=2.2
> pip install PyWavelets>=1.1.1
> pip install ~/Downloads/packaging-21.3-py3-none-any.whl```
> ```

Thank you for providing this information, 
I was facing the same issue as I have a RTX A6000 card, I used your installation and the error is gone but I'm getting a new error :
[image]
can you let me know what mistakes am I doing,
Thanks

SaMeEr9597, 2022-03-12:
> sudo apt install gcc-10

What version of gcc did you use ?

makeradventures, 2022-03-16:
> _**Edit: to make it clear that I have not solved this issue yet I edited this comment (March 16th)**_ I thought that maybe `pip install spconv-cu114` was the problem my `nvcc --version` showed that my cuda toolkit was version 11.3! Also I installed pytorch with: `pip3 install --pre torch torchvision torchaudio -f https://download.pytorch.org/whl/nightly/cu113/torch_nightly.html` this time.
> 
> Here are all my bash commands for the now working version in case it helps .
.
.

> pip install networkx>=2.2
> pip install PyWavelets>=1.1.1
> pip install networkx>=2.2
> pip install PyWavelets>=1.1.1
> pip install ~/Downloads/packaging-21.3-py3-none-any.whl
> ```
> 
> But still I get the following error:
> 
> [image]

@SaMeEr9597  I think I used gcc-10. But I still haven't solve the issue! I edited the post above and added a new image with my error message to make this more clear. Sorry for being unclear and giving you hope, but unfortunately I haven't solved it myself. However I can not recommend the docker solution I mentioned above enough!
Docker provides a way cleaner and more reproducible setup and it is way easier to try out multiple versions of a package in docker than it is when you always have to uninstall the old one / install the new one / change the symbolic links etc. The Dockerfile and Docker-compose file in #811 worked for me on my RTX3080. You could try it out or read the docker file to see which combination of package versions solved the issue for me. Because at the end of the day I think, I could get it to run on my system if I had Ubuntu 18.04 and if I would install exactly the same versions as in the dockerfile!

closed 2022-04-30 by github-actions[bot]