open-mmlab/OpenPCDet #757: the result of running test.py

qianjinhao, 2022-01-16:
Could you please explain what the output of running test.py means?

2022-01-16 16:20:02,251   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:96.2683, 89.4933, 89.2394
bev  AP:90.0946, 87.8890, 87.4085
3d   AP:89.3376, 83.6905, 78.7188
aos  AP:96.24, 89.38, 89.06
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:98.2582, 94.4205, 92.2754
bev  AP:93.0256, 90.3162, 88.5318
3d   AP:92.1085, 84.3891, 82.5032
aos  AP:98.24, 94.26, 92.07
Car AP@0.70, 0.50, 0.50:
bbox AP:96.2683, 89.4933, 89.2394
bev  AP:96.3029, 89.4889, 89.2866
3d   AP:96.2557, 89.4697, 89.2496
aos  AP:96.24, 89.38, 89.06
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:98.2582, 94.4205, 92.2754
bev  AP:98.2532, 94.5873, 94.4329
3d   AP:98.2347, 94.5250, 94.3256
aos  AP:98.24, 94.26, 92.07
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:73.1607, 68.0815, 64.3235
bev  AP:65.1551, 59.4448, 54.5137
3d   AP:63.1019, 54.8400, 51.7715
aos  AP:67.76, 62.48, 58.71
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:73.6766, 68.2827, 64.3525
bev  AP:65.9296, 58.5252, 54.1260
3d   AP:62.6968, 54.5130, 49.8612
aos  AP:67.82, 62.16, 58.06
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:96.1222, 81.3675, 76.5044
bev  AP:88.5292, 73.3466, 70.3849
3d   AP:86.0637, 69.4916, 64.5456
aos  AP:95.99, 81.08, 76.18
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:97.1514, 82.4188, 78.2345
bev  AP:93.4754, 74.5449, 70.0973
3d   AP:89.1011, 70.4043, 66.0205
aos  AP:97.04, 82.12, 77.89
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:96.1222, 81.3675, 76.5044
bev  AP:95.0655, 78.2808, 73.3191
3d   AP:95.0655, 78.2717, 73.3051
aos  AP:95.99, 81.08, 76.18
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:97.1514, 82.4188, 78.2345
bev  AP:96.2319, 79.1350, 75.8273
3d   AP:96.2319, 79.1293, 75.8013
aos  AP:97.04, 82.12, 77.89

closed 2022-03-07 by github-actions[bot]

udacityyy, 2022-05-23:
Hello, did u manage to understand what it means ? if yes, do you mind explaining it to me please ?

Petros626, 2024-10-24:
@udacityyy take a look at papers about kitti evaluation and the official kitti website.