open-mmlab/OpenPCDet #758: Validating Performace of Pointpillar on Waymo

aalmuzairee, 2022-01-17:
Hi @sshaoshuai,

I'm trying to reproduce the results of  the Pointpillar model on Waymo with the default configuration that you have, and I'm getting a Level 1 mAP/mAPH of 65.14/64.42 on vehicles, while you're reported results claim that the model is able to achieve a Level 1 mAP/mAPH of 70.43/69.83 on vehicles. Is the difference in results I'm having normal, or is it something wrong with my setup?

jihanyang, 2022-01-17:
That's not normal. Can you reproduce other model's performance on waymo?

hz3014, 2022-01-17:
> Hi @sshaoshuai,
>
> I'm trying to reproduce the results of the Pointpillar model on Waymo with the default configuration that you have, and I'm getting a Level 1 mAP/mAPH of 65.14/64.42 on vehicles, [...]

how many epoch did you train? my experiment shows it doesnt converge within 30 epoch(2 batch). Probably train with more epoch

aalmuzairee, 2022-01-17:
@jihanyang : haven't tried other models yet, wanted to start with Pointpillar.

@hz3014 : That may probably be the reason, I have only trained for 30 epochs until now, how many more epochs did you train for?

hz3014, 2022-01-17:
@aalmuzairee 
I trained for 60 epoch with batch size 2 and the result was very good(better than claimed performance). I dont think batch size have much impact to performance. 
I am actually interested in what cause the difference between opencpdet claimed performance and actual performance.
@jihanyang @sshaoshuai  Kindly ask if the claimed performance@(train with 20 % data) achieved by 30 epoch with batch size 4?

jihanyang, 2022-01-17:
Yes, the performance is achievable. I have also reproduce them several times. Have you re-processed waymo data from raw after the released of pcdet v0.5.0?

hz3014, 2022-01-17:
> Yes, the performance is achievable. I have also reproduce them several times. Have you re-processed waymo data from raw after the released of pcdet v0.5.0?

Yes, I did regeneraged waymo data from raw. I can also achieve high accuracy but it just takes more epoch. I hope batch size wouldnt have this much of impact? 
Below is my training result for waymo centerpoint 31 epoch and 60 epoch with batch size 2, for reference

`centerpoint 20% 31epoch
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.6207 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.6144 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.5272 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.5217 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.6865 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6144 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6194 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5530 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7042 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6896 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6750 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6610 

60 epoch
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7095 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7037 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6120 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6068 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7562 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6925 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6850 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.6259 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7508 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.7366 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.7210 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.7073 `

jihanyang, 2022-01-17:
As we know, the batch_size_per_gpu = 2 would produce almost same or slightly better performance compared to  batch_size_per_gpu = 4 since more updating the optimizer does. However, it's still strange for me why you cannot reproduce the released result with given cfg with the latest processed waymo data. An example validation result of Centerpoint (resnet) trained by meself recently is given as below

```
2022-01-05 15:09:29,062   INFO  ==> Checkpoint trained from version: pcdet+0.5.0+d3d6893+py9eae8e7
2022-01-05 15:09:29,823   INFO  ==> Done (loaded 279/279)
2022-01-05 15:09:30,076   INFO  *************** EPOCH 30 EVALUATION *****************
2022-01-05 15:13:23,977   INFO  *************** Performance of EPOCH 30 *****************
2022-01-05 15:13:23,983   INFO  Generate label finished(sec_per_example: 0.0291 second).
2022-01-05 15:13:23,983   INFO  recall_roi_0.3: 0.000000
2022-01-05 15:13:23,983   INFO  recall_rcnn_0.3: 0.865053
2022-01-05 15:13:23,984   INFO  recall_roi_0.5: 0.000000
2022-01-05 15:13:23,984   INFO  recall_rcnn_0.5: 0.799069
2022-01-05 15:13:23,984   INFO  recall_roi_0.7: 0.000000
2022-01-05 15:13:23,984   INFO  recall_rcnn_0.7: 0.528865
2022-01-05 15:13:23,990   INFO  Average predicted number of objects(7998 samples): 208.681
2022-01-05 15:27:26,607   INFO
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7340
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7287
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6527
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6478
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7460
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6841
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6666
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.6098
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7222
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.7100
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6984
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6866

mAP
LEVEL_2/AP mAP: 0.6726
LEVEL_2/APH mAP: 0.6480

```

hz3014, 2022-01-17:
@jihanyang Which spconv version do you use? I am using spconv 2.1.21.

jihanyang, 2022-01-17:
1.2.1

closed 2022-01-21 by jihanyang

mc171819, 2022-04-19:
> @jihanyang Which spconv version do you use? I am using spconv 2.1.21.

hi, did you solve this by changing the spconv version?

hz3014, 2022-04-19:
> > @jihanyang Which spconv version do you use? I am using spconv 2.1.21.
> 
> hi, did you solve this by changing the spconv version?

Hey, I did not treat this as an issue as I achieved higher accuracy for some other models. However, I believe if you switch back spconv version you should get similar performance.