open-mmlab/OpenPCDet #762: 'NoneType' object has no attribute 'contiguous' when dataset has only lidar information

iskra3138, 2022-01-18:
Hi there,

When I train pv_rcnn with my own dataset, I got an the following error message in v0.5.2.

  File "train.py", line 203, in <module>
    main()
  File "train.py", line 172, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/OpenPCDet/tools/train_utils/train_utils.py", line 118, in train_model
    dataloader_iter=dataloader_iter
  File "/OpenPCDet/tools/train_utils/train_utils.py", line 47, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "../pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/root/anaconda3/envs/tree/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/root/anaconda3/envs/tree/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 379, in forward
    rois=batch_dict.get('rois', None)
  File "../pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 330, in aggregate_keypoint_features_from_one_source
    features= xyz_features.contiguous() #if xyz_features is not None else None,
AttributeError: 'NoneType' object has no attribute 'contiguous'

I think it happened because NUM_POINT_FEATURES of my dataset is 3.
So I modified line 330 like this.
features= xyz_features.contiguous() if xyz_features is not None else None,

Am I right?
Thanks in advance.

https://github.com/open-mmlab/OpenPCDet/blob/5861246c31c876f0e5a3e125086ed2b0be7bc9da/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py#L330

iskra3138, 2022-01-19:
I found that it was an error caused by not removing 'raw_points' from FEATURES_SOURCE in pv_rcnn.yaml.
I close this issue.

closed 2022-01-19 by iskra3138

im-djh, 2022-08-23:
[image]
    I think you are doing the right thing. Actually, there might be a little bug here. 
    When calling function: pooled_points, pooled_features = aggregate_func(....), the parameter "features=xyz_features" might be "None" if we set used_feature_list: ['x', 'y', 'z'] for xyz_features are those features excluding x,y,z. By just adding a conditional branch to use .continguous() function can easily aviod that bug.