open-mmlab/OpenPCDet #766: Waymo Info Generation Expected Output

OrcunCanDeniz, 2022-01-20:
I ran the waymo generation and got ;
[image]
I found the skipped infos numbers suspicious. I can see tqdm progress as 
`Infos are saved to (sampled_interval=1): /home/ubuntu/OpenPCDet/data/waymo/waymo_processed_data_v0_5_0/segment-1022527355599519580_4866_960_4886_960/segment-1022527355599519580_4866_960_4886_960.pkl
  2%|███▋                                                                                                                                                                                                 | 15/798 [19:44<8:15:23, 37.96s/it]`

Since there was no info on expected outputs of data generation scripts, I couldnt be sure about this behaviour. Is this normal?

jihanyang, 2022-01-21:
No. You don't sample any data here.

OrcunCanDeniz, 2022-01-21:
What would be the problem? I have exactly 1000 .tfrecords under `data/waymo/raw_data`. As the script continues, directories are being created into `data/waymo/waymo_processed_data_v0_5_0/segment-xxxxxx` with bunch of `.npy` files and a .pkl file in them. By the way, raw_data was soft linked to some other location but still satisfies the convention to access data from data/waymo/raw_data/XXXX.tfrecord.

jihanyang, 2022-01-21:
You are supposed to step in to the `include_waymo_data` in waymo_dataset.py. I cannot dertermine the exact problem, since the information is not enough for me.

sshaoshuai, 2022-01-21:
I think this log may be normal, since the first initialization of WaymoDataset will also try to load infos - which you still do not create yet. 

You just need to check the infos after the process of data generation is complete.

OrcunCanDeniz, 2022-01-21:
Script ended with 
`gt_database sample: 157963/157965`
`gt_database sample: 157964/157965`
`gt_database sample: 157965/157965`
`Database Vehicle: 1194258`
`Database Pedestrian: 1110186`
`Database Cyclist: 53270`
`---------------Data preparation Done---------------`

sshaoshuai, 2022-01-21:
Yes, if you want to exactly match our results,  you need to use v1.2.0 of WOD. But v1.0 could also achieve good results (with a small gap with v1.2.0).

closed 2022-01-21 by OrcunCanDeniz