open-mmlab/OpenPCDet #769: Run python setup.py develop wrong

sumanghhx, 2022-01-21:
hello, I met this in
$ python setup.py develop
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
1.10.2.git.kitware.jobserver-1
g++ -pthread -shared -B /home/nebula/anaconda3/envs/OpenPcDet/compiler_compat -L/home/nebula/anaconda3/envs/OpenPcDet/lib -Wl,-rpath=/home/nebula/anaconda3/envs/OpenPcDet/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_cpu.o /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -L/home/nebula/anaconda3/envs/OpenPcDet/lib/python3.6/site-packages/torch/lib -L/usr/local/cuda-11.0/bin/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda -o build/lib.linux-x86_64-3.6/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-36m-x86_64-linux-gnu.so
g++: error: /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_cpu.o: No such file or directory
g++: error: /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o: No such file or directory
g++: error: /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms.o: No such file or directory
g++: error: /home/nebula/hhx/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o: No such file or directory
error: command 'g++' failed with exit status 1

python 3.6.13
cuda 11.0
gcc 9.3.0

djiajunustc, 2022-01-22:
Hi @sumanghhx ,

Please refer to issue #240.

sumanghhx, 2022-01-24:
@djiajunustc Hello, I read issue #240.，but it didn't work，please help

sumanghhx, 2022-01-24:
thank you,I already solve it, my environmental problems

closed 2022-01-24 by sumanghhx

nicholaswhf, 2022-01-28:
> thank you,I already solve it, my environmental problems

how did you solve it， bro