open-mmlab/OpenPCDet #771: Training using our own dataset

SaMeEr9597, 2022-01-23:
Good evening,
First I would like to thank you for this project, I had a query regarding training using our own dataset. I see that in Demo you have shown how to use our own dataset of point cloud and save to numpy format and test it on a pre-trained model, is there a way we can train using our own dataset, if so could you please guide me as to how to do it, 
Thanks for your help

155cannon, 2022-01-27:
我来分享一下如何训练自己的数据集以及如何标注点云获得3D标注框：

1.标注点云获得3D标注框
我使用cloudcompare软件，本来有一些.pcd文件，用cloudcompare打开后，选择cross section功能，把你要检测的目标框出来，然后形成新的点云，选择后在属性面板可以看到目标中心坐标和3D标注框的长宽高，将这6个值存在标注文件里（可以直接ctrl+c复制属性面板上的文字，然后编写简单的预处理程序完善一下，形成下面说的标注文件）。
使用cloudcompare另存.pcd文件，去除掉其他信息，只保留xyz坐标，另存为.txt文件。
最终形成：
点云文件0000.txt（每行x y z）~xxxx.txt
标注文件0000.txt(每行x, y, z, dx, dy, dz, heading）~xxxx.txt，x, y, z, dx, dy, dz就是我上面说的那6个值，按顺序放，heading我统一设为1.57，也就是90°，这个根据需要，我的情况是这样。
这样直接就得到了OpenPCDet统一坐标系下的点云和3D标注框。

2.训练自己的数据集
我之前看了好多人发的帖子，有的人模仿kitti数据集的形式制作自己的数据集，然后使用原程序，有的人写自己的dataset.py，经过考虑我选择了后者。我比较仔细地看了demo.py、kitti_dataset.py和dataset.py（网络上也有其他人写的OpenPCDet的注释，可以看一下），搞明白一些东西后，就开始模仿着写自己的dataset.py。主要一些代码见下：

import glob
import numpy as np
from pathlib import Path
from ..dataset import DatasetTemplate

class CoilsDataset(DatasetTemplate):
    def __init__(self, dataset_cfg, class_names, training=True, root_path=None, logger=None):
        """
        Args:
            root_path:
            dataset_cfg:
            class_names:
            training:
            logger:
        """
        super().__init__(
            dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
        )

        points_file_list = glob.glob(str(self.root_path / 'training/points' / '*.txt'))
        labels_file_list = glob.glob(str(self.root_path / 'training/labels' / '*.txt'))

        points_file_list.sort()
        labels_file_list.sort()
        self.sample_file_list = points_file_list
        self.samplelabel_file_list = labels_file_list

    def __len__(self):
        return len(self.sample_file_list)

    def __getitem__(self, index):
        sample_idx = Path(self.sample_file_list[index]).stem  # 0000.txt -> 0000 样本id(文件编号) n：点的个数 m：标注的个数
        points = np.loadtxt(self.sample_file_list[index], dtype=np.float32).reshape(-1, 3)  # 每个点云文件里的所有点 n*3
        points_label = np.loadtxt(self.samplelabel_file_list[index], dtype=np.float32).reshape(-1, 7) # 每个点云标注文件里的所有点 m*7
        gt_names = np.array(['Coil']*points_label.shape[0])

        input_dict = {
            'points': points,
            'frame_id': sample_idx,
            'gt_names': gt_names,
            'gt_boxes': points_label
        }

        data_dict = self.prepare_data(data_dict=input_dict)
        return data_dict
以上是我的coils_dataset.py，此文件和kitti_dataset.py放在同一目录下。

DATASET: 'CoilsDataset'
DATA_PATH: '../data/coils'

POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]

DATA_SPLIT: {
    'train': train,
    'test': val
}

[... 281 lines cut ...]
    model.cuda()
    model.eval()
    with torch.no_grad():
        for idx, data_dict in enumerate(demo_dataset):
            logger.info(f'Visualized sample index: \t{idx + 1}')
            data_dict = demo_dataset.collate_batch([data_dict])
            load_data_to_gpu(data_dict)
            pred_dicts, _ = model.forward(data_dict)

            V.draw_scenes(
                points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
                ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
            )

            if not OPEN3D_FLAG:
                mlab.show(stop=True)

    logger.info('Demo done.')

if __name__ == '__main__':
    main()

然后就可以看效果了！

最后，感谢OpenPCDet团队成员的辛勤付出，向你们致敬！

jihanyang, 2022-01-27:
@155cannon Many thanks for your kindly sharing! Hope you can also post this comments in this issue #253, which collect most discussion about "custom datasets". Also, it will be more helpful if you can post this with English, since it is more widely-used in our community.

VsionQing, 2022-02-23:
Hi，We encountered a problem when training Kitti dataset. No error was reported, but we can't train. #820

yun9993, 2022-02-23:
你好 @155cannon ，请问方便分享你更改后的源文件嘛？训练自己的数据集搞得我头昏脑胀，2448115060@qq.com，这是我的邮箱，感谢！

VsionQing, 2022-02-23:
> 你好 @155cannon ，请问方便分享你更改后的源文件嘛？训练自己的数据集搞得我头昏脑胀，[2448115060@qq.com](mailto:2448115060@qq.com)，这是我的邮箱，感谢！

请问你是在windows下训练kitti的吗？

yun9993, 2022-02-23:
> > 你好@155cannon [，](mailto:2448115060@qq.com)请问源昏方便分享你之后更改的文件吗？
> 
> 请问你是在windows下训练kitti的吗？

我用PCTA做了一个小规模的数据集。想在ubuntu下用自己的数据集训练模型试试水，但是还不太清楚数据集该怎么处理。

soo4826, 2022-02-23:
Hi, I also train with custom dataset.
So I make my own dataloader according to this issue.
I implemented __getitem__ and __len__ function but if i started training, NotImplemented error occurs.

Should i have to edit more code in this issue from @155cannon

jeacwen, 2022-02-24:
> 我来分享一下如何自己的数据集以及如何发现点云获取3D标注框：
> 
> 1.标注点云获取3D标注框 我使用云比较软件，本来就有一些.pcd文件，用云比较打开后，选择截面功能，把你要检测的目标框出来，然后形成新的点云面板可以看到目标坐标和 3D 框的长宽高，将这6个值存在于中心的文字文件里（可以ctrl+c属性复制面板上的，然后直接将一个简单的执行程序完善，形成下面说的标记文件）。 使用cloud比较另存.pcd文件，去掉其他信息，只保留文件xyz坐标，另存为.txt文件。 最终形成： 云文件0000.txt（每行xyz）~xxxx .标记 txt 0000.txt(每行x,y,z,dx,dy,dz,heading)~xxxx.txt，x,y,z,dx,dy,dz就是我上面说的那6个值，按顺序放，标题我统一设为1.57，也就是90°，根据，我的情况是这样 的。
> 
> 2.自己训练 很多人发的帖子，然后有的人根据我的数据集来制作自己的数据集，使用原程序，有的人写自己的数据集。我比较仔细地看看demo.py、kitti_dataset.py和dataset.py（网络上也有OpenPCDet的注释，可以看看），搞定一些东西后，就开始用其他方式写自己的dataset.py。主要一些代码见下：
> 
> import glob import numpy as np from pathlib import Path from ..dataset import DatasetTemplate
> 
> 类 CoilsDataset(DatasetTemplate): def **init** (self, dataset_cfg, class_names, training=True, root_path=None, logger=None): """ Args: root_path: dataset_cfg: class_names: training: logger: """ super()。**初始化**（ dataset_cfg=dataset_cfg，class_names=class_names，training=training，root_path=root_path，logger=logger ）
> 
> ```
>     points_file_list = glob.glob(str(self.root_path / 'training/points' / '*.txt'))
>     labels_file_list = glob.glob(str(self.root_path / 'training/labels' / '*.txt'))
> 
>     points_file_list.sort()
>     labels_file_list.sort()
>     self.sample_file_list = points_file_list
>     self.samplelabel_file_list = labels_file_list
> 
> def __len__(self):
>     return len(self.sample_file_list)
> 
> def __getitem__(self, index):
>     sample_idx = Path(self.sample_file_list[index]).stem  # 0000.txt -> 0000 样本id(文件编号) n：点的个数 m：标注的个数
>     points = np.loadtxt(self.sample_file_list[index], dtype=np.float32).reshape(-1, 3)  # 每个点云文件里的所有点 n*3
>     points_label = np.loadtxt(self.samplelabel_file_list[index], dtype=np.float32).reshape(-1, 7) # 每个点云标注文件里的所有点 m*7
>     gt_names = np.array(['Coil']*points_label.shape[0])
> 
>     input_dict = {
>         'points': points,
>         'frame_id': sample_idx,
>         'gt_names': gt_names,
>         'gt_boxes': points_label
>     }
> 
>     data_dict = self.prepare_data(data_dict=input_dict)
>     return data_dict
> ```
> 
> 以上是我的coils_dataset.py，这个文件和kitti_dataset.py 定位在同一个目录下。
> 
> 数据集：'CoilsDataset' DATA_PATH：'../data/coils'
> 
> POINT_CLOUD_RANGE：[0, -40, -3, 70.4, 40, 1]
> 
> DATA_SPLIT：{ “火车”：火车， “测试”：val }
> 
> INFO_PATH：{ “火车”：[kitti_infos_train.pkl]， “测试”：[kitti_infos_val.pkl]， }
> 
> GET_ITEM_LIST：[“点”] FOV_POINTS_ONLY：真
> 
> DATA_AUGMENTOR： DISABLE_AUG_LIST：['placeholder'] AUG_CONFIG_LIST： -名称：random_world_flip ALONG_AXIS_LIST：['x']
> 
> ```
>     - NAME: random_world_rotation
>       WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]
> 
>     - NAME: random_world_scaling
>       WORLD_SCALE_RANGE: [0.95, 1.05]
> ```
> 
> POINT_FEATURE_ENCODING：{ encoding_type：absolute_coordinates_encoding， used_feature_list：['x'，'y'，'z']， src_feature_list：['x'，'y'，'z']， }
> 
> DATA_PROCESSOR： -名称：mask_points_and_boxes_outside_range REMOVE_OUTSIDE_BOXES：真
> 
> ```
> - NAME: shuffle_points
>   SHUFFLE_ENABLED: {
>     'train': True,
>     'test': False
[... 201 lines cut ...]
>         load_data_to_gpu(data_dict)
>         pred_dicts, _ = model.forward(data_dict)
> 
>         V.draw_scenes(
>             points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
>             ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
>         )
> 
>         if not OPEN3D_FLAG:
>             mlab.show(stop=True)
> 
> logger.info('Demo done.')
> ```
> 
> 如果**名称**== '**主要**': main()
> 
> 然后就可以看到效果了！
> 
> ，感谢OpenPC团队成员的辛勤工作，最后向你们欢迎！

> 我来分享一下如何训练自己的数据集以及如何标注点云获得3D标注框：
>
> 1.标注点云获得3D标注框 我使用cloudcompare软件，本来有一些.pcd文件，用cloudcompare打开后，选择cross [...]

请问可以分享源码吗？感激不敬，付费都行。jeacwen@163.com

jeacwen, 2022-02-24:
> 我来分享一下如何训练自己的数据集以及如何标注点云获得3D标注框：
>
> 1.标注点云获得3D标注框 我使用cloudcompare软件，本来有一些.pcd文件，用cloudcompare打开后，选择cross [...]

请问可以分享源码用于学习吗？刚入门，如果能得到指点就万分感谢。这是邮箱jeacwen@163.com，希望得到您的回复

jeacwen, 2022-02-25:
hi 请问方便分享一下修改的完成源码吗？这个邮箱是我的邮箱[jeacwen@163.com]。

zjr-bit, 2022-03-19:
Hello, bro! I am trying to train the PointPillars with the dataset(DAIR_V2X). And I have already complete data procession. But when I train it, the error comes out：

epochs:   0%|                                                                                                                                                                                 | 0/1 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 202, in <module>
    main()
  File "train.py", line 154, in main
    train_model(
  File "/home/zjr/OpenPCDet/tools/train_utils/train_utils.py", line 111, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/zjr/OpenPCDet/tools/train_utils/train_utils.py", line 47, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/__init__.py", line 54, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/zjr/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/detectors/pointpillar.py", line 14, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/home/zjr/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zjr/OpenPCDet/tools/../pcdet/utils/loss_utils.py", line 59, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (16920) must match the size of tensor b (17280) at non-singleton dimension 1
train:   0%|                                                                          

if you could give me some clue, I would thank you very much!

gashel, 2022-04-15:
@155cannon 哈喽，您的代码可以实现训练，只是训练完成后没有性能评估，而且加上评估代码做训练会报错，请问您有做后续的修改吗？感谢~

155cannon, 2022-04-26:
> @155cannon 哈喽，您的代码可以实现训练，只是训练完成后没有性能评估，而且加上评估代码做训练会报错，请问您有做后续的修改吗？感谢~

性能评估的代码我没有看，加上评估代码做训练肯定会报错，所以去掉了性能评估。我不是搞学术科研的，单纯使用，直接看效果。

ccsself, 2022-05-18:
> 

hello  请问 您自定义这部分的数据是生存pkl格式的数据进行训练的吗   ？

VsionQing, 2022-05-18:
收到谢谢

Christina-Soda, 2022-07-04:
@155cannon 请问您可以分享一下源文件吗？学术小白跪谢！xinxinran_0288@126.com

OrangeSodahub, 2022-07-18:
@Christina-Soda @ghb0224 @ccsself @jeacwen @yun9993 @VsionQing 
Hello, you can refer to my successful example using kitti format custom dataset. README describes how to label, train, inference it including transformation of coordinates. It may solve your problems!
https://github.com/OrangeSodahub/CRLFnet#lid-cam-fusion
https://github.com/OrangeSodahub/CRLFnet/blob/master/src/site_model/src/LidCamFusion/OpenPCDet/pcdet/datasets/custom/README.md

JulyLi2019, 2022-08-10:
> 
您好，我在训练时遇到如下错误，请问您有遇到过吗？
RecursionError: maximum recursion depth exceeded in comparison

VsionQing, 2022-08-10:
收到谢谢

eucww, 2022-08-31:
@155cannon 你好，可以发一份源文件参考一下嘛，谢谢． cy_w0n@163.com

brillint, 2022-09-16:
> >大佬您好 @155cannon ，请问方便分享你更改后的源文件嘛？313089570@qq.com，这是我的邮箱，非常感谢！

closed 2022-11-01 by github-actions[bot]

xcc2731594, 2022-12-02:
> > 
> 
> 您好，我在训练时遇到如下错误，请问您有遇到过吗？ RecursionError: maximum recursion depth exceeded in comparison

您好，你的问题解决了吗，我也遇到这个问题

LiangziWang9, 2023-10-18:
大佬您好，请问可以分享一下源文件吗？这是我的邮箱769487692@qq.com,感谢！ @155cannon

daofeng2007, 2023-11-07:
I followed the instructions provided by @155cannon  and received the error below when I tried to train the model.

RuntimeError: Sizes of tensors must match except in dimension 1. Expected size 1250 but got size 1252 for tensor number 2 in the list.

point range and voxel size are set below.
POINT_CLOUD_RANGE: [-200, -200,-32,200, 200,32]
VOXEL_SIZE: [0.16, 0.16, 64]

How to remove the error?

LiangziWang9, 2023-11-08:
You need to modify the point_cloud_range, please refer to docs/CUSTOM_DATASET_TUTORIAL.md,point cloud range along x&y-axis / voxel_size is the multiple of 16. @daofeng2007

jianxinqwe, 2024-08-15:
大佬您好，请问可以分享一下源文件吗？这是我的邮箱[251409433@qq.com](感谢！ @155cannon

yjbyte, 2025-02-27:
大佬您好，我目前制作了自己的数据集，也使用cloudcompare进行了标注，在训练阶段遇到了问题，请问可以分享一下源文件吗？这是我的邮箱[202412492588@nuist.edu.cn],感谢！ @155cannon