open-mmlab/OpenPCDet #774: Run demo.py but Wrong

RonaldYuren, 2022-01-25:
I was running the demo.py with KITTI dataset but it came out with error like below:

Traceback (most recent call last):
  File "demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/gene/YuRen/OpenPCDet/pcdet/datasets/__init__.py", line 7, in <module>
    from .dataset import DatasetTemplate
  File "/home/gene/YuRen/OpenPCDet/pcdet/datasets/dataset.py", line 8, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/gene/YuRen/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 6, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/gene/YuRen/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 9, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/gene/YuRen/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /home/gene/YuRen/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN3c1011CPUTensorIdEv

I have no idea with it, please help

jihanyang, 2022-01-25:
Notice that spconv 2.x rely on pytorch >= 1.5.0.

RonaldYuren, 2022-01-25:
> Notice that spconv 2.x rely on pytorch >= 1.5.0.

I have finally upgrade pytorch to 1.5.0, but the error is still the same.
do you have some idea on the path suggested in #403 ?

jihanyang, 2022-01-27:
No. I have no idea about his solution.

RonaldYuren, 2022-02-14:
> No. I have no idea about his solution.

python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \
    --ckpt pv_rcnn_8369.pth \
    --data_path ${POINT_CLOUD_DATA}

maybe error is the  "pv_rcnn_8369.pth", is this file included in the OpenPCD folder?

safaATcurrus, 2022-02-21:
The important thing is not stated in the document
what does the model/config file paths point to by default?
if we only write pv_rcnn_9987.pth, where will we put this model file? Inside the operating systems root? What is the absolute path of that model field in the parameter?

MartinHahner, 2022-02-22:
> The important thing is not stated in the document what does the model/config file paths point to by default? if we only write pv_rcnn_9987.pth, where will we put this model file? Inside the operating systems root? What is the absolute path of that model field in the parameter?

You can either use the relative path from where `demo.py` is located or the full absolute path.
Just putting e.g. `--ckpt pv_rcnn_8369.pth` wont work.

**And** you need to download the checkpoint first from the [model zoo](https://github.com/open-mmlab/OpenPCDet#model-zoo), it is not directly included in the repository.

closed 2022-04-08 by github-actions[bot]

wzan0001, 2022-08-10:
I get the same error, do you solve it?