open-mmlab/OpenPCDet #775: run train.py error! about data_augmentor

155cannon, 2022-01-26:
I have my own dataset.
points:x y z
labels:cx, cy, cz, dx, dy, dz, heading
The points and labels are transformed to the unified normative coordinate of OpenPCDet already.I write my customized dataset
class inherited from DatasetTemplate. The __getitem__ function is shown below.

    def __getitem__(self, index):
        sample_idx = Path(self.sample_file_list[index]).stem  # 0000.txt -> 0000 样本id(文件编号) n：点的个数 m：标注的个数
        points = np.loadtxt(self.sample_file_list[index], dtype=np.float32).reshape(-1, 3)  # 每个点云文件里的所有点 n*3
        points_label = np.loadtxt(self.samplelabel_file_list[index], dtype=np.float32).reshape(-1, 7) # 每个点云标注文件里的所有点 m*7
        gt_names = np.array(['Coil'])

        input_dict = {
            'points': points,
            'frame_id': sample_idx,
            'gt_names': gt_names,
            'gt_boxes': points_label
        }

        data_dict = self.prepare_data(data_dict=input_dict)
        return data_dict

I use pointpillar model and remove the gt_sampling from the AUG_CONFIG_LIST.
When run python train.py --cfg_file cfgs/coils_models/pointpillar.yaml --pretrained_model pointpillar_7728.pth, error occurs!

2022-01-26 17:19:30,921   INFO  **********************Start training coils_models/pointpillar(default)**********************
epochs:   0%|          | 0/80 [00:00<?, ?it/s]
epochs:   0%|          | 0/80 [00:02<?, ?it/s]
Traceback (most recent call last):
  File "/home/via/PointCloud/OpenPCDet-master/tools/coilsTrain.py", line 183, in <module>
    main()
  File "/home/via/PointCloud/OpenPCDet-master/tools/coilsTrain.py", line 153, in main
    train_model(
  File "/home/via/PointCloud/OpenPCDet-master/tools/train_utils/train_utils.py", line 111, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/via/PointCloud/OpenPCDet-master/tools/train_utils/train_utils.py", line 25, in train_one_epoch
    batch = next(dataloader_iter)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/usr/local/lib/python3.8/dist-packages/torch/_utils.py", line 434, in reraise
    raise exception
IndexError: Caught IndexError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/via/PointCloud/OpenPCDet-master/tools/../pcdet/datasets/kitti/coils_dataset.py", line 47, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/via/PointCloud/OpenPCDet-master/tools/../pcdet/datasets/dataset.py", line 127, in prepare_data
    data_dict = self.data_augmentor.forward(
  File "/home/via/PointCloud/OpenPCDet-master/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 229, in forward
    data_dict['gt_boxes'] = data_dict['gt_boxes'][gt_boxes_mask]
IndexError: boolean index did not match indexed array along dimension 0; dimension is 7 but corresponding boolean dimension is 1

The dimension in "dimension is 7"  is random(7, 8 or 11). What's the problem? Thank everyone!

MartinHahner, 2022-01-26:
debug the dimensions of this line:  
File "/home/via/PointCloud/OpenPCDet-master/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 229
`data_dict['gt_boxes'] = data_dict['gt_boxes'][gt_boxes_mask]`

What happens if you start training from scratch without the pretrained snapshot _pointpillar_7728.pth_?

155cannon, 2022-01-27:
> debug the dimensions of this line: File "/home/via/PointCloud/OpenPCDet-master/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 229 `data_dict['gt_boxes'] = [...]

I try , the same error occurs

155cannon, 2022-01-27:
The pointpillar.yaml:

CLASS_NAMES: ['Coil']

DATA_CONFIG: 
    _BASE_CONFIG_: cfgs/dataset_configs/coils_dataset.yaml
    POINT_CLOUD_RANGE: [24, 0, 0, 90, 12, 4.5]
    DATA_PROCESSOR:
        - NAME: mask_points_and_boxes_outside_range
          REMOVE_OUTSIDE_BOXES: True

        - NAME: shuffle_points
          SHUFFLE_ENABLED: {
            'train': True,
            'test': False
          }

        - NAME: transform_points_to_voxels
          VOXEL_SIZE: [0.16, 0.16, 4]
          MAX_POINTS_PER_VOXEL: 32
          MAX_NUMBER_OF_VOXELS: {
            'train': 16000,
            'test': 40000
          }
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder']
        AUG_CONFIG_LIST:
            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: PointPillar

    VFE:
        NAME: PillarVFE
        WITH_DISTANCE: False
        USE_ABSLOTE_XYZ: True
        USE_NORM: True
        NUM_FILTERS: [64]

    MAP_TO_BEV:
        NAME: PointPillarScatter
        NUM_BEV_FEATURES: 64

    BACKBONE_2D:
        NAME: BaseBEVBackbone
        LAYER_NUMS: [3, 5, 5]
        LAYER_STRIDES: [2, 2, 2]
        NUM_FILTERS: [64, 128, 256]
        UPSAMPLE_STRIDES: [1, 2, 4]
        NUM_UPSAMPLE_FILTERS: [128, 128, 128]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Coil',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            }
        ]

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.01
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

155cannon, 2022-01-27:
the coils_dataset.yaml:

DATASET: 'CoilsDataset'
DATA_PATH: '../data/coils'

POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]

DATA_SPLIT: {
    'train': train,
    'test': val
}

INFO_PATH: {
    'train': [kitti_infos_train.pkl],
    'test': [kitti_infos_val.pkl],
}

GET_ITEM_LIST: ["points"]
FOV_POINTS_ONLY: True

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x']

        - NAME: random_world_rotation
          WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

        - NAME: random_world_scaling
          WORLD_SCALE_RANGE: [0.95, 1.05]

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z'],
    src_feature_list: ['x', 'y', 'z'],
}

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': False
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.05, 0.05, 0.1]
      MAX_POINTS_PER_VOXEL: 5
      MAX_NUMBER_OF_VOXELS: {
        'train': 16000,
        'test': 40000
      }

155cannon, 2022-01-27:
the coils_dataset.py:

import glob
import numpy as np
from pathlib import Path
from ..dataset import DatasetTemplate

class CoilsDataset(DatasetTemplate):
    def __init__(self, dataset_cfg, class_names, training=True, root_path=None, logger=None):
        """
        Args:
            root_path:
            dataset_cfg:
            class_names:
            training:
            logger:
        """
        super().__init__(
            dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
        )

        points_file_list = glob.glob(str(self.root_path / 'training/points' / '*.txt'))
        labels_file_list = glob.glob(str(self.root_path / 'training/labels' / '*.txt'))

        points_file_list.sort()
        labels_file_list.sort()
        self.sample_file_list = points_file_list
        self.samplelabel_file_list = labels_file_list

    def __len__(self):
        return len(self.sample_file_list)

    def __getitem__(self, index):
        sample_idx = Path(self.sample_file_list[index]).stem  # 0000.txt -> 0000 样本id(文件编号) n：点的个数 m：标注的个数
        points = np.loadtxt(self.sample_file_list[index], dtype=np.float32).reshape(-1, 3)  # 每个点云文件里的所有点 n*3
        points_label = np.loadtxt(self.samplelabel_file_list[index], dtype=np.float32).reshape(-1, 7) # 每个点云标注文件里的所有点 m*7
        gt_names = np.array(['Coil'])

        input_dict = {
            'points': points,
            'frame_id': sample_idx,
            'gt_names': gt_names,
            'gt_boxes': points_label
        }

        data_dict = self.prepare_data(data_dict=input_dict)
        return data_dict

    @staticmethod
    def generate_prediction_dicts(batch_dict, pred_dicts, class_names, output_path=None):
        """
        Args:
            batch_dict:
                frame_id:
            pred_dicts: list of pred_dicts
                pred_boxes: (N, 7), Tensor
                pred_scores: (N), Tensor
                pred_labels: (N), Tensor

            batch_dict: dict of original data from the dataloader

            class_names:
            output_path: if it is not None, save the results to this path
        """
        def get_template_prediction(num_samples):
            ret_dict = {
                'name': np.zeros(num_samples),
                'score': np.zeros(num_samples), 'boxes_lidar': np.zeros([num_samples, 7])
            }
            return ret_dict

        def generate_single_sample_dict(box_dict):
            pred_scores = box_dict['pred_scores'].cpu().numpy()
            pred_boxes = box_dict['pred_boxes'].cpu().numpy()
            pred_labels = box_dict['pred_labels'].cpu().numpy()

            pred_dict = get_template_prediction(pred_scores.shape[0])
            if pred_scores.shape[0] == 0:
                return pred_dict

            pred_dict['name'] = np.array(class_names)[pred_labels - 1]
            pred_dict['score'] = pred_scores
            pred_dict['boxes_lidar'] = pred_boxes

            return pred_dict

        annos = []
        for index, box_dict in enumerate(pred_dicts):
            frame_id = batch_dict['frame_id'][index]

            single_pred_dict = generate_single_sample_dict(box_dict)
            single_pred_dict['frame_id'] = frame_id
            annos.append(single_pred_dict)

            if output_path is not None:
                cur_det_file = output_path / ('%s.txt' % frame_id)
                with open(cur_det_file, 'w') as f:
                    name = single_pred_dict['name']
                    score = single_pred_dict['score']
                    boxes_lidar = single_pred_dict['boxes_lidar']

                    for idx in range(len(single_pred_dict['boxes_lidar'].shape[0])):
                        print('%d===%s -jingyou %.4f %.4f %.4f %.4f %.4f %.4f %.4f %.4f'
                              % (frame_id,name[idx],score[idx],boxes_lidar[idx][0],boxes_lidar[idx][1],
                                 boxes_lidar[idx][2],boxes_lidar[idx][3],boxes_lidar[idx][4],
                                 boxes_lidar[idx][5],boxes_lidar[idx][6]), file=f)

        return annos

155cannon, 2022-01-27:
It's my fault, I I have found the problem. Thank you! @MartinHahner

closed 2022-01-27 by 155cannon