open-mmlab/OpenPCDet #776: demo.py problem on latest pcdet-0.5.2

xuf2, 2022-01-27:
All install without any problem, but runing demo.py with one frame KITTI data which occurred below error:

**RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
Traceback (most recent call last):
  File "demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/u_xuf2/OpenPCDet/pcdet/datasets/__init__.py", line 5, in <module>
    from pcdet.utils import common_utils
  File "/home/u_xuf2/OpenPCDet/pcdet/utils/common_utils.py", line 7, in <module>
    import SharedArray
ImportError: numpy.core.multiarray failed to import**

Follow below environment:
Ubuntu18.04
Python 3.8
PyTorch 1.9
CUDA11.1+cuDNN8.0.5
spconv1.2
pcdet v0.5.2(latest)
open3d-0.14.1/mayavi
numpy-1.20.0

I update numpy version but other error occurred.

Have used the same above configuration environment for pcdet0.3.0 installation on another PC two months ago, the demo.py worked well.

sshaoshuai, 2022-01-29:
Please just try to re-install 'SharedArray'.

closed 2022-02-03 by sshaoshuai

xuf2, 2022-02-07:
Thanks for the reply, I try to reinstall lower version SharedArray==3.1 and solve this problem