open-mmlab/OpenPCDet #777: The performance of PV-RCNN in waymo open dataset

kyrie-H, 2022-01-27:
I have trained the model of PV-RCNN on 20% data of all the training samples. However, our evaluation results are much worse than the evaluation of the pre-trained model, is this related to the selection of training data?

Our performance:
2022-01-26 09:38:52,489   INFO  Generate label finished(sec_per_example: 0.1105 second).
2022-01-26 09:38:52,490   INFO  recall_roi_0.3: 0.750898
2022-01-26 09:38:52,490   INFO  recall_rcnn_0.3: 0.751561
2022-01-26 09:38:52,490   INFO  recall_roi_0.5: 0.687323
2022-01-26 09:38:52,490   INFO  recall_rcnn_0.5: 0.696961
2022-01-26 09:38:52,490   INFO  recall_roi_0.7: 0.433653
2022-01-26 09:38:52,490   INFO  recall_rcnn_0.7: 0.477272
2022-01-26 09:38:52,521   INFO  Average predicted number of objects(39987 samples): 47.752
2022-01-26 09:49:17,125   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7066 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.6996 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6206 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6143 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.5839 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.2952 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.5005 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.2531 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.5756 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.3629 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.5539 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.3492 

The official performance:
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7541
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7474
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6744
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6680
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7198
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6124
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6370
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5395
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.6588
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6425
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6339
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6182

thptai, 2022-01-27:
My results which uses 20% training data Top (LV1): AP/APH, Bottom (LV2):
AP/APH
PV-RCNN 20% 30 74.42 74.05 70.28 60.00 65.66 64.17
Max(30) 66.16 65.54 61.15 52.04 63.21 61.77

============================================================

>> Name : Tran Huu Phuong Tai

>> H.P. : 010-9400-1803

>> E-mail: ***@***.***

>> Automation Lab. (83354), School of Information and Communication
Engineering,

>> Sungkyunkwan University, 2066, Seobu-ro, Jangan-gu, Suwon-si,
Gyeonggi-do, Korea (440-746)

>> Tel : +82-31-290-7197

============================================================

kyrie-H, 2022-01-27:
> 我使用 20% 训练数据的结果 Top (LV1): AP/APH, Bottom (LV2): AP/APH PV-RCNN 20% 30 74.42 74.05 70.28 60.00 65.66 64.17 Max(30) 66.16 65.54 61.15 52.04 63.21 61.77 === ==================================================== =======
> > 姓名：Tran Huu Phuong Tai > HP：010-9400-1803 > 电子邮件：***@***.*** > 自动化实验室。(83354)，信息与传播学院
> 工程，
> > 成均馆大学, 2066, Seobu-ro, Changan-gu, Suwon-si,
> 韩国京畿道 (440-746)
> > 电话：+82-31-290-7197
> ==================================================== ==========
> […](#)

How did you select the training data? I choose training_0000tar~training_0007tar on waymo dataset v1.2.

thptai, 2022-01-27:
I used the default setting, with full - raw waymo dataset.
The code only selects 20% after you decode the raw dataset.
The full waymo dataset is around 158k samples. 20% of the full dataset is like 30-32k samples.

kyrie-H, 2022-01-27:
> I used the default setting, with full - raw waymo dataset. The code only selects 20% after you decode the raw dataset. The full waymo dataset is around 158k samples. 20% of the full dataset is like 30-32k samples.

Thanks a lot! By the way, the  SAMPLED_INTERVAL: {
    'train': 5,
    'test': 1
} means  20% of the raw data, right?

kyrie-H, 2022-01-27:
> > > I used the default setting, with full - raw waymo dataset. The code only selects 20% after you decode the raw dataset. The full waymo dataset is around 158k samples. 20% of the full dataset is like 30-32k samples.
> > 
> > 
> > Thanks a lot! By the way, the SAMPLED_INTERVAL: { 'train': 5, 'test': 1 } means 20% of the raw data, right?
> 
> Yes. 20% of decoded data which you converted from the raw data.

I get it. Thanks again!

closed 2022-01-27 by kyrie-H

jwma0725, 2022-06-05:
@thptai @sshaoshuai  @kyrie-H  Hi, I want to know how to use the full data. I see some methods set SAMPLED_INTERVAL:{'train':5,'test':5}, which means 100% of the raw data, right? And I also want to konw how to calculate Waymo Overall and I only get "0-30","30-50","50-inf". Could u help me? Thank u very much!

Rzlnmc, 2024-01-21:
> @thptai @sshaoshuai @kyrie-H Hi, I want to know how to use the full data. I see some methods set SAMPLED_INTERVAL:{'train':5,'test':5}, which means 100% of the raw data, right? And I also want to [...]

Hi, I have the same questions as you, do you find the answers?