open-mmlab/OpenPCDet #779: Multi Sweep Waymo Models

OrcunCanDeniz, 2022-01-28:
Is it possible to operate on multi sweep data from Waymo?

jihanyang, 2022-01-28:
Sure, but you need to re-generate the dataset info to save sweep information as nuscenes dataset

OrcunCanDeniz, 2022-01-28:
As I understand, I need to run;
`python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos \
    --cfg_file tools/cfgs/dataset_configs/waymo_multi_sweep_dataset.yaml`

After creating a `waymo_multi_sweep_dataset.yaml `

    DATASET: 'WaymoDataset'
    DATA_PATH: '../data/waymo'
    PROCESSED_DATA_TAG: 'waymo_processed_data_v0_5_0'
    MAX_SWEEPS: 10 ## MODIFIED THIS FROM ORIGINAL waymo_dataset.yaml
    POINT_CLOUD_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
    
    DATA_SPLIT: {
        'train': train,
        'test': val
    }
    
    SAMPLED_INTERVAL: {
        'train': 5,
        'test': 1
    }
    
    FILTER_EMPTY_BOXES_FOR_TRAIN: True
    DISABLE_NLZ_FLAG_ON_POINTS: True
    
    USE_SHARED_MEMORY: False  # it will load the data to shared memory to speed up (DO NOT USE IT IF YOU DO NOT FULLY UNDERSTAND WHAT WILL HAPPEN)
    SHARED_MEMORY_FILE_LIMIT: 35000  # set it based on the size of your shared memory
    
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder']
        AUG_CONFIG_LIST:
            - NAME: gt_sampling
              USE_ROAD_PLANE: False
              DB_INFO_PATH:
                  - waymo_processed_data_v0_5_0_waymo_dbinfos_train_sampled_1_msweep.pkl
    
              USE_SHARED_MEMORY: False  # set it to True to speed up (it costs about 15GB shared memory)
              DB_DATA_PATH:
                  - waymo_processed_data_v0_5_0_gt_database_train_sampled_1_global_msweep.npy
    
              PREPARE: {
                 filter_by_min_points: ['Vehicle:5', 'Pedestrian:5', 'Cyclist:5'],
                 filter_by_difficulty: [-1],
              }
    
              SAMPLE_GROUPS: ['Vehicle:15', 'Pedestrian:10', 'Cyclist:10']
              NUM_POINT_FEATURES: 5
              REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
              LIMIT_WHOLE_SCENE: True
    
            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x', 'y']
    
            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]
    
            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]
    
    
    POINT_FEATURE_ENCODING: {
        encoding_type: absolute_coordinates_encoding,
        used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],  ## MODIFIED THIS FROM ORIGINAL waymo_dataset.yaml
        src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
    }
    
    
    DATA_PROCESSOR:
        - NAME: mask_points_and_boxes_outside_range
          REMOVE_OUTSIDE_BOXES: True
    
        - NAME: shuffle_points
          SHUFFLE_ENABLED: {
            'train': True,
            'test': True
          }
    
        - NAME: transform_points_to_voxels
          VOXEL_SIZE: [0.1, 0.1, 0.15]
          MAX_POINTS_PER_VOXEL: 5
          MAX_NUMBER_OF_VOXELS: {
            'train': 150000,
            'test': 150000
          }

jihanyang, 2022-01-28:
However, I haven't find the place to load multiple sweeps in waymo_dataset. I will ask @sshaoshuai to reply this issue later.

sshaoshuai, 2022-01-29:
You do not need to re-generate the data, since the generated infos of OpenPCDet (v0.5) already contains the needed pose matrix for multi-sweep alignment:

https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/waymo/waymo_utils.py#L226-L227

However, you still need to wriite codes in the ``__getitem__`` to align the multi-sweep LiDAR point clouds (remove the ego-motion by using the above pose matrix) as the input points for multi-sweep training and testing.

OrcunCanDeniz, 2022-01-31:
Thank you. Would you accept a PR in case i implement this?

jihanyang, 2022-02-01:
@OrcunCanDeniz Sure.

closed 2022-02-01 by OrcunCanDeniz

sshaoshuai, 2022-02-02:
If you would like to submit a PR related to this, it is preferred to also provide some multi-sweep results.
We will denifitely accept such a PR with more elegent implementation or better results, given the context that we also have an internal implementation of multi-sweep Waymo codes (still not released yet since we do not have many results for multi-sweep training/testing).

SakuraRiven, 2022-08-12:
@sshaoshuai  @jihanyang  @OrcunCanDeniz  Hi, are there any documents to guide multi-sweeps on waymo? I am a beginner and try to learn from nuscene, but these two are quite different...

SakuraRiven, 2022-08-12:
Another question is, do we have to change SAMPLED_INTERVAL and generate new datasets when using different proportions?