open-mmlab/OpenPCDet #782: Use pip instead of the deprecated easy_install

mantkiew, 2022-01-31:
Instead of `python setup.py develop` as indicated in https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md#install-pcdet-v05 users should be instructed to use 

```
python3 -m pip install --user -e .
```
Otherwise, the users receive an error:
```
/<home>/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py:157: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
```

Additionally, even when running `pip install -e` (which is equivalent to `python setup.py develop`, I am getting the following error (in either case):
```
Installing collected packages: pcdet
  Running setup.py develop for pcdet
    error: subprocess-exited-with-error
    
    × python setup.py develop did not run successfully.
    │ exit code: 1
    ╰─> [29 lines of output]
        running develop
        /home/ma/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py:157: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
          warnings.warn(
        Traceback (most recent call last):
          File "<string>", line 2, in <module>
          File "<pip-setuptools-caller>", line 34, in <module>
          File "/home/ma/pcdet-object-detection-ros-docker/pcdet-object-detection-ros/OpenPCDet/setup.py", line 34, in <module>
            setup(
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/__init__.py", line 155, in setup
            return distutils.core.setup(**attrs)
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 148, in setup
            return run_commands(dist)
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
            dist.run_commands()
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
            self.run_command(cmd)
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 985, in run_command
            cmd_obj.ensure_finalized()
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 107, in ensure_finalized
            self.finalize_options()
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/command/develop.py", line 52, in finalize_options
            easy_install.finalize_options(self)
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py", line 276, in finalize_options
            self._fix_install_dir_for_user_site()
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py", line 382, in _fix_install_dir_for_user_site
            self.create_home_path()
          File "/home/ma/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py", line 1338, in create_home_path
            if path.startswith(home) and not os.path.isdir(path):
        AttributeError: 'int' object has no attribute 'startswith'
        [end of output]
```
I was not able to find a fix for this problem.

mantkiew, 2022-01-31:
FYI. I am on Ubuntu 20.04 with Python 3.8.10, Nvidia driver 510, CUDA 11.3, PyTorch `1.10.2+cu113`.

danielgrzenda, 2022-02-01:
I'm running into a similar problem building a docker container. It started occuring today. Ubuntu 20.04, Python 3.9.10, Cuda 11.3.1 and Pytorch nightly. Setuptools is pinned to 60.5 and the same command is failing `python3 -m pip install --user -e .`

msteijaert, 2022-02-02:
I had the same issue with python 3.9. I could solve it by installing setuptools==60.7.0 (which was released earlier today).

jabberwockyCode, 2022-02-02:
Same issue and had it fixed with @msteijaert 's solution

danielgrzenda, 2022-02-02:
Confirmed it worked for me as well.

closed 2022-03-19 by github-actions[bot]

AmrinKareem, 2023-08-08:
Worked for me too. Thank you!

David-ops-s, 2023-11-01:
Confirmed works for me, pip3 --upgrade setuptools does not upgrade automatically, has to specify the version

splendourbell, 2024-03-20:
Worked for me too. Thank you!

vamsikrishna7909, 2024-10-11:
I'm trying to use `python setup.py develop` from below instructions :
# Clone this repo
git clone https://github.com/KaiyangZhou/Dassl.pytorch.git
cd Dassl.pytorch/

# Create a conda environment
conda create -y -n dassl python=3.8

# Activate the environment
conda activate dassl

# Install torch (requires version >= 1.8.1) and torchvision
# Please refer to https://pytorch.org/ if you need a different cuda version
conda install pytorch torchvision cudatoolkit=10.2 -c pytorch

# Install dependencies
pip install -r requirements.txt

# Install this library (no need to re-build if the source code is modified)
python setup.py develop

Facing the below error:`python setup.py develop
running develop
/home/008458886@csusb.edu/ENTER/envs/dassl/lib/python3.8/site-packages/setuptools/command/develop.py:41: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
/home/008458886@csusb.edu/ENTER/envs/dassl/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
running egg_info
creating dassl.egg-info
error: could not create 'dassl.egg-info': Permission denied`

did anyone faced same issue? Please respond as I tried to degrade setup tools from 75.1.0(latest) to lower versions. It didn't work still.

aldakata, 2024-10-20:
FYI, using pip on a conda env, I had to conda install setuptools==60.7.0, for the fix to work. pip install setuptools==60.7.0 did not get the job done