open-mmlab/OpenPCDet #783: error while running ' python setup.py develop '

eiffelbastin, 2022-02-01:
I'm getting this error while trying to use the prebuild

`C:\Users\91828\Desktop\OpenPCDet>python setup.py develop
running develop
C:\Users\91828\AppData\Local\Programs\Python\Python38\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\91828\AppData\Local\Programs\Python\Python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing pcdet.egg-info\PKG-INFO
writing dependency_links to pcdet.egg-info\dependency_links.txt
writing requirements to pcdet.egg-info\requires.txt
writing top-level names to pcdet.egg-info\top_level.txt
reading manifest file 'pcdet.egg-info\SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info\SOURCES.txt'
running build_ext
C:\Users\91828\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\utils\cpp_extension.py:316: UserWarning: Error checking compiler version for cl: [WinError 2] The system cannot find the file specified
  warnings.warn(f'Error checking compiler version for {compiler}: {error}')
error: [WinError 2] The system cannot find the file specified`

can someone help me identify the issue and solve it

eiffelbastin, 2022-02-04:
[image]

djiajunustc, 2022-02-07:
Hi @eiffelbastin ,

We haven't tried to use OpenPCDet in windows system yet.
It would be nice if you can help us to develop the supporting version for windows.

closed 2022-03-24 by github-actions[bot]