open-mmlab/OpenPCDet #785: performance of SECOND on Waymo

pigtigger, 2022-02-03:
I trained second (8GPUs) with config from pcdet  but got much worse performance than the results reported in README.  Will the version of spconv  be the reason?  I  use spconv 2 in my experiment and here is the results of 30 epoch.  
[image]

sshaoshuai, 2022-02-03:
Have you generated the new data info with pcdet v0.5+?

pigtigger, 2022-02-04:
@sshaoshuai Not yet, I created the data with pcdet 0.3, so the new version of data can improve performance?

pigtigger, 2022-02-06:
@sshaoshuai, I generated the new waymo data and train SECOND (cfgs/waymo_models/second.yaml) again with the latest code,  but I met a strange bug: every thing was good at the beginning, but after some epochs the loss became nan and finally the evaluated mAP is zero. I did not change any code, did you meet this problem before?
[image]
[image]

jihanyang, 2022-02-06:
@pigtigger Please just re-run the experiment again. This is a known issue, but will not take place frequently.

pigtigger, 2022-02-06:
@jihanyang Thank you, I will have a try, btw, do you know why this problem occurs frequently? I meet this nan bug in my own model but can not figure out what causes it.

jihanyang, 2022-02-06:
Actually, we are still not clear about the reason, since this is problem not easy to trace and hard to reproduce. From my experience, even the same code, this error could take place frequently in some machines while not appear at all in other machines. If you have any suggestion to solve this, feel free to let us know.

pigtigger, 2022-02-06:
Thank you :) @jihanyang , re-run the experiment does work, I will share my solution if find it.

closed 2022-02-07 by jihanyang