open-mmlab/OpenPCDet #789: How  to test kitti trained models like pv_rcnn on pandaset ?

Leozyc-waseda, 2022-02-09:
I am trying to test kitti trained models like pv_rcnn on pandaset.
Can someone tell me how to do it ? I have loaded pandaset with Dataloader.

I used this command.
`python test.py --cfg_file cfgs/pandaset_models/pv_rcnn.yaml --batch_size 1 --ckpt pv_rcnn_8369.pth     --save_to_file`

```
../pcdet/models/roi_heads/pvrcnn_head.py:126: UserWarning: This overload of nonzero is deprecated:
	nonzero()
Consider using one of the following signatures instead:
	nonzero(*, bool as_tuple) (Triggered internally at  /opt/conda/conda-bld/pytorch_1603729047590/work/torch/csrc/utils/python_arg_parser.cpp:882.)
  dense_idx = faked_features.nonzero()  # (N, 3) [x_idx, y_idx, z_idx]
Traceback (most recent call last):
  File "test.py", line 200, in <module>
    main()
  File "test.py", line 196, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 64, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/algo-4/work/OpenPCDet/tools/eval_utils/eval_utils.py", line 63, in eval_one_epoch
    output_path=final_output_dir if save_to_file else None
  File "../pcdet/datasets/pandaset/pandaset_dataset.py", line 328, in generate_prediction_dicts
    single_pred_df = generate_single_sample_dataframe(index, box_dict, zrot, pose)
  File "../pcdet/datasets/pandaset/pandaset_dataset.py", line 300, in generate_single_sample_dataframe
    world_centers = ps.geometry.ego_to_lidar_points(ego_centers, pose_dict)
AttributeError: module 'pandaset.geometry' has no attribute 'ego_to_lidar_points'
eval:   0%|             
```

VsionQing, 2022-02-23:
Do you realize the training under windows?

Leozyc-waseda, 2022-02-23:
@VsionQing 

Under ubuntu20.04, it works well.

VsionQing, 2022-02-23:
Could you test my files for me? I can't run successfully under windows, and the python configuration of Linux virtual machine is not very smooth. I want to see if it's the reason of the system. If you can, leave an email

Leozyc-waseda, 2022-02-23:
> Could you test my files for me? I can't run successfully under windows, and the python configuration of Linux virtual machine is not very smooth. I want to see if it's the reason of the system. If you can, leave an email

No.

YCAyca, 2022-03-17:
https://github.com/lea-v/pandaset-devkit/blob/feature/addPointCloudTransformations/python/pandaset/geometry.py use this branch of pandaset. 'ego_to_lidar_points' function is inside

Leozyc-waseda, 2022-03-17:
@YCAyca 
thank you.
I already know how to solve this issue.

closed 2022-03-17 by Leozyc-waseda