open-mmlab/OpenPCDet #790: PV-RCNN++ training custom dataset

luoxiaoliaolan, 2022-02-09:
Hi!
      Thank you so much for releasing the latest pv-rcnn++ code，I try to train my own data with this code. I organized my own data according to the format of the kitti dataset. I follow "pv_rcnn_plusplus.yaml" from waymo_models folder to custom_training_model_cfg. But when I start training, I get an error like this：
[image]

Also, I provide information about custom dataset:
                       The dimensions of point cloud data： [x, y, z, intensity]

According to the displayed error message, I am located models/backbones_3d/pfe/voxel_set_abstraction.py  line 375, function self.aggregate_keypoint_features_from_one_source,  
xyz_features=raw_points[:, 4:].contiguous() if raw_points.shape[1] > 4 else None,
where xyz_features' s shape is (246186, 1)  
It caused the code here(pointnet2_modules.py  line 396 
 N, C = features.shape
        assert C % self.num_reduced_channels == 0, \
            f'the input channels ({C}) should be an integral multiple of num_reduced_channels({self.num_reduced_channels})' ) to report an error.
[image]

I feel that this may be caused by the inconsistent dimensional information between my own data and the data of the waymo dataset. Can you tell me how to modify the model configuration （pv_rcnn_plusplus.yaml） to fit custom dataset? (I have read this issue: https://github.com/open-mmlab/OpenPCDet/issues/747)   At the same time, I also want to try to adapt the models trained on the waymo dataset to the format of the kitti dataset. Can you help?

              Among the questions raised above, I look forward to communicating with you more！
                                                              Thank a lot！

Fllacsaa, 2022-02-10:
I have the same error as you by using my own data.

sshaoshuai, 2022-02-10:
Hi, 

Please try to modify the line of ```SA_LAYER.raw_points.NUM_REDUCED_CHANNELS=2```  as follows: 
```
NUM_REDUCED_CHANNELS: 1
```
Since your own data just has points with shape (N, 3+1) instead of (N, 3+2) like waymo dataset.

luoxiaoliaolan, 2022-02-10:
Thanks for your reply, as per your suggestion, I try to modify the line of SA_LAYER.raw_points.NUM_REDUCED_CHANNELS=1:
[image]
When I trained custom dataset, I got this error:
[image]
Do you think I should continue modifying the parameters of the model configuration(pv_rcnn_plusplus.yaml) or should I start by building my own dataset?
Attach the configuration of custom dataset：
`DATASET: 'AsDataset'
DATA_PATH: '/mnt/NAS/X01_datasets/as_label_data/AS_annotation_20/113'

POINT_CLOUD_RANGE: [0.0, -40, -3, 128, 40, 1]

DATA_SPLIT: {
    'train': train,
    'test': val
}

INFO_PATH: {
    'train': [as_infos_train.pkl],
    'test': [as_infos_val.pkl],
}

BALANCED_RESAMPLING: True

GET_ITEM_LIST: ["points"]
FOV_POINTS_ONLY: True

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: gt_sampling
          DB_INFO_PATH:
              - as_dbinfos_train.pkl
          PREPARE: {
             filter_by_min_points: ['Car:20', 'Pedestrian:8', 'bicycle:10', 'tricycle:12', 'cyclist:15', 'motorcyclist:15',
                                    'tricyclist:15', 'van:20', 'bus:40', 'truck:30', 'mini_truck:25', 'special_vehicle:35',
                                    'traffic_cone:10', 'small_movable:5', 'small_unmovable:7', 'crash_barrel:8', 'construction_sign:10',
                                    'noise:17', 'water_horse:14', 'other:15']
          }

          SAMPLE_GROUPS: ['Car:20', 'Pedestrian:8', 'bicycle:10', 'tricycle:12', 'cyclist:15', 'motorcyclist:15',
                                    'tricyclist:15', 'van:20', 'bus:40', 'truck:30', 'mini_truck:25', 'special_vehicle:35',
                                    'traffic_cone:10', 'small_movable:5', 'small_unmovable:7', 'crash_barrel:8', 'construction_sign:10',
                                    'noise:17', 'water_horse:14', 'other:15']
          NUM_POINT_FEATURES: 4
          DATABASE_WITH_FAKELIDAR: False
          REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
          LIMIT_WHOLE_SCENE: True

        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x']

        - NAME: random_world_rotation
          WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

        - NAME: random_world_scaling
          WORLD_SCALE_RANGE: [0.95, 1.05]

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': False
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.05, 0.05, 0.1]
      MAX_POINTS_PER_VOXEL: 20
      MAX_NUMBER_OF_VOXELS: {
        'train': 300000,
        'test': 400000
      }
`

luoxiaoliaolan, 2022-02-10:
Attach pv_rcnn_plusplus.yaml：
`CLASS_NAMES: ['car', 'pedestrian', 'bicycle', 'tricycle', 'cyclist', 'motorcyclist', 'tricyclist',
                   'van', 'bus', 'truck', 'mini_truck', 'special_vehicle', 'traffic_cone', 'small_movable',
                   'small_unmovable', 'crash_barrel', 'construction_sign', 'noise', 'water_horse', 'other']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/as_dataset.yaml

MODEL:
    NAME: PVRCNNPlusPlus

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: CenterHead
        CLASS_AGNOSTIC: False

        CLASS_NAMES_EACH_HEAD: [
            [ 'car', 'pedestrian', 'bicycle', 'tricycle', 'cyclist', 'motorcyclist', 'tricyclist',
                   'van', 'bus', 'truck', 'mini_truck', 'special_vehicle', 'traffic_cone', 'small_movable',
                   'small_unmovable', 'crash_barrel', 'construction_sign', 'noise', 'water_horse', 'other' ]
        ]

        SHARED_CONV_CHANNEL: 64
        USE_BIAS_BEFORE_NORM: True
        NUM_HM_CONV: 2
        SEPARATE_HEAD_CFG:
            HEAD_ORDER: [ 'center', 'center_z', 'dim', 'rot' ]
            HEAD_DICT: {
                'center': { 'out_channels': 2, 'num_conv': 2 },
                'center_z': { 'out_channels': 1, 'num_conv': 2 },
                'dim': { 'out_channels': 3, 'num_conv': 2 },
                'rot': { 'out_channels': 2, 'num_conv': 2 },
            }

        TARGET_ASSIGNER_CONFIG:
            FEATURE_MAP_STRIDE: 8
            NUM_MAX_OBJS: 500
            GAUSSIAN_OVERLAP: 0.1
            MIN_RADIUS: 2

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'code_weights': [ 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0 ]
            }

        POST_PROCESSING:
            SCORE_THRESH: 0.5
            POST_CENTER_LIMIT_RANGE: [ -75.2, -75.2, -2, 75.2, 75.2, 4 ]
            MAX_OBJ_PER_SAMPLE: 500
            NMS_CONFIG:
[... 183 lines cut ...]
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.7
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 1
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.01
    WEIGHT_DECAY: 0.001
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10`

closed 2022-02-14 by sshaoshuai

luoxiaoliaolan, 2022-02-19:
Thanks for your patience！But I still have this problem while training：
`&&&&&&&&&&&&&&&&&&&&&&&&&&&info_path:______ /mnt/pfs/jarvis/at_dataset/at_infos_train.pkl

epochs:   0%|          | 0/80 [00:00<?, ?it/s]/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/models/model_utils/centernet_utils.py:142: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  topk_ys = (topk_inds // width).float()
/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/models/model_utils/centernet_utils.py:146: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  topk_classes = (topk_ind // K).int()
/opt/conda/lib/python3.7/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1634272178570/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]

                                              
Traceback (most recent call last):
  File "tools/train.py", line 204, in <module>
    main()
  File "tools/train.py", line 173, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/tools/train_utils/train_utils.py", line 118, in train_model
    dataloader_iter=dataloader_iter
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/tools/train_utils/train_utils.py", line 47, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/models/detectors/pv_rcnn_plusplus.py", line 28, in forward
    batch_dict = self.pfe(batch_dict)
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 380, in forward
    rois=batch_dict.get('rois', None)
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 330, in aggregate_keypoint_features_from_one_source
    features=xyz_features.contiguous(),
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_modules.py", line 461, in forward
    cur_xyz, cur_features = self.__getattr__(f'layer_{k}')(**kwargs)
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/output/train-result/f68eb42e-29f5-4a20-8465-f29bb3b8fac8/code_2/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_modules.py", line 399, in forward
    features = features.view(N, -1, self.num_reduced_channels).sum(dim=1)
RuntimeError: cannot reshape tensor of 0 elements into shape [0, -1, 1] because the unspecified dimension size -1 can be any value and is ambiguous
========trainExitCode========
1` 

Code：
[image]

pv_rcnn_plusplus.yaml:
CLASS_NAMES: ['car', 'pedestrian', 'bicycle', 'tricycle', 'cyclist', 'motorcyclist', 'tricyclist',
                   'van', 'bus', 'truck', 'mini_truck', 'special_vehicle', 'traffic_cone', 'small_movable',
                   'small_unmovable', 'crash_barrel', 'construction_sign', 'noise', 'water_horse', 'other']

DATA_CONFIG:
    _BASE_CONFIG_: tools/cfgs/dataset_configs/as_dataset.yaml

MODEL:
    NAME: PVRCNNPlusPlus

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
[... 314 lines cut ...]
POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': False
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.05, 0.05, 0.1]
      MAX_POINTS_PER_VOXEL: 10
      MAX_NUMBER_OF_VOXELS: {
        'train': 300000,
        'test': 400000
      }

Should I modify that module to solve this training problem?

luoxiaoliaolan, 2022-02-19:
request to reopen this issue@sshaoshuai

sshaoshuai, 2022-03-12:
Sorry, just notice this issue. 

I didn't try PV-RCNN++ with so many classes and I suggest you to step by step check the error, and I guess it is still the problem of empty RoIs (without any points).

BOBrown, 2022-07-12:
@luoxiaoliaolan 
I have met the same issue, and I solved this issue by setting the NUM_POINTS_OF_EACH_SAMPLE_PART in pv_rcnn_plusplus.yaml:

PFE:
        NAME: VoxelSetAbstraction
        POINT_SOURCE: raw_points
        NUM_KEYPOINTS: 4096
        NUM_OUTPUT_FEATURES: 90
        SAMPLE_METHOD: SPC
        SPC_SAMPLING:
            NUM_POINTS_OF_EACH_SAMPLE_PART: 4000000
            NUM_SECTORS: 6
            SAMPLE_RADIUS_WITH_ROI: 1.6

frothmoon, 2024-01-09:
Hello, did you solve this problem? I also encountered the same problem, looking forward to your reply.