open-mmlab/OpenPCDet #800: PointPillars NuScenes velocity

ahmedius2, 2022-02-13:
Hello,

The multi-head PointPillars model outputs the velocity of detected objects in the X and Y axes. Are these velocities with respect to the ego vehicle's velocity, or are they global velocities? Also, are their metric is meters/second? I am trying to predict the future positions of the objects using the ego pose and pose/velocity of the detected objects.

Best regards,
Ahmet

sshaoshuai, 2022-02-19:
Yes, the predicted velocities are with respect to the ego vehicle's coordinate since the velocity labels are normalized to this coordinate in the training process.

ahmedius2, 2022-02-19:
Thanks! Is the ego vehicle's X-axis always parallel to the global X-axis or could it have a different orientation? Also same question for Y? I am asking this to understand whether there is a need to use the ego vehicle's orientation to determine the velocity of detected objects with respect to the global XY frame.

sshaoshuai, 2022-02-19:
For OpenPCDet, currently we need to align the direction of velocity (dx, dy) in the ego vehicle's coordinate for both training and inference. 

For the global coordinate of nuscene dataset, maybe you can get some information from here: 

https://github.com/nutonomy/nuscenes-devkit/blob/master/python-sdk/nuscenes/utils/data_classes.py#L534-L541

https://github.com/nutonomy/nuscenes-devkit/blob/master/python-sdk/nuscenes/utils/data_classes.py#L594-L601

ahmedius2, 2022-02-27:
Many thanks, I made a lot of progress on the prediction, what you shared with me was very helpful. I couldn't fully understand what you mean by saying that you need to align the direction of velocity. Currently, is it not possible to calculate the global velocity of the objects using the velocity output by the network assuming we have access to global velocity of the ego vehicle?

ahmedius2, 2022-02-28:
If the previous question is not clear, what I am trying to understand is, when the network gives a velocity output, is this velocity relative to the ego vehicle's velocity, or is it global velocity in the ego vehicle's coordinate frame? From the code, the second one seems to be true. So the network outputs absolute velocity but in the ego vehicle's coordinate frame, AFAIU.

closed 2022-04-14 by github-actions[bot]

quan-dao, 2022-09-13:
Hi @ahmedius2,
I think the ground truth velocities are calculated relative to **LiDAR**, not ego vehicle.

The ground truth boxes' info (loc, size, heading, velocity) is calculated [here](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/nuscenes/nuscenes_dataset.py#L284) by the function [`fill_trainval_infos`](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L252) of `nuscenes_utils.py`.

In the function `fill_trainval_infos`, ground truth boxes are parsed by the function [`get_sample_data`](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L185) which uses NuScenes's [`get_boxes`](https://github.com/nutonomy/nuscenes-devkit/blob/28765b8477dbd3331bacd922fada867c2c4db1d7/python-sdk/nuscenes/nuscenes.py#L319) to get ground truth from annotation files. The `get_boxes` function provides gt boxes in the **global** frame (reference: section sample_annotation of NuScenes's [data format](https://www.nuscenes.org/nuscenes#data-format)).

The resulting `boxes` are then mapped to the ego vehicle frame ([line 221](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L221url)) and then to sensor frame ([line 225](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L225)). Notice that sensor frame is defined by `cs_record` ([line 199](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L199)) extracted from `sd_record` (line 198) which is defined by the inputted `sample_data_token`. 

If you come back to function [`fill_trainval_infos`](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L252) , where `get_sample_data` is called, [`ref_sd_token`](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L263) is passed to  `get_sample_data` at [line 269](https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/nuscenes/nuscenes_utils.py#L269) . Therefore, gt boxes are mapped to `ref_sd_token`, i.e. **LiDAR** frame.

ryanyej, 2024-01-26:
you are abosolutely right! @quan-dao

ryanyej, 2024-03-13:
> Hi @ahmedius2, I think the ground truth velocities are calculated relative to **LiDAR**, not ego vehicle.
>
> The ground truth boxes' info (loc, size, heading, velocity) is calculated [...]

After I review the code, I found velocity is directly come from [translation] from sample_annotation. 
And I found defination of **sample_annotation**
A bounding box defining the position of an object seen in a sample. All location data is given with respect **to the global coordinate system.**
so velocity is defined in global coordinate system.