open-mmlab/OpenPCDet #804: how to install pcdet??

luminous33, 2022-02-15:
(openpcdet) user@user-SYS-7049GP-TRT:/data/test_model/OpenPCDet$ python setup.py develop
running develop
/usr/local/anaconda3/envs/openpcdet/lib/python3.8/site-packages/setuptools/command/easy_install.py:157: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/anaconda3/envs/openpcdet/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /data/test_model/OpenPCDet/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
1.10.2.git.kitware.jobserver-1
g++: error: /data/test_model/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o: No such file or directory
g++: error: /data/test_model/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms.o: No such file or directory
g++: error: /data/test_model/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o: No such file or directory
g++: error: /data/test_model/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o: No such file or directory
error: command '/usr/bin/g++' failed with exit code 1

## pip list
Package            Version
------------------ ---------
ccimport           0.3.7
certifi            2020.6.20
charset-normalizer 2.0.12
click              8.0.3
cmake              3.22.2
cumm               0.2.8
easydict           1.9
fire               0.4.0
githubrelease      1.5.8
idna               3.3
imageio            2.16.0
lark               1.1.1
LinkHeader         0.4.3
llvmlite           0.38.0rc1
mkl-fft            1.3.1
mkl-random         1.2.2
mkl-service        2.4.0
networkx           2.6.3
ninja              1.10.2.3
numba              0.55.1
numpy              1.20.0
olefile            0.46
packaging          21.3
pccm               0.3.4
Pillow             8.3.2
pip                21.2.4
portalocker        2.3.2
protobuf           3.19.4
pybind11           2.9.1
pyparsing          3.0.7
PyWavelets         1.2.0
PyYAML             6.0
requests           2.27.1
scikit-image       0.19.1
scipy              1.8.0
setuptools         60.7.0
SharedArray        3.2.1
six                1.16.0
spconv             2.1.21
tensorboardX       2.4.1
termcolor          1.1.0
tifffile           2022.2.9
torch              1.10.2
torchaudio         0.10.2
torchvision        0.11.3
tqdm               4.62.3
typing-extensions  3.10.0.2
urllib3            1.26.8
wheel              0.37.1

## NVIDIA-SMI
Tue Feb 15 11:01:56 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 465.19.01    Driver Version: 465.19.01    CUDA Version: 11.3     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:18:00.0 Off |                  N/A |
| 30%   29C    P8    17W / 350W |      2MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce ...  Off  | 00000000:3B:00.0 Off |                  N/A |
| 55%   28C    P8    26W / 350W |      2MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   2  NVIDIA GeForce ...  Off  | 00000000:86:00.0 Off |                  N/A |
| 53%   27C    P8    23W / 350W |      0MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   3  NVIDIA GeForce ...  Off  | 00000000:AF:00.0 Off |                  N/A |
| 54%   26C    P8    20W / 350W |      0MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

## nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Sun_Mar_21_19:15:46_PDT_2021
Cuda compilation tools, release 11.3, V11.3.58
Build cuda_11.3.r11.3/compiler.29745058_0

## conda & python version
Python 3.8.12
conda 4.10.1

## gcc version
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 

## g++ version
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 

I keep getting g++ errors. 
I dont know what the problem is.
Please tell me how to solve this issue.

Thank you:)

sshaoshuai, 2022-02-19:
Have you tried to run the following two commands before you run `python setup.py develop`?
```
python setup.py clean

rm -rf ./build
```

luminous33, 2022-02-21:
> Have you tried to run the following two commands before you run `python setup.py develop`?
> 
> ```
> python setup.py clean
> 
> rm -rf ./build
> ```

Yes!!
I already tried it, but it didn't install.
So i need your help :(

sshaoshuai, 2022-02-21:
Sorry, I have no idea about this error. 
It seems like some path related errors.

The simpliest way is to create a fully new enviroment and follow README.md to build OpenPCDet.

luminous33, 2022-02-21:
So How to deal with ninja error?
like Command '['ninja', '-v']' returned non-zero exit status 1
I changed '-v' -> '--version' but I know this method is wrong.
What action should I take?

sshaoshuai, 2022-02-21:
No idea about this error.

luminous33, 2022-02-21:
Thk for your reply :)

luminous33, 2022-02-21:
> No idea about this error.

if do you have Docker file???

sshaoshuai, 2022-02-22:
Please refer to here https://github.com/open-mmlab/OpenPCDet/tree/master/docker

luminous33, 2022-02-23:
> Please refer to here https://github.com/open-mmlab/OpenPCDet/tree/master/docker

Awesome!!!!!
Thank u:)

rbldrbld, 2022-03-02:
same error

closed 2022-04-17 by github-actions[bot]