open-mmlab/OpenPCDet #805: Evaluate on pandaset get very low precision

Leozyc-waseda, 2022-02-15:
@sshaoshuai @lindahua 
Thank you so much for your awesome work.

I want to use the model trained on Kitti to evaluate the accuracy on pandaset.

in pandaset_dataset.py, I changed evaluation function just like kitti evaluation.

I searched for a lot of methods on the Internet, mainly there are two methods.
1. Import Pandaset by dataloader, and modify the evaluation function.
2. Convert Pandaset to kitti format for evaluation.

➀ried to use dataloader and imitate kitti's evaluation function.
I am getting very low precision, can you tell me why?Maybe how to get the right answer,please.
```   def evaluation(self, det_annos, class_names, **kwargs):

        # self.logger.warning('Evaluation is not implemented for Pandaset as there is no official one. ' +
        #                     'Returning an empty evaluation result.')
        if 'annos' not in self.pandaset_infos[0].keys():
            return None, {}

        from .kitti_object_eval_python import eval as kitti_eval

        eval_det_annos = copy.deepcopy(det_annos)
        eval_gt_annos = [copy.deepcopy(info['annos']) for info in self.pandaset_infos]
        ap_result_str, ap_dict = kitti_eval.get_official_eval_result(eval_gt_annos, eval_det_annos, class_names)

        return ap_result_str, ap_dict
```  

And I also changed pv_rcnn.yaml

```  
CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/pandaset_dataset.yaml
MODEL:
    NAME: PVRCNN

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
[... 490 lines cut ...]
2022-02-16 05:54:03,555   INFO  Not updated weight pfe.SA_layers.1.mlps.1.4.running_mean: torch.Size([64])
2022-02-16 05:54:03,555   INFO  Not updated weight pfe.SA_layers.1.mlps.1.4.running_var: torch.Size([64])
2022-02-16 05:54:03,555   INFO  Not updated weight pfe.vsa_point_feature_fusion.0.weight: torch.Size([128, 544])
2022-02-16 05:54:03,555   INFO  Not updated weight point_head.cls_layers.0.weight: torch.Size([256, 544])
2022-02-16 05:54:03,555   INFO  ==> Done (loaded 277/319)
2022-02-16 05:54:03,581   INFO  *************** EPOCH 8369 EVALUATION *****************
eval:   0%|                                                                                    | 0/1600 [00:00<?, ?it/s]../pcdet/models/roi_heads/pvrcnn_head.py:126: UserWarning: This overload of nonzero is deprecated:
nonzero()
Consider using one of the following signatures instead:
nonzero(*, bool as_tuple) (Triggered internally at  /opt/conda/conda-bld/pytorch_1603729047590/work/torch/csrc/utils/python_arg_parser.cpp:882.)
  dense_idx = faked_features.nonzero()  # (N, 3) [x_idx, y_idx, z_idx]
eval: 100%|████████████████████████████████████████| 1600/1600 [09:32<00:00,  2.79it/s, recall_0.3=(1841, 959) / 229468]
2022-02-16 06:03:36,369   INFO  *************** Performance of EPOCH 8369 *****************
2022-02-16 06:03:36,370   INFO  Generate label finished(sec_per_example: 0.3580 second).
2022-02-16 06:03:36,370   INFO  recall_roi_0.3: 0.008023
2022-02-16 06:03:36,370   INFO  recall_rcnn_0.3: 0.004179
2022-02-16 06:03:36,370   INFO  recall_roi_0.5: 0.000819
2022-02-16 06:03:36,370   INFO  recall_rcnn_0.5: 0.000296
2022-02-16 06:03:36,370   INFO  recall_roi_0.7: 0.000017
2022-02-16 06:03:36,370   INFO  recall_rcnn_0.7: 0.000000
2022-02-16 06:03:36,371   INFO  Average predicted number of objects(1600 samples): 0.000
2022-02-16 06:03:36,726   INFO  None
2022-02-16 06:03:36,726   INFO  Result is save to /home/work/work/OpenPCDet/output/pandaset_models/pv_rcnn/default/eval/epoch_8369/val/default
2022-02-16 06:03:36,726   INFO  ****************Evaluation done.*****************
```

Leozyc-waseda, 2022-02-15:
Then I converted pandaset to kitti format and tested it, this time there is no precision...
```
python test.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 1 --ckpt pv_rcnn_8369.pth \
>     --save_to_file
/home/work/anaconda3/envs/openpcdet-new/lib/python3.7/site-packages/skimage/io/manage_plugins.py:23: UserWarning: Your installed pillow version is < 8.1.2. Several security issues (CVE-2021-27921, CVE-2021-25290, CVE-2021-25291, CVE-2021-25293, and more) have been fixed in pillow 8.1.2 or higher. We recommend to upgrade this library.
  from .collection import imread_collection_wrapper
2022-02-16 06:13:28,083   INFO  **********************Start logging**********************
2022-02-16 06:13:28,083   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-02-16 06:13:28,083   INFO  cfg_file         cfgs/kitti_models/pv_rcnn.yaml
2022-02-16 06:13:28,083   INFO  batch_size       1
2022-02-16 06:13:28,083   INFO  workers          4
2022-02-16 06:13:28,083   INFO  extra_tag        default
2022-02-16 06:13:28,083   INFO  ckpt             pv_rcnn_8369.pth
2022-02-16 06:13:28,083   INFO  launcher         none
2022-02-16 06:13:28,083   INFO  tcp_port         18888
2022-02-16 06:13:28,083   INFO  local_rank       0
2022-02-16 06:13:28,083   INFO  set_cfgs         None
2022-02-16 06:13:28,083   INFO  max_waiting_mins 30
2022-02-16 06:13:28,083   INFO  start_epoch      0
2022-02-16 06:13:28,083   INFO  eval_tag         default
2022-02-16 06:13:28,083   INFO  eval_all         False
2022-02-16 06:13:28,083   INFO  ckpt_dir         None
2022-02-16 06:13:28,083   INFO  save_to_file     True
2022-02-16 06:13:28,083   INFO  cfg.ROOT_DIR: /home/work/work/OpenPCDet
2022-02-16 06:13:28,083   INFO  cfg.LOCAL_RANK: 0
2022-02-16 06:13:28,083   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2022-02-16 06:13:28,083   INFO  
cfg.DATA_CONFIG = edict()
2022-02-16 06:13:28,083   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2022-02-16 06:13:28,083   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2022-02-16 06:13:28,083   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2022-02-16 06:13:28,084   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-02-16 06:13:28,084   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-02-16 06:13:28,084   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-02-16 06:13:28,084   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2022-02-16 06:13:28,084   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2022-02-16 06:13:28,084   INFO  
cfg.MODEL = edict()
2022-02-16 06:13:28,084   INFO  cfg.MODEL.NAME: PVRCNN
2022-02-16 06:13:28,084   INFO  
cfg.MODEL.VFE = edict()
2022-02-16 06:13:28,084   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-02-16 06:13:28,084   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-02-16 06:13:28,084   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-02-16 06:13:28,084   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2022-02-16 06:13:28,084   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2022-02-16 06:13:28,084   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2022-02-16 06:13:28,084   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2022-02-16 06:13:28,084   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2022-02-16 06:13:28,084   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2022-02-16 06:13:28,084   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
[... 305 lines cut ...]
aos  AP:0.00, 0.00, 0.00
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
[... cut ...]
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00

2022-02-16 06:19:10,540   INFO  Result is save to /home/work/work/OpenPCDet/output/kitti_models/pv_rcnn/default/eval/epoch_8369/val/default
2022-02-16 06:19:10,540   INFO  ****************Evaluation done.*****************
```

LouisSF, 2022-02-18:
Hi, 
I think that this may be due to the fact that you use a model trained on Kitti, which uses different reference coordinates than Pandaset,in Kitti the x axis points in front of the car, whereas in Pandaset it points to the left. You may also have to adjust the height of the lidar point clouds in pandaset to account for the difference in Lidar sensors, though I haven't checked this precisely.

Leozyc-waseda, 2022-02-18:
@LouisSF 
Thank you for your reply.

Are you referring to the second method I used to convert Pandaset data into kitti ?
This reminds me, I will try to adjust the coordinates and use `demo.py` to see the result.

also, i noticed kitti point cloud range is [0, -40, -3, 70, 40, 1],

https://github.com/open-mmlab/OpenPCDet/blob/6a1495ca610186d3b9d5142a4a6506b8dc78bce1/tools/cfgs/dataset_configs/kitti_dataset.yaml#L4

for pandaset is [-70, -40, -3, 70, 40, 1] ,
https://github.com/open-mmlab/OpenPCDet/blob/6a1495ca610186d3b9d5142a4a6506b8dc78bce1/tools/cfgs/dataset_configs/pandaset_dataset.yaml#L4

I've been trying to test pandaset with dataloader. In the case of only detecting the car, good results are achieved, but I still don't know why cyclists and pedestrian are not detected.

I tried to change Xmin from 0 to -70, the detection result of pv-rcnn did not change, it seems that the pv-rcnn model trained on kitti only detected the part from 0 to 70, and the part from -70 to 0 was not detected at all.

If all objects from -70 to 0 cannot be detected, is it due to the use of the pv-rcnn model trained on Kitti?
I am retraining pv-rcnn with pandaset and it looks like it will take some time.

closed 2022-04-04 by github-actions[bot]

1uk4s, 2022-09-13:
@Leozyc-waseda in the last post you said that you trained pv-rcnn on pandaset. Did you succeed with that and got good results? I am also working on PandaSet and have the same issue with low performance.

Leozyc-waseda, 2022-09-13:
@1uk4s 

Due to the wide range of pandaset annotations, the results are not good when evaluated.
If you want to correctly evaluate something like PV-RCNN on Pandaset you need to remove the ground truth out of range.
After I visualized the results, it turned out pretty good with pandaset.

1uk4s, 2022-09-13:
Hi @Leozyc-waseda 
thank you so much for your reply. 

I already removed ground truth boxes out of range and also removed boxes with less than 10 points in it as there are still in the 40 x 70 meter area a lot of parked cars with very few points. 

I am using the following 5 classes, merged from original classes: (and removed boxes of other classes)
CAR: Car, Pickup Truck
PEDESTRIAN: Pedestrian, Pedestrian with object
BICYCLE: Bicycle
MOTORCYCLE: Motorcycle, Motorized Scooter
TRUCK: Medium-sized Truck, Bus, Emergency Vehicle, Semi-Truck

My best evaluation results are:
CAR @ 0.5 IOU:                        53 AP
CAR @0.7 IOU:                         65 AP
PEDESTRIAN @ 0.5 IOU:         25 AP
PEDESTRIAN @ 0.3 IOU:         34 AP
BICYCLE @ 0.5 IOU:                 0.02 AP
MOTORCYCLE @ 0.5 IOU:       9 AP 
TRUCK @ 0.7 IOU:                    6 AP

As you can see the results for all classes are far behind the results of the baseline in pandaset paper. Especially for the classes with not so much bounding boxes in the dataset (bicycle, truck and motorcycle)

Did you achieved better results ? I am especially wondering about bicycle, truck and motorcycle class.

I used PV-RCNN++ but that should not make so much difference. I am using no data augmentation.

Leozyc-waseda, 2022-09-13:
@1uk4s 
I only tested in Car, Pedestrian, and Bicycle.

Here are something I find out,

pandaset_dataset.yaml

best in 
 `POINT_CLOUD_RANGE: [-70, -40, -3, 70, 40, 1] # xmin, ymin, zmin, xmax, ymax, zmax`

not pandaset paper mentioned.

> [70:4m; 70:4m]
> for the mechanical spinning LiDAR and [0m; 211:2m] for
> the forward-facing LiDAR. Other configuration parameters
> are shared in both models. The detection range along the
> y-axis is set to [51:2m; 51:2m] and [2m; 4m]

and also The voxel size is set to 

 `VOXEL_SIZE: [0.05, 0.05, 0.1]`

not the paper's
`(0:1m; 0:1m; 0:15m).`

try different NMS thresholds and score thresh in POST_PROCESSING  in  `pv_rcnn.yaml`

On my computer, the result looks good, but it doesn't match the result on the paper.

1uk4s, 2022-09-21:
What i missed was the data augmentation. This has greatly improved the results of the not so common classes. 
But thanks for the hint on the NMS threshold i will try that

Mingqj, 2023-10-28:
@Leozyc-waseda Hi, could you share your evaluation codes on Pandaset? Thanks!