open-mmlab/OpenPCDet #810: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)

daeunni, 2022-02-16:
Hi, 
Why this kind of Numba error occurred?
This warning occurred in valid prediction part. 

[image]

I'd really appreciate it if someone helped me.
Best,

jihanyang, 2022-02-17:
Sorry, I never meet this numba error. Hope some other researchers can help you.

LouisSF, 2022-02-18:
I think this is due to type inference issues, you may find more details on the official [Numba deprecation page](https://numba.pydata.org/numba-doc/latest/reference/deprecation.html)

A workaround I used to suppress these warnings was to use the following code:
```
from numba.core.errors import (NumbaDeprecationWarning, 
                                    NumbaPendingDeprecationWarning,
                                    NumbaPerformanceWarning)
import warnings
warnings.simplefilter('ignore', category=NumbaDeprecationWarning)
warnings.simplefilter('ignore', category=NumbaPendingDeprecationWarning)
warnings.simplefilter('ignore', category=NumbaPerformanceWarning)
```

You have to be careful where you place the following block of code, generally in the same python file as the called numba functions.

daeunni, 2022-02-21:
> I think this is due to type inference issues, you may find more details on the official [Numba deprecation page](https://numba.pydata.org/numba-doc/latest/reference/deprecation.html)
>
> A workaround I [...]

Thanks for reply :)

closed 2022-02-21 by daeunni