open-mmlab/OpenPCDet #811: no Docker image for RTX 30 series GPU  and CUDA >=11.1

makeradventures, 2022-02-16:
I don't know if I am doing something obviously wrong but I have had massive struggles with this repository! Basically due to my GPU being a 3080, the common way of using cuda 10.0 or 10.1 or 10.2 does not work for me. Since my graphics card only supports CUDA>=11.1  When directly installing OpenPCDet the recommended way, I always run into some for me unsolvable undefined symbols errors (#755) The provided Docker image also does not work for me due to it being incompatible with the compute capabilities 8.6! I would really appreciate if somebody with an RTX30 series graphics card could describe either the used python/pytroch/gcc/g++/nvcc/cuda/spconv version or maybe share a docker file!

Help is appreciated!

jihanyang, 2022-02-17:
I guess you can try to simply modify some lines in the docker file such as https://github.com/open-mmlab/OpenPCDet/blob/830fba91605f236bc2183f83ae1e986df35200d7/docker/Dockerfile#L1
and https://github.com/open-mmlab/OpenPCDet/blob/6a1495ca610186d3b9d5142a4a6506b8dc78bce1/docker/Dockerfile#L55 to build on your needed cuda version.

makeradventures, 2022-02-17:
Thank you very much for your quick reply!!!

I did the suggested changes and replaced the lines you suggest as well as the PyTorch line with:
```
FROM nvidia/cuda:11.3.0-cudnn8-devel-ubuntu18.04
```
```
RUN pip install --user torch==1.6 torchvision==0.7.0 -f https://download.pytorch.org/whl/cu113/torch_stable.html
```
```
RUN pip install spconv-cu113
```
How ever I still encounter an error this time it explicitly says RTX3080 not supported for my pytorch version:
```
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
/root/.local/lib/python3.7/site-packages/torch/cuda/__init__.py:125: UserWarning: 
NVIDIA GeForce RTX 3080 Laptop GPU with CUDA capability sm_86 is not compatible with the current PyTorch installation.
The current PyTorch install supports CUDA capabilities sm_37 sm_50 sm_60 sm_70 sm_75.
If you want to use the NVIDIA GeForce RTX 3080 Laptop GPU GPU with PyTorch, please check the instructions at https://pytorch.org/get-started/locally/

  warnings.warn(incompatible_device_warn.format(device_name, capability, " ".join(arch_list), device_name))
Traceback (most recent call last):
  File "setup.py", line 116, in <module>
    'src/sampling_gpu.cu',
  File "/usr/local/lib/python3.7/dist-packages/setuptools/__init__.py", line 155, in setup
    return distutils.core.setup(**attrs)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/core.py", line 148, in setup
    return run_commands(dist)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/core.py", line 163, in run_commands
    dist.run_commands()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/develop.py", line 114, in install_for_development
    self.run_command('build_ext')
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/root/.local/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 649, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "/root/.local/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 464, in unix_wrap_ninja_compile
    cuda_post_cflags = unix_cuda_flags(cuda_post_cflags)
  File "/root/.local/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 372, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/root/.local/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1414, in _get_cuda_arch_flags
    raise ValueError("Unknown CUDA arch ({}) or GPU not supported".format(arch))
```

makeradventures, 2022-02-17:
Solved! Thank you very much @jihanyang for your help!

I am new to docker and understanding it took me quite a bit, so in case it helps someone, here is exactly what I did:

I had to change the pytorch line again to :
```
RUN pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html
```

because I encountered some version conflicts with numpy after adding opencv and open3d to my docker image I also added the following line:
```
RUN pip3 install numpy==1.20 #to solve numpy dependency issues with opencv!
```

So here is my working docker file:
```
FROM nvidia/cuda:11.3.0-cudnn8-devel-ubuntu18.04

RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections

# Install basics
RUN apt-get update -y \
    && apt-get install build-essential \
    && apt-get install -y apt-utils git curl ca-certificates bzip2 tree htop wget \
    && apt-get install -y libglib2.0-0 libsm6 libxext6 libxrender-dev bmon iotop g++ python3.7 python3.7-dev python3.7-distutils

# Install cmake v3.13.2
RUN apt-get purge -y cmake && \
    mkdir /root/temp && \
    cd /root/temp && \
    wget https://github.com/Kitware/CMake/releases/download/v3.13.2/cmake-3.13.2.tar.gz && \
    tar -xzvf cmake-3.13.2.tar.gz && \
    cd cmake-3.13.2 && \
    bash ./bootstrap && \
    make && \
    make install && \
    cmake --version && \
    rm -rf /root/temp

# Install python
RUN ln -sv /usr/bin/python3.7 /usr/bin/python
RUN wget https://bootstrap.pypa.io/get-pip.py && \
	python get-pip.py && \
	rm get-pip.py

# Install python packages
RUN PIP_INSTALL="python -m pip --no-cache-dir install" && \
    $PIP_INSTALL numpy==1.19.3 llvmlite numba 

# Install torch and torchvision
# See https://pytorch.org/ for other options if you use a different version of CUDA
RUN pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html

RUN pip3 install numpy==1.20 #to solve numpy dependency issues with opencv!

# Install python packages
RUN PIP_INSTALL="python -m pip --no-cache-dir install" && \
    $PIP_INSTALL tensorboardX easydict pyyaml scikit-image tqdm SharedArray six

WORKDIR /root

# Install Boost geometry
RUN wget https://jaist.dl.sourceforge.net/project/boost/boost/1.68.0/boost_1_68_0.tar.gz && \
    tar xzvf boost_1_68_0.tar.gz && \
    cp -r ./boost_1_68_0/boost /usr/include && \
    rm -rf ./boost_1_68_0 && \
    rm -rf ./boost_1_68_0.tar.gz 

# A weired problem that hasn't been solved yet
RUN pip uninstall -y SharedArray && \
    pip install SharedArray

RUN pip install spconv-cu113

``` 
And if you want to run the demo add this to the end of your Dockerfile:
```

#just for running the Demo(without visualization) we need to add:
RUN apt-get update && apt-get install -y python3-opencv
RUN pip install opencv-python
RUN pip install open3d

```

I replaced the original docker file with my file and ran:
```
docker build ./ -t openpcdet-docker
```
Then I ran the Docker image and mounted a volume to it, which I opened on my host computer and put the OpenPCDet git folder in it (located in /var/lib/docker/volumes/data/_data):
```
docker run  -i -t -v data:/root/data --gpus all openpcdet-docker

```
Then I built OpenPCDet inside the docker image and inside the mounted volume, so it stays built!
I now have to export the build location to PYTHONPATH every time I run the docker image to use it:
```
export PYTHONPATH=$HOME/data/OpenPCDet/:$PYTHONPATH
```
Then I made sure that the data folder structure was correct and that it had all the reqired data in it and that the model was in the right place and then I could run the Demo successfully but unfortunately I don't know if docker supports a visualization?
```
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path ~/data/OpenPCDet/data
```

closed 2022-02-17 by makeradventures

jihanyang, 2022-02-17:
@makeradventures Great!

vikasnr, 2022-12-05:
Thanks for sharing a detailed explanation.

I faced issue with pcdet while running demo.py code. Installation pcdet was required. Used below command in OpenPCDet root folder to install pcdet

`python setup.py develop`

Coming to my question.
How to run docker without interactive shell?  I don't need visualization so currently using the docker file you shared earlier.

I have added below code at the end of without visualization Dockerfile to install pcdet

`
#to permanently export the location of OpenPCdet (installed in a persistent volume) to PYTHONPATH 
ENV PYTHONPATH="/root/data/OpenPCDet/:$PYTHONPATH"

# copy the contents from local to docker 
WORKDIR /
ADD . /app

# To install pcdet
WORKDIR /app
RUN python setup.py develop

# To execute demo code
WORKDIR /app/tools
CMD ["python", "demo_analysis_waymo.py", "--cfg_file","cfgs/waymo_models/pv_rcnn.yaml", "--ckpt","../H_pvrcnn_checkpoint_epoch_30.pth", "--data_path", "../DataAnalysis"]
`
However its failing at  `RUN python setup.py develop` 

`
(openpc) vikasnr@lambda-dual:~/calibration/code/new/OpenPCDet$ docker build ./ -t openpcdet-docker
Sending build context to Docker daemon  425.1MB
Step 1/24 : FROM nvidia/cuda:11.3.0-cudnn8-devel-ubuntu18.04
 ---> 0aa6703ebcf1
Step 2/24 : RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
 ---> Using cache
 ---> 6425b0112b31
Step 3/24 : RUN apt-get update -y     && apt-get install build-essential     && apt-get install -y apt-utils git curl ca-certificates bzip2 tree htop wget     && apt-get install -y libglib2.0-0 libsm6 libxext6 libxrender-dev bmon iotop g++ python3.7 python3.7-dev python3.7-distutils
 ---> Using cache
 ---> 54b58526c4d2
Step 4/24 : RUN apt-get purge -y cmake &&     mkdir /root/temp &&     cd /root/temp &&     wget https://github.com/Kitware/CMake/releases/download/v3.13.2/cmake-3.13.2.tar.gz &&     tar -xzvf cmake-3.13.2.tar.gz &&     cd cmake-3.13.2 &&     bash ./bootstrap &&     make &&     make install &&     cmake --version &&     rm -rf /root/temp
 ---> Using cache
 ---> 0f6a08f59f74
Step 5/24 : RUN ln -sv /usr/bin/python3.7 /usr/bin/python
 ---> Using cache
 ---> 919132b1ca87
Step 6/24 : RUN wget https://bootstrap.pypa.io/get-pip.py &&    python get-pip.py &&    rm get-pip.py
 ---> Using cache
 ---> 7981e8faed37
Step 7/24 : RUN PIP_INSTALL="python -m pip --no-cache-dir install" &&     $PIP_INSTALL numpy==1.19.3 llvmlite numba
 ---> Using cache
 ---> 6a465c7e3c92
Step 8/24 : RUN pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html
 ---> Using cache
 ---> a73a38bf24f1
Step 9/24 : RUN pip3 install numpy==1.20 #to solve numpy dependency issues with opencv!
 ---> Using cache
 ---> 25da8f105c3d
Step 10/24 : RUN PIP_INSTALL="python -m pip --no-cache-dir install" &&     $PIP_INSTALL tensorboardX easydict pyyaml scikit-image tqdm SharedArray six
 ---> Using cache
 ---> ce96f9b67669
Step 11/24 : WORKDIR /root
 ---> Using cache
 ---> 316667c251f3
Step 12/24 : RUN wget https://jaist.dl.sourceforge.net/project/boost/boost/1.68.0/boost_1_68_0.tar.gz &&     tar xzvf boost_1_68_0.tar.gz &&     cp -r ./boost_1_68_0/boost /usr/include &&     rm -rf ./boost_1_68_0 &&     rm -rf ./boost_1_68_0.tar.gz
 ---> Using cache
 ---> 37fdaa7caed9
Step 13/24 : RUN pip uninstall -y SharedArray &&     pip install SharedArray
 ---> Using cache
 ---> a67c529f257e
[... 70 lines cut ...]
    super().run_command(command)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 84, in run
    _build_ext.run(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
    self.build_extensions()
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 468, in build_extensions
    self._build_extensions_serial()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 494, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 246, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 556, in build_extension
    depends=ext.depends,
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 551, in unix_wrap_ninja_compile
    cuda_post_cflags = unix_cuda_flags(cuda_post_cflags)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 450, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 1606, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range
`

vikasnr, 2022-12-05:
Thanks for sharing a detailed explanation.

I faced issue with pcdet while running demo.py code. Installation pcdet was required. Used below command in OpenPCDet root folder to install pcdet

`python setup.py develop`

Coming to my question.
How to run docker without interactive shell?  I don't need visualization so currently using the docker file you shared earlier.

I have added below code at the end of without visualization Dockerfile to install pcdet
```
...... 
 # just for running the Demo we need to add:
RUN apt-get update && apt-get install -y python3-opencv
RUN pip install opencv-python
RUN pip install open3d

# To permanently export the location of OpenPCdet (installed in a persistent volume) to PYTHONPATH 
ENV PYTHONPATH="/root/data/OpenPCDet/:$PYTHONPATH"

 # copy OpenPCDet code from local to docker 
WORKDIR /
ADD . /app

# To install pcdet
WORKDIR /app
RUN python setup.py develop

# To execute demo code
WORKDIR /app/tools
CMD ["python", "demo_analysis_waymo.py", "--cfg_file","cfgs/waymo_models/pv_rcnn.yaml", "--ckpt","../H_pvrcnn_checkpoint_epoch_30.pth", "--data_path", "../DataAnalysis"]`

```
However its failing at  `RUN python setup.py develop` 

```
(openpc) vikasnr@lambda-dual:~/calibration/code/new/OpenPCDet$ docker build ./ -t openpcdet-docker
Sending build context to Docker daemon  425.1MB
Step 1/24 : FROM nvidia/cuda:11.3.0-cudnn8-devel-ubuntu18.04
 ---> 0aa6703ebcf1
Step 2/24 : RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
 ---> Using cache
 ---> 6425b0112b31
Step 3/24 : RUN apt-get update -y     && apt-get install build-essential     && apt-get install -y apt-utils git curl ca-certificates bzip2 tree htop wget     && apt-get install -y libglib2.0-0 libsm6 libxext6 libxrender-dev bmon iotop g++ python3.7 python3.7-dev python3.7-distutils
 ---> Using cache
 ---> 54b58526c4d2
Step 4/24 : RUN apt-get purge -y cmake &&     mkdir /root/temp &&     cd /root/temp &&     wget https://github.com/Kitware/CMake/releases/download/v3.13.2/cmake-3.13.2.tar.gz &&     tar -xzvf cmake-3.13.2.tar.gz &&     cd cmake-3.13.2 &&     bash ./bootstrap &&     make &&     make install &&     cmake --version &&     rm -rf /root/temp
 ---> Using cache
 ---> 0f6a08f59f74
Step 5/24 : RUN ln -sv /usr/bin/python3.7 /usr/bin/python
 ---> Using cache
 ---> 919132b1ca87
Step 6/24 : RUN wget https://bootstrap.pypa.io/get-pip.py &&    python get-pip.py &&    rm get-pip.py
 ---> Using cache
 ---> 7981e8faed37
Step 7/24 : RUN PIP_INSTALL="python -m pip --no-cache-dir install" &&     $PIP_INSTALL numpy==1.19.3 llvmlite numba
 ---> Using cache
 ---> 6a465c7e3c92
Step 8/24 : RUN pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html
 ---> Using cache
 ---> a73a38bf24f1
Step 9/24 : RUN pip3 install numpy==1.20 #to solve numpy dependency issues with opencv!
 ---> Using cache
 ---> 25da8f105c3d
Step 10/24 : RUN PIP_INSTALL="python -m pip --no-cache-dir install" &&     $PIP_INSTALL tensorboardX easydict pyyaml scikit-image tqdm SharedArray six
 ---> Using cache
 ---> ce96f9b67669
Step 11/24 : WORKDIR /root
 ---> Using cache
 ---> 316667c251f3
[... 78 lines cut ...]
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 84, in run
    _build_ext.run(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
    self.build_extensions()
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 468, in build_extensions
    self._build_extensions_serial()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 494, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 246, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 556, in build_extension
    depends=ext.depends,
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 551, in unix_wrap_ninja_compile
    cuda_post_cflags = unix_cuda_flags(cuda_post_cflags)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 450, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 1606, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range

```
Note : I'm running this on a GPU machine.