open-mmlab/OpenPCDet #812: Is model.train() in each iteration necessary?

gfkri, 2022-02-17:
Hi,

thx for your great work. I am currently trying to keep certain parameters frozen whilst finetuning others. Thus, I need to set the BatchNorm modules within this part of the network to eval. 

Thereby, I realized that model.train() is called every iteration (additionally to the call in train.py):
https://github.com/open-mmlab/OpenPCDet/blob/6a1495ca610186d3b9d5142a4a6506b8dc78bce1/tools/train_utils/train_utils.py#L44

Is there a certain reason for that or rather a precaution? I dont want to miss something here.
Thank you very much in in advance, 
BR Georg

jihanyang, 2022-02-18:
To the best of my knowledge, this is a precaution. You can just put it before training and do a quick verification on small datasets like KITTI.

sshaoshuai, 2022-02-20:
Yes, it is a precaution in case some lines call model.eval() before next iteration, since I think the time cost of this operation is negligible.

gfkri, 2022-02-20:
Thank you for your replies, I disabled it, did some sanity checks, looks good.

closed 2022-02-20 by sshaoshuai