open-mmlab/OpenPCDet #814: A question about the objects confidence of the output.

mmmmaomao, 2022-02-19:
Hello!

I only detect vehicle, pedestrian and cyclist in Kitti. If count the background class, there are four classes. Then the score corresponding to the largest class generated by softmax in the classification stage should not be less than 0.25, but during visualization, I found that the score corresponding to some bbox is 0.12. So is this score the probability of softmax's output? 

And is the thresh in the post-processing filtering the score of the softmax?
`
POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False
`
Thank you！

mmmmaomao, 2022-02-19:
[image]
For example, the score of one car in this result is 0.11. Is this score processed by raw softmax score?

jihanyang, 2022-02-20:
No. That's score after sigmoid.

closed 2022-02-20 by mmmmaomao

Leozyc-waseda, 2022-05-17:
[image]
If this is the output of the sigmoid function, what's the meaning in these numbers?

It doesn't look like a binary classification problem.