open-mmlab/OpenPCDet #816: CUDA error in boxes_iou3d_gpu

Guoxs, 2022-02-21:
I trained PointPillars on KITTI datasets using multi-gpu on single machine using this command:
`bash ./scripts/dist_train.sh 4 --cfg_file cfgs/kitti_models/pointpillar.yaml`

when I finished training and started evaluation, error happened. 
CUDA version 10.2, ubuntu 18.04, pcdet+0.5.2+70857b8

`2022-02-21 10:03:14,113   INFO  Loading KITTI dataset
2022-02-21 10:03:14,338   INFO  Total samples for KITTI dataset: 3769
2022-02-21 10:03:14,342   INFO  ==> Loading parameters from checkpoint /home/guoxs3/project/OpenPCDet/output/kitti_mod
els/pointpillar/default/ckpt/checkpoint_epoch_80.pth to CPU
2022-02-21 10:03:14,386   INFO  ==> Checkpoint trained from version: pcdet+0.5.2+70857b8
2022-02-21 10:03:14,415   INFO  ==> Done (loaded 127/127)
2022-02-21 10:03:14,417   INFO  *************** EPOCH 80 EVALUATION *****************
eval:   0%|                                                                                   | 0/236 [00:00<?, ?it/s]
Error!
Error!
Error!
Error!
Traceback (most recent call last):
  File "train.py", line 202, in <module>
    main()
  File "train.py", line 195, in main
    dist_test=dist_train
  File "/home/guoxs3/project/OpenPCDet/tools/test.py", line 122, in repeat_eval_ckpt
    result_dir=cur_result_dir, save_to_file=args.save_to_file
  File "/home/guoxs3/project/OpenPCDet/tools/eval_utils/eval_utils.py", line 57, in eval_one_epoch
    pred_dicts, ret_dict = model(batch_dict)
  File "/home/guoxs3/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call
_impl
    return forward_call(*input, **kwargs)
  File "/home/guoxs3/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/parallel/distributed.py", line 886, in 
forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/guoxs3/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call
_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/models/detectors/pointpillar.py", line 21, in forward
    pred_dicts, recall_dicts = self.post_processing(batch_dict)
  File "../pcdet/models/detectors/detector3d_template.py", line 274, in post_processing
    thresh_list=post_process_cfg.RECALL_THRESH_LIST
  File "../pcdet/models/detectors/detector3d_template.py", line 308, in generate_recall_record
    iou3d_rcnn = iou3d_nms_utils.boxes_iou3d_gpu(box_preds[:, 0:7], cur_gt[:, 0:7])
  File "../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 69, in boxes_iou3d_gpu
    max_of_min = torch.max(boxes_a_height_min, boxes_b_height_min)
RuntimeError: CUDA error: invalid device function
`

MartinHahner, 2022-02-21:
Which GPU do you use? Is it an RTX 30..?

Maybe this thread helps:
https://github.com/open-mmlab/OpenPCDet/issues/273

Guoxs, 2022-02-21:
my GPUs are RTX 2080. I have install spconv-cu102==2.1.12

Guoxs, 2022-02-21:
> Which GPU do you use? Is it an RTX 30..?
> 
> Maybe this thread helps: #273

 I rebuild pcdet after update cumm-cu102  to 0.2.8 and spconv-cu102 to 2.1.21, but this problem remains.

Guoxs, 2022-02-21:
when I run `python demo.py` , "Error!" also occurs, I can not find where it was printed.

[image]

MartinHahner, 2022-02-21:
> when I run `python demo.py` , "Error!" also occurs, I can not find where it was printed.

https://github.com/open-mmlab/OpenPCDet/blob/e1bfcecbf4966fe417bf56fa60d6e177a1909724/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp#L133

https://github.com/open-mmlab/OpenPCDet/blob/e1bfcecbf4966fe417bf56fa60d6e177a1909724/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp#L183

Guoxs, 2022-02-22:
Thank you! So how can I fix this? @MartinHahner

MartinHahner, 2022-02-22:
The `Error!` printouts indicate that CUDA is not working properly on your machine. 
Try to use a different / newer CUDA version and rebuild pcdet.

Guoxs, 2022-02-22:
I found my CUDA version (10.0) is not matched with nvidia driver (10.2), and when I update CUDA to 10.2 and rebuild pcdet, it was solved. Thank you for your kindness help!

closed 2022-02-22 by Guoxs