open-mmlab/OpenPCDet #817: What is the effect of 'adam_onecycle' optimizer ?

daeunni, 2022-02-22:
Hi :)
Thx for your nice work. 

btx, I wonder the effect of '**adam_onecyle**' optimizer. (used in pointpillar model's config)

What part of this adam_onecycle method helps our learning? 
And why did you use adam_onecycle as a default? (origin pointpillar paper used just adam optimizer) 

Best, 
Daeun Lee

Gsivap, 2022-03-10:
I think it helps in converging faster and allows to use higher learning rates. Refer section 4.1 in this paper: https://arxiv.org/pdf/1803.09820.pdf for more details.

daeunni, 2022-03-15:
Thanks ! 

I experienced AP degradation when I used just '**Adam**' optimizer. (not Adam_onecycle optimizer).
it seems that learning didnt converge and fluctuate.. ;(

is there any reason why performance is poor? 
and is this common issue ?

Gsivap, 2022-03-15:
I think when you just use adam the learning rate starts with the learning rate specified in the model config yaml file and then decays according to this:

 https://github.com/open-mmlab/OpenPCDet/blob/0642cf06d0fd84f50cc4c6c01ea28edbc72ea810/tools/train_utils/optimization/__init__.py#L40
 
Whereas if you use a one cycle lr scheduler, it ramps up to the learning rate specified in the model config yaml file and then ramps down.

Maybe try using a smaller learning rate if you are not using one_cycle lr scheduler?

daeunni, 2022-03-23:
really helpful answer. thanks XD

closed 2022-03-23 by daeunni