open-mmlab/OpenPCDet #818: anchor_rotations in config.yaml

csj3684, 2022-02-22:
Hi, I really appreciate your works and I was gonna do train with my own dataset.

And I saw that 'anchor_rotations'  is 1.57, which is pi / 2 (rad), I think
why do all config have1.57 in anchor_rotations ?

Only for kitti dataset ? 
And I saw that The yaw value of kitti label_2 files from 'http://www.cvlibs.net/datasets/kitti/raw_data.php' is not correct.
It should + or - pi / 2 ( actually I don't remember + or - )

Then, when I train with my own dataset without calib set, Do I have to set it '0' ?

Thank you.

safaATcurrus, 2022-02-23:
The rotations are between +/- 2 pi. It was written like that on the description of the labels. 2 * pi radians = 360 degrees
so in fact it makes sense. pi/2 makes only 90 degrees and some objects might be turning their back completely to the camera, in that case you can not assess that.

I dont have a exact answer for calibration but I guess some techniques can not be trained without calibration.  for the techniques that train without calibration, I guess you don't need to do anything. It will completely ignore calibration folder. 
This settings are stated in yaml files. (data & model)

csj3684, 2022-02-23:
I really Thank you.

If I understood correctly,

In world coordinates system, if the yaw value of objects can be any value,
Do I have to append more expected yaw values to anchor rotations ??

Again, I really appreciate your works.

safaATcurrus, 2022-02-25:
I dont know what is yaw value. 

Well in fact you might be right about the maximum minimum angle between +pi and -pi
because -1.5pi is equal to 0.5pi angle.

You can check the dataset and if the maximum value for a column is pi (not 2 pi) then you are probably right.
the angle is between +pi and -pi

considering that pi=3.14159265...

csj3684, 2022-03-04:
Question for OpenPCDet

( I wasn't able to find any mail to send so I talk to you in here, sorry
to bother you.)

Hello

I'm seungjin choi in KOR.

I actually don't speak good english. but I have one to ask really..

and again, I really appreciate your works and I'm trying to train my own
dataset with yours.

I can train it but I absolutely think it's wrong

I don't know what the process is. in my situation,

I don't use calibration file and train with only lidar and label file.

in HERE, I don't know how to deal with this label file format,

in this case, I use Voxel rcnn which use Kitti dataset format, and

right after calling 'build_dataloader' in train.py

it must be class 'KittiDataset' and after loading or some processing that I
defined for data

About my some data in value 'infos' ( kitti_infos ),

my gt_boxes_lidar has format [center_xs, center_ys, center_zs,
dim_xs(=lengths), dim_ys(widths), dim_zs(heights), yaws]
 in lidar coordinate system (=world coordinate system)

actually my data start with camera coordinate system, it finally is
converted into lidar coordinate system (world coordinate system)

in this process, anything that I misunderstood ?

Right after when __getitem__ in Kittidataset is called, what format should
'gt_boxes_lidar' in  infos have ?

really Thank you for your works again.

csj3684, 2022-03-05:
I've solved my problem. I transformed not correctly. Thank you.

closed 2022-04-19 by github-actions[bot]