open-mmlab/OpenPCDet #819: CADDN demo run problem

safaATcurrus, 2022-02-22:
I am getting this error 
File "/media/safa/nvme/pycharmProjects/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 195, in downsample_depth_map
    image=data_dict['depth_maps'],
KeyError: 'depth_maps'

When I run 

python demo.py --cfg_file cfgs/kitti_models/CaDDN.yaml --ckpt caddn_pcdet.pth --data_path ../data/kitti/training/velodyne/000008.bin

I have downloaded depth_2 folder near calib and image_2 folder in train dataset. 

What am I doing wrong?

Regards

Safa

MartinHahner, 2022-02-22:
Have you generated the .pkl files as described [here](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#dataset-preparation) before after you downloaded the depth maps?

Maybe you need to re-generate them if you downloaded the depth maps afterwards. 
Otherwise, perhaps @codyreading can help.

safaATcurrus, 2022-02-22:
Thank you @MartinHahner  for the reply. Stating pkl file generation, I guess you mean running this command below:

`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
`
I had done before I downloaded. I did after I downloaded, per your advice. Still I am getting the same error.

```
Traceback (most recent call last):
  File "demo.py", line 113, in <module>
    main()
  File "demo.py", line 95, in main
    for idx, data_dict in enumerate(demo_dataset):
  File "demo.py", line 60, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/media/safa/nvme/pycharmProjects/OpenPCDet/pcdet/datasets/dataset.py", line 148, in prepare_data
    data_dict = self.data_processor.forward(
  File "/media/safa/nvme/pycharmProjects/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 213, in forward
    data_dict = cur_processor(data_dict=data_dict)
  File "/media/safa/nvme/pycharmProjects/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 195, in downsample_depth_map
    image=data_dict['depth_maps'],
KeyError: 'depth_maps'
```

I am open to other suggestions as well.

Regards

closed 2022-04-10 by github-actions[bot]

Mi97ch, 2022-11-28:
Hi, do someone have the updated Demo code to solve this issue?