open-mmlab/OpenPCDet #820: How 头

VsionQing, 2022-02-23:


VsionQing, 2022-02-23:
2022-02-23 09:30:29,155   INFO  **********************Start logging**********************
2022-02-23 09:30:29,155   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-02-23 09:30:29,156   INFO  cfg_file         cfgs/kitti_models/pv_rcnn.yaml
2022-02-23 09:30:29,156   INFO  batch_size       1
2022-02-23 09:30:29,156   INFO  epochs           80
2022-02-23 09:30:29,156   INFO  workers          1
2022-02-23 09:30:29,156   INFO  extra_tag        default
2022-02-23 09:30:29,156   INFO  ckpt             None
2022-02-23 09:30:29,156   INFO  pretrained_model None
2022-02-23 09:30:29,156   INFO  launcher         none
2022-02-23 09:30:29,157   INFO  tcp_port         18888
2022-02-23 09:30:29,157   INFO  sync_bn          False
2022-02-23 09:30:29,157   INFO  fix_random_seed  False
2022-02-23 09:30:29,157   INFO  ckpt_save_interval 1
2022-02-23 09:30:29,157   INFO  local_rank       0
2022-02-23 09:30:29,157   INFO  max_ckpt_save_num 30
2022-02-23 09:30:29,157   INFO  merge_all_iters_to_one_epoch False
2022-02-23 09:30:29,157   INFO  set_cfgs         None
2022-02-23 09:30:29,157   INFO  max_waiting_mins 0
2022-02-23 09:30:29,157   INFO  start_epoch      0
2022-02-23 09:30:29,158   INFO  num_epochs_to_eval 0
2022-02-23 09:30:29,158   INFO  save_to_file     False
2022-02-23 09:30:29,158   INFO  cfg.ROOT_DIR: E:\PythonFile\OpenPCDet-master1
2022-02-23 09:30:29,158   INFO  cfg.LOCAL_RANK: 0
2022-02-23 09:30:29,158   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2022-02-23 09:30:29,158   INFO  
cfg.DATA_CONFIG = edict()
2022-02-23 09:30:29,158   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2022-02-23 09:30:29,158   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2022-02-23 09:30:29,158   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2022-02-23 09:30:29,158   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-02-23 09:30:29,158   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-02-23 09:30:29,158   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-02-23 09:30:29,159   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-02-23 09:30:29,159   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-02-23 09:30:29,159   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-02-23 09:30:29,159   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-02-23 09:30:29,160   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-02-23 09:30:29,160   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2022-02-23 09:30:29,160   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2022-02-23 09:30:29,160   INFO  
cfg.MODEL = edict()
2022-02-23 09:30:29,160   INFO  cfg.MODEL.NAME: PVRCNN
2022-02-23 09:30:29,160   INFO  
cfg.MODEL.VFE = edict()
2022-02-23 09:30:29,160   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-02-23 09:30:29,160   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-02-23 09:30:29,160   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-02-23 09:30:29,161   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2022-02-23 09:30:29,161   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2022-02-23 09:30:29,161   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2022-02-23 09:30:29,161   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2022-02-23 09:30:29,161   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2022-02-23 09:30:29,161   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2022-02-23 09:30:29,161   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
2022-02-23 09:30:29,161   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [128, 256]
[... 489 lines cut ...]
      (6): ReLU()
    )
    (cls_layers): Sequential(
      (0): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
      (1): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (2): ReLU()
      (3): Dropout(p=0.3, inplace=False)
      (4): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
[... cut ...]
      (1): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (2): ReLU()
      (3): Dropout(p=0.3, inplace=False)
      (4): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
      (5): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (6): ReLU()
      (7): Conv1d(256, 7, kernel_size=(1,), stride=(1,))
    )
  )
)
2022-02-23 09:30:30,579   INFO  **********************Start training kitti_models/pv_rcnn(default)**********************

MartinHahner, 2022-02-23:
Maybe it is related to [these]( https://github.com/open-mmlab/OpenPCDet/pull/815) changes?

VsionQing, 2022-02-23:
收到谢谢

VsionQing, 2022-02-23:
> 
I downloaded the latest version, but I don't know how to use sh file to find TCP port under windows. How to solve it?
And how to test whether the TCP port works

MartinHahner, 2022-02-23:
Aha, maybe Windows is then the issue here. I am not sure if this codebase works on Windows at all.

VsionQing, 2022-02-23:
> Aha, maybe Windows is then the issue here. I am not sure if this codebase works on Windows at all.

I have a new guess. Is it possible that there is something wrong with my data set.
In addition, how to check whether your TCP port is correct?

VsionQing, 2022-02-25:
> Aha, maybe Windows is then the issue here. I am not sure if this codebase works on Windows at all.

Then,i try to train the model pointpillar ,then i success.
So,can you tell me the direct Bev detection method and PV with the following pointpillar_ What are the differences of RCNN? Or do you have any idea what is the most likely cause of training failure?
In addition, I tested that pointrcnn can not run. I think it may be due to the configuration of 3dbackbone and so on

closed 2022-04-11 by github-actions[bot]