open-mmlab/OpenPCDet #826: AttributeError: 'EasyDict' object has no attribute 'DATASET'

elegentbamboo, 2022-02-25:
(HDMapNet) lyl@lyl:/Project/pycharm_project/OpenPCDet-master/tools$ python train.py  --cfg_file cfgs/kitti_models/pointrcnn.yaml  --batch_size 4 --epochs 60
2022-02-25 11:29:03,324   INFO  **********************Start logging**********************
2022-02-25 11:29:03,324   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-02-25 11:29:03,324   INFO  cfg_file         cfgs/kitti_models/pointrcnn.yaml
2022-02-25 11:29:03,324   INFO  batch_size       4
2022-02-25 11:29:03,324   INFO  epochs           60
2022-02-25 11:29:03,324   INFO  workers          4
2022-02-25 11:29:03,324   INFO  extra_tag        default
2022-02-25 11:29:03,324   INFO  ckpt             None
2022-02-25 11:29:03,324   INFO  pretrained_model None
2022-02-25 11:29:03,324   INFO  launcher         none
2022-02-25 11:29:03,324   INFO  tcp_port         18888
2022-02-25 11:29:03,324   INFO  sync_bn          False
2022-02-25 11:29:03,324   INFO  fix_random_seed  False
2022-02-25 11:29:03,324   INFO  ckpt_save_interval 1
2022-02-25 11:29:03,324   INFO  local_rank       0
2022-02-25 11:29:03,324   INFO  max_ckpt_save_num 30
2022-02-25 11:29:03,324   INFO  merge_all_iters_to_one_epoch False
2022-02-25 11:29:03,324   INFO  set_cfgs         None
2022-02-25 11:29:03,324   INFO  max_waiting_mins 0
2022-02-25 11:29:03,324   INFO  start_epoch      0
2022-02-25 11:29:03,324   INFO  num_epochs_to_eval 0
2022-02-25 11:29:03,324   INFO  save_to_file     False
2022-02-25 11:29:03,324   INFO  cfg.ROOT_DIR: /Project/pycharm_project/OpenPCDet-master
2022-02-25 11:29:03,324   INFO  cfg.LOCAL_RANK: 0
2022-02-25 11:29:03,324   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2022-02-25 11:29:03,324   INFO  
cfg.DATA_CONFIG = edict()
2022-02-25 11:29:03,324   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2022-02-25 11:29:03,324   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'sample_points', 'NUM_POINTS': {'train': 16384, 'test': 16384}}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}]
2022-02-25 11:29:03,324   INFO  
cfg.MODEL = edict()
2022-02-25 11:29:03,324   INFO  cfg.MODEL.NAME: PointRCNN
2022-02-25 11:29:03,324   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.BACKBONE_3D.NAME: PointNet2MSG
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.BACKBONE_3D.SA_CONFIG = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.NPOINTS: [4096, 1024, 256, 64]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.RADIUS: [[0.1, 0.5], [0.5, 1.0], [1.0, 2.0], [2.0, 4.0]]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.NSAMPLE: [[16, 32], [16, 32], [16, 32], [16, 32]]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.MLPS: [[[16, 16, 32], [32, 32, 64]], [[64, 64, 128], [64, 96, 128]], [[128, 196, 256], [128, 196, 256]], [[256, 256, 512], [256, 384, 512]]]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.BACKBONE_3D.FP_MLPS: [[128, 128], [256, 256], [512, 512], [512, 512]]
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.POINT_HEAD = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.NAME: PointHeadBox
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.CLS_FC: [256, 256]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.REG_FC: [256, 256]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.CLASS_AGNOSTIC: False
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.USE_POINT_FEATURES_BEFORE_FUSION: False
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.POINT_HEAD.TARGET_CONFIG = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.TARGET_CONFIG.GT_EXTRA_WIDTH: [0.2, 0.2, 0.2]
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.TARGET_CONFIG.BOX_CODER: PointResidualCoder
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.POINT_HEAD.TARGET_CONFIG.BOX_CODER_CONFIG = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.TARGET_CONFIG.BOX_CODER_CONFIG.use_mean_size: True
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.TARGET_CONFIG.BOX_CODER_CONFIG.mean_size: [[3.9, 1.6, 1.56], [0.8, 0.6, 1.73], [1.76, 0.6, 1.73]]
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.POINT_HEAD.LOSS_CONFIG = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.LOSS_CONFIG.LOSS_REG: WeightedSmoothL1Loss
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.POINT_HEAD.LOSS_CONFIG.LOSS_WEIGHTS = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.LOSS_CONFIG.LOSS_WEIGHTS.point_cls_weight: 1.0
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.LOSS_CONFIG.LOSS_WEIGHTS.point_box_weight: 1.0
2022-02-25 11:29:03,325   INFO  cfg.MODEL.POINT_HEAD.LOSS_CONFIG.LOSS_WEIGHTS.code_weights: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
2022-02-25 11:29:03,325   INFO  
cfg.MODEL.ROI_HEAD = edict()
2022-02-25 11:29:03,325   INFO  cfg.MODEL.ROI_HEAD.NAME: PointRCNNHead
2022-02-25 11:29:03,325   INFO  cfg.MODEL.ROI_HEAD.CLASS_AGNOSTIC: True
[... 74 lines cut ...]
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.WEIGHT_DECAY: 0.01
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.MOMENTUM: 0.9
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.MOMS: [0.95, 0.85]
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.PCT_START: 0.4
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.DIV_FACTOR: 10
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.DECAY_STEP_LIST: [35, 45]
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.LR_DECAY: 0.1
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.LR_CLIP: 1e-07
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.LR_WARMUP: False
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.WARMUP_EPOCH: 1
2022-02-25 11:29:03,327   INFO  cfg.OPTIMIZATION.GRAD_NORM_CLIP: 10
2022-02-25 11:29:03,327   INFO  cfg.TAG: pointrcnn
2022-02-25 11:29:03,327   INFO  cfg.EXP_GROUP_PATH: kitti_models
Traceback (most recent call last):
  File "train.py", line 202, in <module>
    main()
  File "train.py", line 113, in main
    total_epochs=args.epochs
  File "../pcdet/datasets/__init__.py", line 50, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
AttributeError: 'EasyDict' object has no attribute 'DATASET'

how to solve this problem？

my easydict version is the latest 1.9.0

closed 2022-02-26 by jihanyang

elegentbamboo, 2022-03-01:
yes, I have solved it,thanks

Natty-wang, 2022-10-30:
I also encountered this problem,How did you solve it?

Uary-zjtd, 2023-03-14:
I also encountered this problem,How did you solve it?

hediimohamed, 2023-05-15:
anyone got any solution for this ? I have the same probem  'EasyDict' object has no attribute 'BACKUP_DB_INFO' ?

soumithreddyb, 2023-06-06:
@hediimohamed I'm also getting the same error 'EasyDict' object has no attribute 'BACKUP_DB_INFO'. Did you resolve it?

hediimohamed, 2023-06-06:
> @hediimohamed I'm also getting the same error 'EasyDict' object has no attribute 'BACKUP_DB_INFO'. Did you resolve it?

I think that you just forgot to Create data info : try this if you are using the KITTI before training 
python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml

soumithreddyb, 2023-06-06:
> I think that you just forgot to Create data info : try this if you are using the KITTI before training python -m pcdet.datasets.custom.custom_dataset create_custom_infos [...]

I'm using the custom dateset. Anyway, I'll try creating data info and see if it works. Thanks

hediimohamed, 2023-06-06:
> I'm using the custom dateset. Anyway, I'll try creating data info and see if it works. Thanks

python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml

soumithreddyb, 2023-06-07:
> python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml

Thank you @hediimohamed . Now, I'm able to train and test the model on my custom dataset.

3623687277, 2023-11-10:
> > > > > 我也收到同样的错误“EasyDict”对象没有属性“BACKUP_DB_INFO”。你解决了吗？
> > > > 
> > > > 
> > > > 我认为您只是忘记了创建数据信息：如果您在训练python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml 之前使用 KITTI，请尝试此操作
> > > 
> > > 
> > > 我正在使用自定义日期集。无论如何，我将尝试创建数据信息，看看它是否有效。谢谢
> > 
> > 
> > python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml
> 
> 谢谢。现在，我能够在自定义数据集上训练和测试模型。

How did you fix it? Can you tell me?

hediimohamed, 2023-11-10:
> How did you fix it? Can you tell me?

you need to create the data info using : python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml ! run this command if you are using a custom dataset !

muyanqaq, 2023-12-06:
'EasyDict' object has no attribute 'CLASS_NAMES_EACH_HEAD'     how to solve ，when i use centerpoint kitti to train

muyanqaq, 2023-12-06:
python train.py --cfg_file ./cfgs/kitti_models/centerpoint.yaml
2023-12-06 15:34:51,423   INFO  **********************Start logging**********************
2023-12-06 15:34:51,424   INFO  CUDA_VISIBLE_DEVICES=ALL
2023-12-06 15:34:51,424   INFO  cfg_file         ./cfgs/kitti_models/centerpoint.yaml
2023-12-06 15:34:51,424   INFO  batch_size       8
2023-12-06 15:34:51,424   INFO  epochs           80
2023-12-06 15:34:51,424   INFO  workers          8
2023-12-06 15:34:51,424   INFO  extra_tag        default
2023-12-06 15:34:51,424   INFO  ckpt             None
2023-12-06 15:34:51,424   INFO  pretrained_model None
2023-12-06 15:34:51,424   INFO  launcher         none
2023-12-06 15:34:51,424   INFO  tcp_port         18888
2023-12-06 15:34:51,424   INFO  sync_bn          False
2023-12-06 15:34:51,424   INFO  fix_random_seed  False
2023-12-06 15:34:51,424   INFO  ckpt_save_interval 1
2023-12-06 15:34:51,424   INFO  local_rank       0
2023-12-06 15:34:51,424   INFO  max_ckpt_save_num 30
2023-12-06 15:34:51,424   INFO  merge_all_iters_to_one_epoch False
2023-12-06 15:34:51,424   INFO  set_cfgs         None
2023-12-06 15:34:51,424   INFO  max_waiting_mins 0
2023-12-06 15:34:51,424   INFO  start_epoch      0
2023-12-06 15:34:51,424   INFO  save_to_file     False
2023-12-06 15:34:51,424   INFO  cfg.ROOT_DIR: /media/yb/data1/Myan/OpenPCDet
2023-12-06 15:34:51,424   INFO  cfg.LOCAL_RANK: 0
2023-12-06 15:34:51,424   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2023-12-06 15:34:51,424   INFO  ----------- DATA_CONFIG -----------
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2023-12-06 15:34:51,424   INFO  ----------- DATA_SPLIT -----------
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2023-12-06 15:34:51,424   INFO  ----------- INFO_PATH -----------
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2023-12-06 15:34:51,424   INFO  ----------- DATA_AUGMENTOR -----------
2023-12-06 15:34:51,424   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2023-12-06 15:34:51,425   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:20', 'Pedestrian:15', 'Cyclist:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2023-12-06 15:34:51,425   INFO  ----------- POINT_FEATURE_ENCODING -----------
2023-12-06 15:34:51,425   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2023-12-06 15:34:51,425   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2023-12-06 15:34:51,425   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2023-12-06 15:34:51,425   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2023-12-06 15:34:51,425   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2023-12-06 15:34:51,425   INFO  ----------- MODEL -----------
2023-12-06 15:34:51,425   INFO  cfg.MODEL.NAME: CenterPoint
2023-12-06 15:34:51,425   INFO  ----------- VFE -----------
2023-12-06 15:34:51,425   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2023-12-06 15:34:51,425   INFO  ----------- BACKBONE_3D -----------
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2023-12-06 15:34:51,425   INFO  ----------- MAP_TO_BEV -----------
2023-12-06 15:34:51,425   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2023-12-06 15:34:51,425   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2023-12-06 15:34:51,425   INFO  ----------- BACKBONE_2D -----------
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5]
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1]
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [128]
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_2D.UPSAMPLE_STRIDES: [2]
2023-12-06 15:34:51,425   INFO  cfg.MODEL.BACKBONE_2D.NUM_UPSAMPLE_FILTERS: [256]
2023-12-06 15:34:51,425   INFO  ----------- DENSE_HEAD -----------
2023-12-06 15:34:51,425   INFO  cfg.MODEL.DENSE_HEAD.NAME: CenterHead
2023-12-06 15:34:51,425   INFO  cfg.MODEL.DENSE_HEAD.CLASS_AGNOSTIC: False
2023-12-06 15:34:51,425   INFO  cfg.MODEL.DENSE_HEAD.USE_DIRECTION_CLASSIFIER: False
[... cut ...]
    module, model_info_dict = getattr(self, 'build_%s' % module_name)(
  File "/media/yb/data1/Myan/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 128, in build_dense_head
    dense_head_module = dense_heads.__all__[self.model_cfg.DENSE_HEAD.NAME](
  File "/media/yb/data1/Myan/OpenPCDet/pcdet/models/dense_heads/center_head.py", line 64, in __init__
    for cur_class_names in self.model_cfg.CLASS_NAMES_EACH_HEAD:
AttributeError: 'EasyDict' object has no attribute 'CLASS_NAMES_EACH_HEAD'

tyi68or7ee6io, 2024-01-13:
AttributeError: 'EasyDict' object has no attribute 'OPTIMIZATION'这个怎么解决

Zixiu99, 2024-02-24:
> you need to create the data info using : python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml ! run this command if you are using a [...]

I am getting exactly the same error reported, I can confirm that I have generated a valid bin file in gt_database and the DB_INFO_PATH is custom_dbinfos_train.pkl. is there anyone else who is facing the same problem as me?(And also commented out (#filter_by_difficulty: [-1])

kossijj, 2024-04-11:
> AttributeError: 'EasyDict' object has no attribute 'OPTIMIZATION'这个怎么解决

解决了吗

Petros626, 2024-10-11:
https://github.com/open-mmlab/OpenPCDet/issues/826#issuecomment-1051460540 like mentioned here at the beginning of the .yaml file this must stand there:

```yaml
DATASET: 'KittiDataset'
DATA_PATH: '../data/kitti'
```