open-mmlab/OpenPCDet #829: About the support of PV-RCNN++

Orbis36, 2022-02-26:
The corresponding .ymal file can't be found when I want to use it in the KITTI dataset.
So does it mean I can use or modified the 'PV_RCNN.yaml' directly to train the PV_RCNN++?
Or it has not been supported yet by OpenPCDet?

Thanks for any help.

sshaoshuai, 2022-02-27:
The KITTI config of PV-RCNN++ is not supported yet since the codes have been refactored and most experiments are conducted on Waymo. 

We do not tune much parameters in KITTI so one thing you can try is to modify the PV_RCNN++ config of Waymo to KITTI dataset.

MartinHahner, 2022-03-01:
@sshaoshuai: I tried doing that and unfortunately could not succeed.
That is why I opened issue https://github.com/open-mmlab/OpenPCDet/issues/747.

sshaoshuai, 2022-03-01:
Ok, I will find time to look into it recently. 
Note that the performance is not expected to be much better than PV-RCNN in KITTI dataset since the KITTI dataset is small and the performance is also unstable (that's also why I did most experiments on Waymo Dataset). But it should be more memory-efficient than PV-RCNN.

sshaoshuai, 2022-03-12:
I wrote of PV-RCNN++ config for KITTI dataset by slightly modifying the config of WOD. 
(Use AnchorHead, keypoint:2048, and we can train PV-RCNN++ with larger batch_size=4 on 11GB GPUs)

The results and config are as follows:  

Here is a result from my side: 
```
022-03-11 02:13:05,193   INFO  recall_roi_0.3: 0.968314
2022-03-11 02:13:05,193   INFO  recall_rcnn_0.3: 0.968940
2022-03-11 02:13:05,193   INFO  recall_roi_0.5: 0.927925
2022-03-11 02:13:05,193   INFO  recall_rcnn_0.5: 0.931452
2022-03-11 02:13:05,193   INFO  recall_roi_0.7: 0.706752
2022-03-11 02:13:05,193   INFO  recall_rcnn_0.7: 0.750839
2022-03-11 02:13:05,195   INFO  Average predicted number of objects(3769 samples): 9.499
2022-03-11 02:13:30,316   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:90.6572, 89.6768, 89.2032
bev  AP:89.8610, 87.9713, 87.3142
3d   AP:88.8617, 83.6246, 78.6445
aos  AP:90.64, 89.56, 88.98
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:96.1084, 94.7267, 94.2113
bev  AP:92.8120, 90.6260, 88.4745
3d   AP:91.5264, 84.5111, 82.2970
aos  AP:96.08, 94.56, 93.93
Car AP@0.70, 0.50, 0.50:
bbox AP:90.6572, 89.6768, 89.2032
bev  AP:90.5914, 94.7703, 89.2633
3d   AP:90.5914, 94.7173, 89.2181
aos  AP:90.64, 89.56, 88.98
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:96.1084, 94.7267, 94.2113
bev  AP:96.0526, 96.5511, 94.4686
3d   AP:96.0423, 96.1660, 94.3803
aos  AP:96.08, 94.56, 93.93
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:77.6912, 72.7228, 68.9964
bev  AP:66.5163, 62.0834, 56.8533
3d   AP:64.0408, 58.5153, 53.9799
aos  AP:73.21, 68.18, 63.99
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:78.3331, 73.7831, 70.3085
bev  AP:67.2273, 61.2886, 56.8762
3d   AP:63.5890, 57.5630, 52.1814
aos  AP:73.52, 68.67, 64.79
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:77.6912, 72.7228, 68.9964
bev  AP:80.2445, 77.0728, 74.7157
3d   AP:80.2238, 76.9786, 74.6168
aos  AP:73.21, 68.18, 63.99
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:78.3331, 73.7831, 70.3085
bev  AP:82.9161, 79.2863, 75.9079
3d   AP:82.8934, 79.1893, 75.7954
aos  AP:73.52, 68.67, 64.79
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:90.0376, 83.0204, 78.0418
bev  AP:88.5554, 75.9622, 73.4186
3d   AP:87.6325, 74.5702, 70.9378
aos  AP:89.95, 82.57, 77.58
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:95.9116, 84.1527, 80.9133
bev  AP:94.1572, 78.4721, 73.8510
3d   AP:93.0001, 75.5205, 70.9390
aos  AP:95.77, 83.69, 80.36
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:90.0376, 83.0204, 78.0418
bev  AP:88.8730, 80.0251, 75.3880
3d   AP:88.8730, 80.0251, 75.3880
aos  AP:89.95, 82.57, 77.58
Cyclist AP_R40@0.50, 0.25, 0.25:
[... 269 lines cut ...]
            NMS_THRESH: 0.1
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.01
    WEIGHT_DECAY: 0.001
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10
```

Orbis36, 2022-03-13:
Thanks for your work. The output looks similar to mine. So is that mean on the validation set of KITTI, PV-RCNN++ does not perform better than PV-RCNN? @sshaoshuai

sshaoshuai, 2022-03-13:
From this run, you can see that PV-RCNN++ achieves better performance on Cyclist category, and for the vehicle category, previously and I got better and stable performance (84.5%~85% mAP@R11) than PV-RCNN in my initial codebase, but now I do not have much time to tune it on KITTI dataset based on current codebase. 
Also, if you run the same configs for several times, you can also observe that the performance fluctuates during different runs, so I think we can not get stable conclusion on KITTI dataset. 

However, as I mentioned before, for KITTI dataset (also Waymo Dataset), we can still know that PV-RCNN++ consumes much less GPU memory and also is faster than PV-RCNN. (e.g., PV-RCNN++ can be trained with batch_size=4 on 11GB GPUs while PV-RCNN can not). 

I think this should be great for others to build more modifications based on PV-RCNN/PV-RCNN++ frameworks.

Orbis36, 2022-03-18:
Thank you so much for your patience, I think that can solve my question perfectly.

closed 2022-03-18 by Orbis36

MartinHahner, 2022-03-22:
> Here is the config of PV-RCNN++ for KITTI dataset:
>
> ```
> CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
>
> DATA_CONFIG:
>     _BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
>
>
> MODEL:
>     NAME: [...]

When I use this config as is, I get the following error:
```bash
2022-03-22 17:41:20,446   INFO  **********************start of training**********************
epochs:   0% 0/80 [00:00<?, ?it/s]
epochs:   0% 0/80 [00:09<?, ?it/s]
Traceback (most recent call last):
  File "/home/mhahner/repositories/PCDet/tools/train_utils/train_utils.py", line 165, in train_model
    accumulated_iter, n_errors = train_one_epoch(model, optimizer, train_loader, model_func,
  File "/home/mhahner/repositories/PCDet/tools/train_utils/train_utils.py", line 79, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/mhahner/repositories/PCDet/tools/../pcdet/models/__init__.py", line 44, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/mhahner/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mhahner/repositories/PCDet/tools/../pcdet/models/detectors/pv_rcnn_plusplus.py", line 28, in forward
    batch_dict = self.pfe(batch_dict)
  File "/home/mhahner/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mhahner/repositories/PCDet/tools/../pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 372, in forward
    pooled_features = self.aggregate_keypoint_features_from_one_source(
  File "/home/mhahner/repositories/PCDet/tools/../pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 325, in aggregate_keypoint_features_from_one_source
    pooled_points, pooled_features = aggregate_func(
  File "/home/mhahner/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mhahner/repositories/PCDet/tools/../pcdet/ops/pointnet2/pointnet2_stack/pointnet2_modules.py", line 462, in forward
    cur_xyz, cur_features = self.__getattr__(f'layer_{k}')(**kwargs)
  File "/home/mhahner/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mhahner/repositories/PCDet/tools/../pcdet/ops/pointnet2/pointnet2_stack/pointnet2_modules.py", line 396, in forward
    assert C % self.num_reduced_channels == 0, \
AssertionError: the input channels (1) should be an integral multiple of num_reduced_channels (2)
python-BaseException
```

sshaoshuai, 2022-03-22:
@MartinHahner , 
Sorry, I forget one thing. You need to change a value of the config as here:
```
MODEL.PFE.SA_LAYER.raw_points.NUM_REDUCED_CHANNELS: 2 --> 1
```
Since KITTI dataset only has one extra point-wise feature channel for intensity.

MartinHahner, 2022-03-23:
With your suggested change, it does not crash anymore. 
I also noticed and changed `EVAL_METRIC: waymo` to `EVAL_METRIC: kitti`.

sshaoshuai, 2022-03-23:
Great, thank you! 
I have also updated the config in my answer.

krauwu, 2023-04-27:
hello！could I know some tricks in train a pvrcnn++ net, or some set of hyperparam?

r-gthb, 2023-10-25:
Hello, would I like to know about pandaset dataset training on PV RCNN model, have been trying to perform it but unsuccessful? any help would be appreciated.