open-mmlab/OpenPCDet #833: assert len(gt_annos) == len(dt_annos) AssertionError

dk-liang, 2022-03-01:
When I run python test.py, it shows:

eval: 100%|████| 942/942 [02:32<00:00,  6.19it/s, recall_0.3=(0, 16787) / 17544]
2022-03-01 14:53:39,144   INFO  *************** Performance of EPOCH 80 *****************
2022-03-01 14:53:39,144   INFO  Generate label finished(sec_per_example: 0.0404 second).
2022-03-01 14:53:39,145   INFO  recall_roi_0.3: 0.000000
2022-03-01 14:53:39,145   INFO  recall_rcnn_0.3: 0.956851
2022-03-01 14:53:39,145   INFO  recall_roi_0.5: 0.000000
2022-03-01 14:53:39,145   INFO  recall_rcnn_0.5: 0.911024
2022-03-01 14:53:39,145   INFO  recall_roi_0.7: 0.000000
2022-03-01 14:53:39,145   INFO  recall_rcnn_0.7: 0.711012
2022-03-01 14:53:39,163   INFO  Average predicted number of objects(3768 samples): 35.277

Traceback (most recent call last):
  File "/data/dkliang/projects/synchronous/OpenPCDet-master/tools/test.py", line 200, in <module>
    main()
  File "/data/dkliang/projects/synchronous/OpenPCDet-master/tools/test.py", line 196, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "/data/dkliang/projects/synchronous/OpenPCDet-master/tools/test.py", line 64, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/data/dkliang/projects/synchronous/OpenPCDet-master/tools/eval_utils/eval_utils.py", line 113, in eval_one_epoch
    output_path=final_output_dir
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 361, in evaluation
    ap_result_str, ap_dict = kitti_eval.get_official_eval_result(eval_gt_annos, eval_det_annos, class_names)
  File "../pcdet/datasets/kitti/kitti_object_eval_python/eval.py", line 675, in get_official_eval_result
    gt_annos, dt_annos, current_classes, min_overlaps, compute_aos, PR_detail_dict=PR_detail_dict)
  File "../pcdet/datasets/kitti/kitti_object_eval_python/eval.py", line 588, in do_eval
    min_overlaps, compute_aos)
  File "../pcdet/datasets/kitti/kitti_object_eval_python/eval.py", line 469, in eval_class
    assert len(gt_annos) == len(dt_annos)
AssertionError

Process finished with exit code 1

LouisSF, 2022-03-01:
Hi, this is due to a difference between the number of annotated scenes in the val set (3769) and the number of scenes that you predicted (3768). You may be skipping a scene, check if you have all the correct data or if you did a modification which makes you skip one of the predictions. For reference this is what I get for a PV-RCNN model trained for 80 epochs:
```
2022-02-07 16:25:03,791   INFO  ==> Checkpoint trained from version: pcdet+0.5.0+0000000+py4d51b52
2022-02-07 16:25:03,842   INFO  ==> Done (loaded 367/367)
2022-02-07 16:25:03,885   INFO  *************** EPOCH 80 EVALUATION *****************
2022-02-07 16:29:38,453   INFO  *************** Performance of EPOCH 80 *****************
2022-02-07 16:29:38,453   INFO  Generate label finished(sec_per_example: 0.0728 second).
2022-02-07 16:29:38,454   INFO  recall_roi_0.3: 0.969359
2022-02-07 16:29:38,454   INFO  recall_rcnn_0.3: 0.969587
2022-02-07 16:29:38,454   INFO  recall_roi_0.5: 0.930459
2022-02-07 16:29:38,454   INFO  recall_rcnn_0.5: 0.936610
2022-02-07 16:29:38,454   INFO  recall_roi_0.7: 0.711072
2022-02-07 16:29:38,454   INFO  recall_rcnn_0.7: 0.752421
2022-02-07 16:29:38,457   INFO  Average predicted number of objects(3769 samples): 10.314
```

closed 2022-04-16 by github-actions[bot]

SweetBittErz, 2022-11-23:
I have the same problem as you. Have you solved it? What causes it?

wangyakunn, 2023-06-16:
Batch_size need to be set to 1 when testing or evaluating.

SweetBittErz, 2023-06-16:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

Adeel-farukh3, 2023-07-05:
I am facing this error while evaluating: sudo python3 evaluate.py --ckpt /home/adeel/PointPillars-Camera-LiDAR-Fusion/CLOC_fusion/pretrained/Point_pillars.pth --data_root /home/adeel/PointPillars-Camera-LiDAR-Fusion/PointPillars/dataset/kitti
/home/adeel/.local/lib/python3.6/site-packages/numba/errors.py:137: UserWarning: Insufficiently recent colorama version found. Numba requires colorama >= 0.3.9
  warnings.warn(msg)
Predicting and Formatting the results.
100%|███████████████████████████████████████████████████████████████| 3712/3712 [03:40<00:00, 16.80it/s]
Evaluating.. Please wait several seconds.
Traceback (most recent call last):
  File "evaluate.py", line 405, in <module>
    main(args)
  File "evaluate.py", line 389, in main
    do_eval(format_results, val_dataset.data_infos, CLASSES, saved_path)
  File "evaluate.py", line 42, in do_eval
    assert len(det_results) == len(gt_results)
AssertionError

junmeng6025, 2024-03-04:
I happened to set drop_last as True in `pcdet/datasets/__init__.py`, you can check if you had this