open-mmlab/OpenPCDet #835: RuntimeError: Error compiling objects for extension #19

zhen12079, 2022-03-01:
你好，我安装的环境是
ubantu18.04
torch1.8
cuda11.1
我的编译一直出错
**(auto) auto@auto-System-Product-Name:~/OpenPCDet$ python setup.py develop**
/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/setuptools/dist.py:487: UserWarning: Normalizing '0.5.2+0000000' to '0.5.2+0'
  warnings.warn(tmpl.format(**locals()))
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /home/auto/OpenPCDet/build/temp.linux-x86_64-3.6/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/1] /usr/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/auto/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o.d -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include/TH -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include/THC -I/home/auto/anaconda3/envs/auto/include/python3.6m -c -c /home/auto/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /home/auto/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
FAILED: /home/auto/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o 
/usr/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/auto/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o.d -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include/TH -I/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/include/THC -I/home/auto/anaconda3/envs/auto/include/python3.6m -c -c /home/auto/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /home/auto/OpenPCDet/build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
nvcc fatal   : Unknown option '-generate-dependencies-with-compile'
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 1673, in _run_ninja_build
    env=env)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "setup.py", line 116, in <module>
    'src/sampling_gpu.cu',
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/setuptools/command/develop.py", line 114, in install_for_development
    self.run_command('build_ext')
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 708, in build_extensions
    build_ext.build_extensions(self)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 538, in unix_wrap_ninja_compile
    with_cuda=with_cuda)
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 1359, in _write_ninja_file_and_compile_objects
    error_prefix='Error compiling objects for extension')
  File "/home/auto/anaconda3/envs/auto/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 1683, in _run_ninja_build
    raise RuntimeError(message) from e
**RuntimeError: Error compiling objects for extension**

麻烦指导一下，感谢！

SaMeEr9597, 2022-03-09:
Hi Zhen, 
I am facing the same issue, were you able to solve it 
Thanks

Keenthan, 2022-04-22:
I solved this problem by switching pytorch version from 1.11 to 1.10

closed 2022-06-07 by github-actions[bot]