open-mmlab/OpenPCDet #837: How to use testing data

ObsisMc, 2022-03-03:
test.py is used on validation data and demo.py can be used on testing data but it seems that it can only test one .bin at one time. 
Are there any codes that can test multiple  testing data at one time?

sshaoshuai, 2022-03-09:
You need to modify the demo.py by yourself if you would like to show several examples at the same time.

ObsisMc, 2022-03-09:
Ok, thanks.

ObsisMc, 2022-03-09:
By the way, is it possible for you to add more parameters to demo.py to implement more functions like ouputing the inference results to a txt?

MartinHahner, 2022-03-09:
> By the way, is it possible for you to add more parameters to demo.py to implement more functions like ouputing the inference results to a txt? If it is ok, maybe I can help you.

You can always make code changes by yourself and if you think others could make use of and appreciate this functionality as well, create a pull request. The code snippets you mentioned (e.g. see below) should all be in the repo already, they just need to be glued together and added to [demo.py](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/demo.py) according to your wishes. Be aware though, that demo.py should be kept as generic as possible in order to handle all kinds of different custom datasets.

https://github.com/open-mmlab/OpenPCDet/blob/9c8e583e28e45582565175f848b7a2538eb6acf6/pcdet/datasets/kitti/kitti_dataset.py#L336-L349

sshaoshuai, 2022-03-10:
Thanks for Martin, this is exactly what I want to say. 

The `demo.py` is just utilized to show the usage of our visualization tool, and you can always custimize it according to your requirements.

ObsisMc, 2022-03-11:
Ok, I got it. I will keep my code in my own repo.

closed 2022-04-25 by github-actions[bot]