open-mmlab/OpenPCDet #839: CUDA kernel failed : invalid configuration argument

Leozyc-waseda, 2022-03-03:
Happened when I was trying to train PV-RCNN++, by the way I can train PV-RCNN normally.
Any suggestions on this error？

***
openpcdet/lib/python3.7/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  ../aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
epochs:   0%|                                                           | 0/30 [00:18<?, ?it/s, loss=12, lr=0.001, d_time=0.00(0.02), f_time=0.14(0.26), b_time=0.30(0.58)]CUDA kernel failed : invalid configuration argument                                                                         | 31/4880 [00:17<32:08,  2.52it/s, total_it=31]
***

sshaoshuai, 2022-03-08:
Sorry, the given information is not enough to debug this error since I trained it succssfully with the released codes.

Leozyc-waseda, 2022-03-09:
After I reinstall my  Nvidia driver, Cudnn, Cuda, Solved.

closed 2022-03-09 by Leozyc-waseda

xiangruhuang, 2022-03-30:
Hi @Leozyc-waseda , do you minding sharing the versions you end up with? I'm having the exact same error. For me, it occurs indefinitely (about 5~6 times out of 10).

LZDSJTU, 2022-04-11:
@xiangruhuang I also met the problem. How did you solve this problem?

xiangruhuang, 2022-04-11:
@LZDSJTU I didn't. I tried multiple environments including the one @Leozyc-waseda mentioned. On a server using 8 Tesla V100 GPUs, the program ends up with the error indefinitely. But if you keep repeating and trying, it eventually runs. Yet for the next run, you have to repeat again.

xiangruhuang, 2022-04-13:
@LZDSJTU @Leozyc-waseda 

I think I found a bug.

In [this line](https://github.com/open-mmlab/OpenPCDet/blob/878ad2e0016f3a5c4b3c4d5c86c9e9eebb230a62/pcdet/ops/pointnet2/pointnet2_stack/src/vector_pool_gpu.cu#L473), `blocks.x` is determined by dividing `num_max_sum_points` by 256 and rounding up. For `num_max_sum_points=0`, `blocks.x` is set to 0 and hence triggering cuda's error : `invalid configuration error`.

The code is clearly assuming `num_max_sum_points > 0`, yet in my experiments, there are situations it equals 0 and causing the error.

I added another line in the same .cu file to set `blocks.x` to 1 if it equals 0, and rebuilt `pcdet`. The error has never came back since then.

@sshaoshuai Can you verify if this is a bug?

sshaoshuai, 2022-04-13:
Thank you for reporting the bug!

Yes, I have also met a similar situation before. Maybe we need add an assert line to explicitly assume that there is at least 1 point before we call this function (in its corresponding python function).
Your solution can avoid this situation but I am not sure what it will return since we should never call this function if you do not have any points. 
I prefer to add an assert and leave the problem to the user who needs to take care their specific situations.  

Can you help to submit a PR?

xiangruhuang, 2022-04-13:
Your suggestion makes sense. I'll think about it and get back to you soon.

xiangruhuang, 2022-04-13:
Actually, to my understanding, getting `grouped_idxs.shape[0]==0` from `VectorPoolWithVoxelQuery` (see [here](https://github.com/open-mmlab/OpenPCDet/blob/878ad2e0016f3a5c4b3c4d5c86c9e9eebb230a62/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_utils.py#L407)) is a random event due to bad choices of pooling center locations. 

In this case, pooled features will be all zeros as initialized and hence no gradient should propagate from pooled features to the support features. I personally would prefer to check if `grouped_idxs.shape[0] > 0` before calling the pooling cuda function in [this line](https://github.com/open-mmlab/OpenPCDet/blob/878ad2e0016f3a5c4b3c4d5c86c9e9eebb230a62/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_utils.py#L444). 

What do you think? I can create a PR if this works for you.

sshaoshuai, 2022-04-13:
It does make sense. 
I just found that it is not the problem of the data and the error just comes from some bad center locations as you said. 

Thank you so much for finding the bug!

xiangruhuang, 2022-04-13:
Got it! Will make a PR tomorrow.