open-mmlab/OpenPCDet #840: Evaluation during training

YCAyca, 2022-03-04:
Hello, I was wondering if there is a way to evaluate the model "during training". Not at the end of the training with a second command like test.py but to obtain validation accuracy and loss to compare the ones with training loss and accuracy (I didnt see any training accuracy information in mi tensorboard graphs though) Do you have any idea? It would be very useful to make a real research and finetuning to have train loss + accuracy - validation loss + accuracy informations. For example here is my train log in tensorboard with only loss information

/home/yagmur/Pictures/Screenshot from 2022-03-04 19-46-28.png

MartinHahner, 2022-03-04:
https://github.com/open-mmlab/OpenPCDet/issues/414

YCAyca, 2022-03-05:
thank you for the answer! I got the reason but if I wouldnt care about the time of training how would it be possible to add this mechanism do you have any idea? The code seemed to me a little bit complicated for this

MartinHahner, 2022-03-05:
In the train loop after calling `train_one_epoch` (e.g. [here](https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/tools/train_utils/train_utils.py#L120)), you could add code from [here](https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/tools/train.py#L180-190) and [there](https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/tools/test.py#L115-132).

Then these [lines](https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/tools/train.py#L192) can be omitted since you already evaluated during training.
If you can make it work, it would be great if you could create a pull request.
As I said before, I think this should be the way to go. 

Something that came to my mind while I was formulating this answer is that maybe the memory footprint gets too large when you construct the `test_set, test_loader, sampler` in the train loop (because then you effectively have two datasets loaded at the same time - training and evaluation), maybe that's the reason why it was done one (training) after the other (evaluation) and not intertwined.
https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/tools/train.py#L182

YCAyca, 2022-03-05:
Im done with training time evaluation thank you for your instructions. Another very important thing for the research part and I try to implement is just to add "training accuracy" information. So basically I just do the same steps for training data but I dont know why, I get an error as below. Doesnt train and test loader loads the data in a similar way?

  #evaluate validation dataset - calculate validation loss + accuracy
    tmp = str(ckpt_name) +'.pth'
    eval_output_dir = ckpt_save_dir / 'validation_accuracy' / ('checkpoint_epoch_%d' % trained_epoch) #result.pkl
                
    tb_log = SummaryWriter(log_dir=str(ckpt_save_dir / 'validation_accuracy' / 'tensorboard')) 
    eval_single_ckpt(model, test_loader, tmp, eval_output_dir, logger, trained_epoch, dist_test=False, cfg=cfg,tb_log=tb_log, val_loss=True)
             
                    
             
  # evaluate train dataset - calculate train accuracy
  eval_output_dir = ckpt_save_dir / 'train_accuracy' / ('checkpoint_epoch_%d' % trained_epoch)
  tb_log = SummaryWriter(log_dir=str(ckpt_save_dir / 'train_accuracy' / 'tensorboard')) 
              
   eval_single_ckpt(model, train_loader, tmp, eval_output_dir, logger, trained_epoch, dist_test=False, cfg=cfg, tb_log=tb_log)
              

[image]

YCAyca, 2022-03-05:
Sure  I will send a pull request after finishing train accuracy problem !!

MartinHahner, 2022-03-06:
I don't know, you need to debug where 'calib' gets added (or not added) to the `batch_dict`.

LouisSF, 2022-03-07:
I was working on this for my own research problems, but faced the same issue regarding calib, I'm not quite sure what causes this so for now i'm just using the following workaround:
```
for idx_anno, anno in enumerate(annos):
   try:
                  calib = annos[idx_anno]['calib']
              except KeyError:
                  print("No calib found!")
                  calib = self.prev_calib
              self.prev_calib = calib
              try:
                  image_shape = annos[idx_anno]['image_shape']
              except KeyError:
                  print("No image_shape found!")
                  image_shape = self.prev_image_shape
              self.prev_image_shape = image_shape
```
I also had to do this for image_shape as I was facing the same issue. This will probably cause some slight accuracy errors but there are only a few frames of KITTI for which we get a KeyError

YCAyca, 2022-03-07:
Actually I updated the code and now I obtain train loss, train acc and validation acc during training. But in somehow it effected my results so now when I make an inference I obtain worse results than before. Train acc doesnt seem nice too. 

The main problem is when you use dataloader for training set (where you give training=True as parameter) you dont obtain calib info in your dataset info, otherwise you obtain. 

def build_dataloader(dataset_cfg, class_names, batch_size, dist, root_path=None, workers=4,
                     logger=None, training=True, merge_all_iters_to_one_epoch=False, total_epochs=0):

    dataset = __all__[dataset_cfg.DATASET](
        dataset_cfg=dataset_cfg,
        class_names=class_names,
        root_path=root_path,
        training=training,
        logger=logger,
    )
    
    print("DATASET",training, dataset[0]) # Test dataset includes "calib" info but train dataset doesnt

    if merge_all_iters_to_one_epoch:
        assert hasattr(dataset, 'merge_all_iters_to_one_epoch')
        dataset.merge_all_iters_to_one_epoch(merge=True, epochs=total_epochs)

    if dist:
        if training:
            sampler = torch.utils.data.distributed.DistributedSampler(dataset)
        else:
            rank, world_size = common_utils.get_dist_info()
            sampler = DistributedSampler(dataset, world_size, rank, shuffle=False)
    else:
        sampler = None
    dataloader = DataLoader(
        dataset, batch_size=batch_size, pin_memory=True, num_workers=workers,
        shuffle=(sampler is None) and training, collate_fn=dataset.collate_batch,
        drop_last=False, sampler=sampler, timeout=0
    )

    return dataset, dataloader, sampler

I tracked the issue and even at first initialize in kitti_dataset.py class KittiDataset(DatasetTemplate): both train and test datasets comes with the equal informations, the calibration info is deleted in dataset.py prepare_dataset() function. I commented out the lines where training mode make the calib deleted and it worked. But as I said my results went bad so I guess I will just give up from train accuracy information...

x-special/nautilus-clipboard
copy
file:///home/yagmur/Pictures/Screenshot%20from%202022-03-07%2011-43-59.png

YCAyca, 2022-03-07:
As a result we obtain these 2 tensorboard logs one for train loss and acc one for validation acc

[image]
[image]

YCAyca, 2022-03-07:
And finally I realized the situation...

in datasey.py, the following part causes to lose calib info training dataset as I said. But I understand the reason now: if its in training mode, we apply "data augmentation" in self.data_augmentor.forward() function. And in this function calib info is deleted, only the ground truth boxes are updated according to the augmentation technique. So since I removed data augmentation from training,  my results went kind a bad. To not lose data augmentation and still be able to obtain train time accuracy, I will try to load train data in test mode with some additional control flags 

if self.training:
            assert 'gt_boxes' in data_dict, 'gt_boxes should be provided for training'
            gt_boxes_mask = np.array([n in self.class_names for n in data_dict['gt_names']], dtype=np.bool_)

            data_dict = self.data_augmentor.forward(
                data_dict={
                    **data_dict,
                    'gt_boxes_mask': gt_boxes_mask
                }
            )

YCAyca, 2022-03-07:
Its done and I will do a pull request but the commits got a little bit complicated since the last 3 commit includes the changes for this part but I dont know exactly how to select the parts to pull request and the parts not to. Maybe you can take a look at the pull request and we can talk about @MartinHahner

MartinHahner, 2022-03-07:
> I don't know, you need to debug where 'calib' gets added (or not added) to the `batch_dict`.

https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/pcdet/datasets/augmentor/database_sampler.py#L165
https://github.com/open-mmlab/OpenPCDet/blob/a9c66fee5348dcfb5f73ccca71becc2871ce3a78/pcdet/datasets/augmentor/data_augmentor.py#L246

closed 2022-04-21 by github-actions[bot]