open-mmlab/OpenPCDet #844: web cam camera integration

kurtwagner19, 2022-03-06:
I am trying to test your network on a web camera, however, since there is no need for LIDAR devices in testing, what steps should be taken as an easy fix to project the estimated 3D bounding boxes in LIDAR devices in testing, what steps should be taken as an easy fix to project the estimated 3D bounding boxes in LIDAR coordinate system to image plabe ( since the model unprojects to LIDAR coordinate system). I have the intrinsic matrix of the camera estimated by opencv library. Is P2 the projection matrix of the camera? Do we need to  change the values of tr_imu_to_velo and tr_velo_to_cam values? And what's your suggestion on how to change P2 projection matrix using only the intrinsic matrix? Do we weed to assume any rotation or translation? or for construction of such matrix, should we assume there is no rotation and transformation?

MartinHahner, 2022-03-06:
@codyreading

kurtwagner19, 2022-03-11:
In other words, would you provide a guidline on how to test the model with new images from a camera and calibration matrices acquired from opencv?
We are using webcam images and replaced the calibration matrices, however we do not get proper results.
Since we are using single cameras, i believe for testing, the model should not require camera extrinsic parameters. (therefore it is confusing, however we assumed any extrinsic parameters givens should not affect the performance- Though not sure if that is the case).

-A general guideline would be helpful on how to test the pretrained model on a custom RGB camera. What hyper-parameters should be considered to be changed in order for the system to run properly and what hyper parameters should be tweaked with respect to the camera intrinsic parameters to have better performance.

-Another question would be, is there any work around to test the network on an image without having the intrinsic parameters of the camera? Let's say we have a youtube video, and all we have is the resolution of the image? Do you think the detector is capable of estimating the depth of the objects up to a scale?( meaning the relative positioning of objects in 3D world is estimated up to a scale and then using a reference(such as size of the vehicles) we can rectify the distances.)

closed 2022-04-25 by github-actions[bot]