open-mmlab/OpenPCDet #848: Question about train.py

yun9993, 2022-03-07:
Traceback (most recent call last):
  File "train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/nas/lyp/OpenPCDet-master/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/nas/lyp/OpenPCDet-master/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/nas/lyp/OpenPCDet-master/tools/../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/nas/lyp/OpenPCDet-master/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/nas/lyp/OpenPCDet-master/tools/../pcdet/models/detectors/detector3d_template.py", line 8, in <module>
    from .. import backbones_2d, backbones_3d, dense_heads, roi_heads
  File "/nas/lyp/OpenPCDet-master/tools/../pcdet/models/backbones_3d/__init__.py", line 2, in <module>
    from .spconv_backbone import VoxelBackBone8x, VoxelResBackBone8x
  File "/nas/lyp/OpenPCDet-master/tools/../pcdet/models/backbones_3d/spconv_backbone.py", line 30, in <module>
    class SparseBasicBlock(spconv.SparseModule):
AttributeError: module 'spconv' has no attribute 'SparseModule'

LouisSF, 2022-03-07:
Hey there, this is probably due to the version of spconv you are using, see the following page for more information: [[](https://github.com/traveller59/spconv/blob/master/docs/SPCONV_2_BREAKING_CHANGEs.md#breaking-changes-in-spconv-2x-for-spconv-1x-users)Breaking changes in Spconv 2.x for spconv 1.x users](https://github.com/traveller59/spconv/blob/master/docs/SPCONV_2_BREAKING_CHANGEs.md)

You can probably just replace all instances of `spconv.xxx` by `spconv.pytorch.xxx`

yun9993, 2022-03-11:
> 嘿，这可能是由于您正在使用的 spconv 版本，有关更多信息，请参阅以下页面：[Spconv 2.x 中针对 spconv 1.x 用户的重大更改](https://github.com/traveller59/spconv/blob/master/docs/SPCONV_2_BREAKING_CHANGEs.md)
> 
> 您可能只需替换`spconv.xxx`by 的所有实例`spconv.pytorch.xxx`

thank you very much! this is useful！

SaMeEr9597, 2022-03-11:
I am getting the same error, I did not understand which file should we change, could you please let me know which file should we edit, Thanks

SaMeEr9597, 2022-03-12:
> > 嘿，这可能是由于您正在使用的 spconv 版本，有关更多信息，请参阅以下页面：[Spconv 2.x 中针对 spconv 1.x 用户的重大更改](https://github.com/traveller59/spconv/blob/master/docs/SPCONV_2_BREAKING_CHANGEs.md)
> > 您可能只需替换`spconv.xxx`by 的所有实例`spconv.pytorch.xxx`
> 
> thank you very much! this is useful！

Hi Yun, 
I'm getting the same error and I'm new to this, can you please let me know which file should I edited and what should I edit in that, 
Thanks

SaMeEr9597, 2022-03-12:
> Hey there, this is probably due to the version of spconv you are using, see the following page for more information: [Breaking changes in Spconv 2.x for spconv 1.x [...]

Hi Lous,
I'm getting the same error and I'm new to this, can you please let me know which file should I edited and what should I edit in that,
And also what version of spconv were you using
Thanks

LegendSun0, 2022-03-18:
> > 嘿，这可能是由于您正在使用的 spconv 版本，有关更多信息，请参阅以下页面：[Spconv 2.x 中针对 spconv 1.x 用户的重大更改](https://github.com/traveller59/spconv/blob/master/docs/SPCONV_2_BREAKING_CHANGEs.md)
> > 您可能只需替换`spconv.xxx`by 的所有实例`spconv.pytorch.xxx`
> 
> 嗨，Lous， 我遇到了同样的错误，我是新手，你能告诉我应该编辑哪个文件，我应该在其中编辑什么， 还有你使用的是什么版本的 spconv 谢谢 我正在使用spconv 的 2.1.21 版本

Hello, has the problem been solved?

SaMeEr9597, 2022-03-18:
> Hello, has the problem been solved?

Nope, not yet, I was getting this error first : 
[image]

And then I tried changing the spconv_backbone.py and spconv_unet.py initial importing lines to this ;
from functools import partial
import torch.nn as nn
from spconv.pytorch import ops
from spconv.pytorch.conv import (SparseConv2d, SparseConv3d, SparseConvTranspose2d,
                         SparseConvTranspose3d, SparseInverseConv2d,
                         SparseInverseConv3d, SubMConv2d, SubMConv3d)
from spconv.pytorch.core import SparseConvTensor
from spconv.pytorch.identity import Identity
from spconv.pytorch.modules import SparseModule, SparseSequential
from spconv.pytorch.ops import ConvAlgo
from spconv.pytorch.pool import SparseMaxPool2d, SparseMaxPool3d
from spconv.pytorch.tables import AddTable, ConcatTable
from ...utils.spconv_utils import replace_feature, spconv

the error I got was this : 
[image]

closed 2022-05-02 by github-actions[bot]