open-mmlab/OpenPCDet #854: Is anyone working on implementing a CARLA dataset in OpenPCDet for use of simulation data in training?

Javier-DlaP, 2022-03-09:
I have been trying to create an implementation for a CARLA dataset by recording the data given by this simulator.Then I have been trying to emulate the simulator data through the KITTI dataset so that I don't have to create a new implementation of this dataset in OpenPCDet.

Has anyone tried to implement a similar dataset using the CARLA simulation data in this project?

LouisSF, 2022-03-15:
Hi, I might work one day on implementing Carla-like datasets in the future, such as [KITTI-CARLA dataset](https://npm3d.fr/kitti-carla) the style of KITTI. I will keep you informed if I start this.

closed 2022-04-29 by github-actions[bot]

AOOOOOA, 2022-05-10:
> Hi, I might work one day on implementing Carla-like datasets in the future, such as [KITTI-CARLA dataset](https://npm3d.fr/kitti-carla) the style of KITTI. I will keep you informed if I start this.

Hi, have you started the work yet? If yes, maybe we can discuss some problems with it.

makaveli10, 2022-05-20:
@AOOOOOA Can you please tell what problems are you facing?