open-mmlab/OpenPCDet #856: waymo dataset evaluation problem

dotorl127, 2022-03-10:
Hello, I just wanna evaluation pre-trained model and I installed waymo open dataset via pip3

```
$ pip3 freeze | grep waymo
waymo-open-dataset-tf-2-5-0==1.4.1
```

But error occur about tensorflow when testing model

```
2022-03-10 12:55:17,115   INFO  *************** EPOCH 20 EVALUATION *****************
eval: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 7998/7998 [09:04<00:00, 14.69it/s, recall_0.3=(0, 154468) / 360251]
2022-03-10 13:04:21,455   INFO  *************** Performance of EPOCH 20 *****************
2022-03-10 13:04:21,455   INFO  Generate label finished(sec_per_example: 0.0681 second).
2022-03-10 13:04:21,455   INFO  recall_roi_0.3: 0.000000
2022-03-10 13:04:21,455   INFO  recall_rcnn_0.3: 0.428779
2022-03-10 13:04:21,455   INFO  recall_roi_0.5: 0.000000
2022-03-10 13:04:21,455   INFO  recall_rcnn_0.5: 0.411560
2022-03-10 13:04:21,455   INFO  recall_roi_0.7: 0.000000
2022-03-10 13:04:21,455   INFO  recall_rcnn_0.7: 0.280488
2022-03-10 13:04:21,457   INFO  Average predicted number of objects(7998 samples): 19.844
2022-03-10 13:04:22.144611: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
Traceback (most recent call last):
  File "test.py", line 199, in <module>
    main()
  File "test.py", line 195, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 63, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/media/moon/extraDB/OpenPCDet/tools/eval_utils/eval_utils.py", line 113, in eval_one_epoch
    output_path=final_output_dir
  File "/media/moon/extraDB/OpenPCDet/pcdet/datasets/waymo/waymo_dataset.py", line 246, in evaluation
    ap_result_str, ap_dict = waymo_eval(eval_det_annos, eval_gt_annos)
  File "/media/moon/extraDB/OpenPCDet/pcdet/datasets/waymo/waymo_dataset.py", line 226, in waymo_eval
    from .waymo_eval import OpenPCDetWaymoDetectionMetricsEstimator
  File "/media/moon/extraDB/OpenPCDet/pcdet/datasets/waymo/waymo_eval.py", line 11, in <module>
    from waymo_open_dataset.metrics.python import detection_metrics
  File "/home/moon/.local/lib/python3.7/site-packages/waymo_open_dataset/metrics/python/detection_metrics.py", line 20, in <module>
    from waymo_open_dataset.metrics.ops import py_metrics_ops
  File "/home/moon/.local/lib/python3.7/site-packages/waymo_open_dataset/metrics/ops/py_metrics_ops.py", line 24, in <module>
    tf.compat.v1.resource_loader.get_path_to_datafile('metrics_ops.so'))
  File "/home/moon/.local/lib/python3.7/site-packages/tensorflow/python/framework/load_library.py", line 57, in load_op_library
    lib_handle = py_tf.TF_LoadLibrary(library_filename)
tensorflow.python.framework.errors_impl.NotFoundError: /home/moon/.local/lib/python3.7/site-packages/waymo_open_dataset/metrics/ops/metrics_ops.so: undefined symbol: _ZN10tensorflow11GetNodeAttrERKNS_9AttrSliceEN4absl14lts_2020_09_2311string_viewEPSs
```

What should I do for testing model

closed 2022-04-25 by github-actions[bot]

yancie-yjr, 2022-05-03:
hi, I meet the same issue, how to tackle?

dotorl127, 2022-05-05:
> hi, I meet the same issue, how to tackle?

Waymo open dataset's version was the problem.
I solved it by lowering the waymo open dataset tf version to 2.4.
try `$pip3 intstall waymo-open-dataset-tf-2-4-0`

YuRui-Learning, 2024-01-11:
ello, when i run test or train function , i will also meet this question in eval. It is some issue aboud tensorflow ?
Traceback (most recent call last): File "eval.py", line 207, in <module> main() File "eval.py", line 203, in main eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test) File "eval.py", line 65, in eval_single_ckpt eval_result.eval_one_epoch( File "/root/MSF/tools/eval_utils/eval_result.py", line 62, in eval_one_epoch result_str, result_dict = dataset.evaluation( File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 476, in evaluation ap_result_str, ap_dict = waymo_eval(eval_det_annos, eval_gt_annos) File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 459, in waymo_eval ap_dict = eval.waymo_evaluation( File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_eval.py", line 211, in waymo_evaluation metrics = self.build_graph(graph) File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_eval.py", line 127, in build_graph metrics = detection_metrics.get_detection_metric_ops( File "/root/.local/lib/python3.8/site-packages/waymo_open_dataset/metrics/python/detection_metrics.py", line 134, in get_detection_metric_ops ground_truth_speed = tf.zeros((num_gt_boxes, 2), tf.float32) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/util/dispatch.py", line 206, in wrapper return target(*args, **kwargs) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2912, in wrapped tensor = fun(*args, **kwargs) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2962, in zeros output = _constant_if_small(zero, shape, dtype, name) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2897, in _constant_if_small if np.prod(shape) < 1000: File "<__array_function__ internals>", line 200, in prod File "/usr/local/lib/python3.8/dist-packages/numpy/core/fromnumeric.py", line 3076, in prod return _wrapreduction(a, np.multiply, 'prod', axis, dtype, out, File "/usr/local/lib/python3.8/dist-packages/numpy/core/fromnumeric.py", line 86, in _wrapreduction return ufunc.reduce(obj, axis, dtype, out, **passkwargs) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/ops.py", line 867, in __array__ raise NotImplementedError( NotImplementedError: Cannot convert a symbolic Tensor (strided_slice:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a NumPy call, which is not supported eval:   0%|          | 0/4931 [02:31<?, ?it/s]